ZuluSCSI/ZuluSCSI-firmware #260: Imaged drive details

djfarrell, 2023-07-08:
Maybe add a option to write more drive details , serial number, etc to a base ini file.  Maybe this exists already? If a software license is locked to a drive serial number it would be nice to accomodate this.

MeTalOneOEight, 2023-07-09:
Take a look at the example zuluscsi.ini:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/main/zuluscsi.ini
It's already there.

djfarrell, 2023-07-09:
Thanks for the response.  I meant implement features in the initiator function, I see that you can write the serial number, but you have to know it first.

csulaguy, 2023-10-31:
Same here. When I look at the logs, it shows the drive size, but doesn't pull information, like vendor, product, version, and serial.
`[7569ms] SCSI id 0 capacity 82080 sectors x 512 bytes
[7570ms] Drive total size is 40 MiB
[7573ms] Preallocating image file
[7575ms] Starting to copy drive data to HD00_imaged.hda
[7890ms] SCSI read succeeded, sectors done: 512 / 82080 speed 842 kB/s

blah blah blah
[54796ms] Finished imaging drive with id 0
`

Even pulling geometry reported might be nice, say when imaging drives and then deploying said image via ZuluSCSI on a similar or same host that's expecting that same drive (but doesn't have to know it's emulated, of course).