ZuluSCSI/ZuluSCSI-firmware #261: v2023.07.01 ZuluSCSI V1.1 Slow Writes

mypcisamiga, 2023-07-09:
With latest version v2023.07.01, I am getting extremely slow writes.   I did a copy of a barebone AmigaOS 3.2.2.1 install between partitions of a single HD image and is taking more than 2+ hours.   I stopped it before it completed.  

With previous versions of the FW, I am getting an Amiga software failure during boot when  IPrefs is loading from startup-sequence and get guru 8000 0004.  This was verified on 2 systems -- on board SCSI of the A4000T as well as the A4091 board on an A4000 desktop.

aperezbios, 2023-07-09:
@mypcisamiga thanks for the report. Can you please specify which previous versions you've tested with? Be as specific as possible. Is your kickstart ROM also version 3.2.2.1, or another version? We don't have access to a 3.2.2.1 system to test with, nor an A4000T, so we're going to need additional specificity, and, ideally, a debug log. It would probably also be worth testing whether or not this also happens if you limit your ZuluSCSI to asynchronous mode, by setting the following in a barebones ZuluSCSI.ini:

```[SCSI]
MaxSyncSpeed=0```

If that makes it work reliably, please test with:

```[SCSI]
MaxSyncSpeed=5```

mypcisamiga, 2023-07-09:
Attached is the debug log.
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11995492/zululog.txt)

mypcisamiga, 2023-07-09:
Thanks ... let me try that *.ini config.    I have real 47.111 physical ROMs on both machines.

For ZuluSCSI FW, I have tried these 
v2023.07.01 -- this one, the AmigaOS can boot okay but writes are slow.
v2023.06.13 - Amiga failing to boot with a signature of a 800 004 guru when Iprefs is loaded.  
v2023.05.04 -- Amiga failing to boot with a signature of a 800 004 guru when Iprefs is loaded

mypcisamiga, 2023-07-09:
[zululog_23.06.13_MaxSyncSpeed0.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11995574/zululog_23.06.13_MaxSyncSpeed0.txt)
[zululog_23.06.13_MaxSyncSpeed5.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11995575/zululog_23.06.13_MaxSyncSpeed5.txt)
[zululog_23.07.01_MaxSyncSpeed0.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11995576/zululog_23.07.01_MaxSyncSpeed0.txt)

mypcisamiga, 2023-07-09:
I didn't mean to close issue.   A few comments on the logs:

zululog_23.06.13_MaxSyncSpeed0.txt--> boots okay but slow writes.
zululog_23.06.13_MaxSyncSpeed5.txt--> fails at IPREFS loading during boot.
zululog_23.07.01_MaxSyncSpeed0.txt--> boots okay but slow writes.

aperezbios, 2023-07-09:
@mypcisamiga thanks for the debug log. In it, there are 110 occurrences of the following:

```[100747ms] DBG SD card readSector(12937) slow transfer, buffer0x200003C4 vs. 0x20008A4C
[108734ms] DBG SD card readSector(12937) slow transfer, buffer0x200003C4 vs. 0x20008A4C
[116824ms] DBG SD card readSector(12937) slow transfer, buffer0x200003C4 vs. 0x20008A4C
[124810ms] DBG SD card readSector(12937) slow transfer, buffer0x200003C4 vs. 0x20008A4C
[132825ms] DBG SD card readSector(12937) slow transfer, buffer0x200003C4 vs. 0x20008A4C
[139802ms] DBG SD card readSector(12937) slow transfer, buffer0x200003C4 vs. 0x200017B4
[147812ms] DBG SD card readSector(12937) slow transfer, buffer0x200003C4 vs. 0x200017B4
[155803ms] DBG SD card readSector(12937) slow transfer, buffer0x200003C4 vs. 0x200017B4```

Given this information, can you share with us the model/specifications of the SD card you're using? Have you tried testing with a different SD card? Was the current SD card you're using purchased with the board? 

You could also try disabling the prefetch buffer, by setting PrefetchBytes=0 in the [SCSI] section of your zuluscsi.ini

I also noticed that your SCSI termination is disabled. Can you confirm that your ZuluSCSI is NOT on the end of the SCSI device chain, as that's the only circumstance in which it's appropriate to have termination disabled, unless there's a separate termination pack on the end of a ribbon cable, or something like that.

mypcisamiga, 2023-07-09:
Yes, this SD card was purchased with the ZuluSCSI card from Rabbit  Hole Computing (SanDisk Ultra 120MB/s).  I haven't tried a different SD card.    Yes, the termination is disabled on the ZuluSCSI as I have a separate active termination pack at the end of the cable.   Once I have a chance I will try to  disable the prefetch in the "ini" file.

mypcisamiga, 2023-07-09:
For 7MB (892 files), it took 42 minutes with PrefetchBytes=0 (all other settings not changed and are the default).  Attached is log with that.  This copy 
[zululog_23.07.01_PrefechBytes0.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11996496/zululog_23.07.01_PrefechBytes0.txt)
was done 2x and so maybe will see more transactions in the logs.

aperezbios, 2023-07-10:
@mypcisamiga thanks for the log. The one thing that sticks out is this:

```[325ms] ---- WARNING: file /HD10.bin is not contiguous. This will increase read latency.```

It's possible this could also be contributing to the slow writes, so I would recommend either defragmenting your SD card, or re-formatting it, and then copying the HD10.bin file back to the SD card, which will result in it effectively being defragmented, in a roundabout way. 

Modern versions of Windows don't seem to allow you to defragment an SD card easily, but you could also use Defraggler (https://www.ccleaner.com/defraggler/download) instead, if you're using Windows.

mypcisamiga, 2023-07-10:
Thanks -- will look into that.   I am also attaching log of using default settings of the PrefetchBypass (and everything else is also default).   
[zululog_23.07.01_PrefechBytes_Default.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11997163/zululog_23.07.01_PrefechBytes_Default.txt)

This also was ~43 minutes to copy the same set of files. 

For the 2+ hour that I mentioned above this was something that I was not timing exactly but also I was using an A4000 desktop with the A4091 --- I would have to try that combo again that maybe is worse with the A4091 scsi controller.

aperezbios, 2023-07-10:
@mypcisamiga great, I'm glad you got this resolved. we'll look in to seeing if it's possible to ensure we create contiguous files when formatting. It may or may not be possible, as we use a library to accomplish this.

closed 2023-07-10 by aperezbios