ZuluSCSI/ZuluSCSI-firmware #264: ZuluSCSI RP2040 Initiator Mode - Need help to get it working

aperezbios, 2023-07-13:
### Discussed in https://github.com/ZuluSCSI/ZuluSCSI-firmware/discussions/262

<div type='discussions-op-text'>

<sup>Originally posted by **guytaylor** July 12, 2023</sup>
I am trying to get Initiator Mode working. 
My understanding is that it allows the Zulu to image a drive, which is what I would like to do.

I have a full size RP2040, and have the latest firmware installed, v2023.07.01 and the drive is a Quantum Fireball SE (which was an a Yamaha A3000 sampler)
The drive was working fine in the sampler. The drive SCSI ID is 0

What I have tried is, connecting the ZuluSCSI directly to the drive using a 50 pin cable, and powering both.
I set the 'Initiator' jumper on the Zulu as well as the debug out.

But  I can't get it to see the drive. I tried all 4 combos of SCSI termination, and here are the logs.
I wait for 60 seconds for each run, and the logs just repeat, I trimmed the log output as they just are repeating

Any tips to what I may be doing wrong?
```</div>

@PetteriAimonen, FYI, I'm able to replicate this behavior locally with an Apple co-branded 1.2GB Quantum Fireball drive, model 655-0394.  Other Quantum drives I attempted to image completed successfully. 

Attached is the debug log

[zululog-initiator-quantum-fireball-dump-failure.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/12041860/zululog-initiator-quantum-fireball-dump-failure.txt)

aperezbios, 2023-08-15:
@PetteriAimonen any idea what might be going on here? See original log in linked discussion

PetteriAimonen, 2023-08-17:
If I understood correctly, the basic problem is that with specific model of drive, SCSI selection in initiator mode is failing. The watchdog timeouts when there is no termination on bus are to be expected, as signals would be stuck in weird states.

I wonder if this might be SCSI1/SCSI2 thing. Currently ZuluSCSI is not setting initiator ID and not setting ATN bit, i.e. it is behaving like a SCSI-1 host. Most drives support this but standard doesn't require it.

I have a test build here that sets the initiator ID, could be worth a try even though it is just a guess:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/5889395480

aperezbios, 2023-08-17:
@PetteriAimonen your educated guess seems to have been correct. See attached debug log file. I aborted the imaging early. Is this change safe to merge?

[zululog-issue-264-fix.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/12371657/zululog-issue-264-fix.txt)

I did observe multiple occurrences of the following, however. Trying different types/brands of brand new SD cards resulted in the same slow transfer messages, every few seconds, but drive imaging was working. Data was being written to the SD card at around 2.5-2.6MB/sec.

```[8205ms] DBG SD card writeSector(9356) slow transfer, buffer0x2001F0B4 vs. 0x20023D40
[8211ms] DBG SD card writeSector(16966) slow transfer, buffer0x2001F0B4 vs. 0x20023D40
[8214ms] DBG SD card readSector(9357) slow transfer, buffer0x2001F0B4 vs. 0x20023D40
[8215ms] DBG SD card writeSector(9357) slow transfer, buffer0x2001F0B4 vs. 0x20023D40
[8217ms] DBG SD card writeSector(16967) slow transfer, buffer0x2001F0B4 vs. 0x20023D40
[8219ms] DBG SD card readSector(9356) slow transfer, buffer0x2001F0B4 vs. 0x20023D40
[8231ms] DBG SD card writeSector(9356) slow transfer, buffer0x2001F0B4 vs. 0x2002BD40
[8232ms] DBG SD card writeSector(16966) slow transfer, buffer0x2001F0B4 vs. 0x2002BD40
[8234ms] DBG SD card readSector(9357) slow transfer, buffer0x2001F0B4 vs. 0x2002BD40
```

PetteriAimonen, 2023-08-18:
The slow transfer messages are probably caused by fragmented free space on the SD card. I think SdFat just selects the next free block instead of looking for a larger one. Not sure if it makes sense to log for each one, the debug messages may be slower than the actual transfer.

aperezbios, 2023-11-25:
This issue was fixed in PR #280

closed 2023-11-25 by aperezbios