ZuluSCSI/ZuluSCSI-firmware #265: ZuluSCSI RP2040 on Adaptec 1542CP - Inquiry Timeout

ajacocks, 2023-07-16:
I have a 386DX/40 PC with an Adaptec 1542CP SCSI controller, and while I can get the disk images on my SD card to be seen in the Adaptec SCSIselect BIOS utility, when trying to let the system boot, it times out on SCSI inquiry.

Based on previous issues, I have tried:
1) setting EnableUnitAttention = 1
2) setting EnableSCSI2 = 0
3) setting MaxSyncSpeed = 0

None of these have had an effect.

Here is the debug log from the most recent boot, which differs only in reported config settings from any of the other attempts:

```
[10ms] Platform: ZuluSCSI RP2040
[10ms] FW Version: 23.07.01-release Jun 30 2023 16:12:07
[11ms] DIP switch settings: debug log 1, termination 1
[11ms] SCSI termination is enabled
[12ms] Flash chip size: 2048 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[24ms] SD card detected, FAT32 volume size: 30436 MB
[25ms] SD MID: 0x03, OID: 0x53 0x44
[25ms] SD Name: SD32G
[25ms] SD Date: 9/2022
[25ms] SD Serial: 0xFD9D5623
[27ms] Reading configuration from zuluscsi.ini
[27ms] Active configuration:
[27ms] -- SelectionDelay = 255
[28ms] -- EnableUnitAttention = No
[28ms] -- EnableSCSI2 = No
[28ms] -- EnableSelLatch = No
[29ms] -- MapLunsToIDs = No
[29ms] -- EnableParity = Yes
[34ms] Finding HDD images in directory /:
[35ms] -- Opening /HD0.img for id:0 lun:0
[500ms] DBG ---- Image file is contiguous, SD card sectors 2121792 to 6316095
[501ms] ---- Read prefetch enabled: 8192 bytes
[501ms] -- Opening /HD1.img for id:1 lun:0
[962ms] DBG ---- Image file is contiguous, SD card sectors 6316096 to 10510399
[963ms] ---- Read prefetch enabled: 8192 bytes
[963ms] -- Platform supports ROM drive up to 1692 kB
[964ms] ---- ROM drive image not detected
[964ms] SCSI ID:0 BlockSize:512 Type:0 Quirks:0 ImageSize:2097152kB
[965ms] SCSI ID:1 BlockSize:512 Type:0 Quirks:0 ImageSize:2097152kB
[1066ms] Initialization complete!
[13330ms] DBG BUS RESET
[13458ms] DBG BUS RESET
```

Is there any other information that I can provide that might help?

Thanks!

[image]

aperezbios, 2023-07-16:
@ajacocks interesting. given the lack of verbosity in the debug log, I almost wonder if there might be a cabling/signalling issue here. Is the 50-pin ribbon cable you're using known to work with other devices? Is it new? old?

ajacocks, 2023-07-17:
It’s a random 50 pin 1’ long 2-connector SCSI cable from my very large collection of them. I’ll give a shot with a different cable.

ajacocks, 2023-07-21:
Trying a different cable allows the 2 defined 2gb drives to be formatted, but after MS-DOS 6.22 installed (successfully) the machine just beeped repeatedly on a boot attempt. Booting from floppy allows the drive to be read, but booting does not succeed. Running ‘fdisk /mbr’ on the drive results in a drive that, instead of attempting to boot, is immediately skipped.

aperezbios, 2023-09-07:
@PetteriAimonen any thoughts on what might be going on here? There's no real activity on the SCSI bus.

PetteriAimonen, 2023-09-07:
A debug log from the new cable that got further along would be helpful.

aperezbios, 2025-02-04:
@ajacocks as this issue has been open for 570 days, with no further response from you for over 500 days, I'm going to close it out, as I'm unable to reproduce it with my own AHA-1542CP and a ZuluIDE RP2040.

closed 2025-02-04 by aperezbios