ZuluSCSI/ZuluSCSI-firmware #267: ZuluSCSI RP2040

boudap, 2023-07-23:
I recently bought a zuluscsi rp2040 adapter,
the SD card appears when I plug it into my windows PC (Win11 64bit)
but the files i save on the SD card do'nt show up on yamaha ex5r
I have it set up as one device presenting itself as hd drive on the SCSI bus.
my settings:
SD card 8gb

ex5r menu : Disk:Device Select
Device=HDO [YAMAHA_EX]
Used 0 KB Free 2047 MB
it doesn't show any MO drive options
sd card is formatted in Ex5r
The Termination switch on the EX5R SCSI card is set to 'on'.
Zululog reports:
[10ms] Platform: ZuluSCSI RP2040
[10ms] FW Version: 23.04.11-release Apr 11 2023 16:09:22
[11ms] DIP switch settings: debug log 0, termination 1
[11ms] SCSI termination is enabled
[12ms] Flash chip size: 4096 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[25ms] SD card detected, FAT64 volume size: 7587 MB
[26ms] SD MID: 0x27, OID: 0x50 0x48
[26ms] SD Name: SD08G
[26ms] SD Date: 3/2019
[27ms] SD Serial: 0xDA949345
[32ms] Config file zuluscsi.ini not found, using defaults
[35ms] Active configuration:
[40ms] -- SelectionDelay = 255
[43ms] -- EnableUnitAttention = No
[46ms] -- EnableSCSI2 = Yes
[48ms] -- EnableSelLatch = No
[51ms] -- MapLunsToIDs = No
[54ms] -- EnableParity = Yes
[555ms] Finding HDD images in directory /:
[582ms] -- Platform supports ROM drive up to 3740 kB
[582ms] ---- ROM drive image not detected
[583ms] No images found, enabling RAW fallback partition
[583ms] Limiting RAW image mapping to SD card sector count: 15564800
[584ms] ---- Read prefetch enabled: 8192 bytes
[2084ms] Initialization complete!

**how can i set up options device hdo to MO?** 

thanks in advance for any advice!

aperezbios, 2023-07-23:
@boudap  can you clarify what you mean by "how can i set up options device hdo to MO?"

Are you referring to Magneto Optical, or does MO mean something else? It's unclear.

You don't have any image files on your SD card yet. You'll need to create one, of the size you desire. If you create a text file on the SD card, "Create HD0 1024M.txt" the firmware will create an image for you.

boudap, 2023-07-23:
yes i mean Magneto Optical. When i put sound files on SD card, after putting ex5r these files don't show.
could you tell me how to create that text file?
thank you