ZuluSCSI/ZuluSCSI-firmware #270: ZuluSCSI RP4020 Still issue with Akai MPC 3000 (fw

Glammer58, 2023-07-25:
Hi everybody, 

I had internally installed the Scsi2sd in the AKAI MPC 3000 and it worked great with it. 

Now I wanted to switch to Zuluscsi (what I have already succesfully installed in the MPC 60)

But I don´t get the MPC 3000 to work with the Zuluscsi 
(Vailixi OS 3.5 installed)
When it comes to the screen where the MPC is looking for drives, it doesn´t find some. 
Now the strange thing. (The MPC is still open and I can access the Zuluscsi hardware)
When the MPC is looking for the drives and I switch on the termination switch on the Zuluscsi, the MPC finds the drives. I have then access to the files in the Disk menue and can load up everything. Also have access to all images. It works like it should. 

You would say then.. "Ok, just put on termination from the start!.."
But, that doesn´t work. 
When I set termination from the start, the Zuluscsi starts blinking and the MPC finds nothing. 

I first used a .ini file that I have from Sampler Zone when I bought a external Zuluscsi there. The configs were very specific on the single Scsi IDs , so I made my own .ini config file with less commands. (I have the same behavior with both .ini files)
I edited my new .ini file for the recommended 
InitPreDelay = 600
Also put in 
System="MPC3000"

My question is: 
Can the termination be set from the .ini file

and can this command be set with a time delay? 
(that´s actually what I did manually on the Zuluscsi card by activating the termination dipswitch while the MPC3000 was looking for the drives.)

Users in a FB Group wrote, their MPC3000 works with the Zulu, but I did not get any answer yet from them asking for their .ini parameters.

Did I miss something to add in the file? 

Thanks for help. 

Here are the parameters I set active in my .ini file. 

[SCSI]
System="MPC3000"

Debug = 0   # Same effect as DIPSW2, enables verbose log messages
SelectionDelay = 255   # Millisecond delay after selection, 255 = automatic, 0 = no delay
Dir = "ActiveDrives"   # Optionally look for image files in subdirectory

# Settings that can be needed for compatibility with some hosts
Quirks = 0   # 0: Standard, 1: Apple, 2: OMTI, 4: Xebec, 8: VMS
EnableUnitAttention = 0 # Post UNIT_ATTENTION status on power-on or SD card hotplug
EnableSCSI2 = 1 # Enable faster speeds of SCSI2
EnableSelLatch = 0 # For Philips P2000C and other devices that release SEL signal before BSY
InitPreDelay = 600  # How many milliseconds to delay before the SCSI interface is initialized
InitPostDelay = 600 # How many milliseconds to delay after the SCSI interface is initialized

aperezbios, 2023-07-25:
@Glammer58 termination can not be controlled from the ini file. It can only be controlled via the DIP switch. 

It's still not clear to my why you can't leave the DIP switch for termination left on at power up. The state of the DIP switch is ONLY read at power-up, never after power up, so changing the switch post-power-on accomplishes nothing, because the firmware only detects the state of the DIP switches once, at initial power on. 

Nowhere do you state the version of firmware on your ZuluSCSI. This is critical information, which can be answered easily by simply uploading a copy of your zululog.txt file, which also contains it.

Generally, there is no need change or set InitPreDelay or InitPostDelay unless you know what you're doing. All that should be necessary is for you to set System=MPC3000 in the [SCSI] section of your zuluscsi.ini file. No other parameters are necessary.

```
[SCSI]
System=MPC3000
```

chickeneps, 2023-07-25:
My suggestions would be:

- Look at your zululog.txt to verify you are running the latest firmware (2023.07.01).
- Just leave termination on (the DIP switch). It's a red herring.
- This is a RED board, the RP2040, right?
- Don't bother with any Facebook group, 99% of the people there don't know what they are talking about
- As a cheat, take out the SD card and power up the MPC. Then insert the SD card, then activate it with the MPC. See if that helps.
- No ID conflicts, make sure you don't have an image (ID) set to 7.
- If you got the ZuluSCSI from SamplerZone, the images are straight-MPC3000 format, not Vailixi "hybrid". Make sure you turn that off.
- Make sure your ZuluSCSI is powered.
- This is actually easier than it looks, don't worry about anything else.

chickeneps, 2023-07-26:
Did you get the ZuluSCSI from SamplerZone, with the internal cable? If so, was that recent, or months ago? If not, where did you get the internal cable? The reason I ask is that the MPC3000 requires a special cable since the internal port is wired backwards.

Glammer58, 2023-07-26:
Hi Alex,
thanks for the fast reply! :)

I had 
System="MPC3000"
I will change it to 
System=MPC3000 
and try again.... I will put on the termination switch on from the start...
Same behavier
I will set back the delays I set.. (after that try I will copy the logfile text into this post. 
I have reduced my .ini commands to

[SCSI]
System=MPC3000
Dir = "ActiveDrives" #kept it as my images are in there ;)
EnableSCSI2 = 1

(Termination dip on)

Tried, same behavier..

Here is the log text:
[10ms] Platform: ZuluSCSI RP2040
[10ms] FW Version: 23.07.01-release Jun 30 2023 16:12:07
[11ms] DIP switch settings: debug log 0, termination 1
[11ms] SCSI termination is enabled
[12ms] Flash chip size: 4096 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[20ms] SD card detected, FAT32 volume size: 29988 MB
[21ms] SD MID: 0x00, OID: 0x00 0x00
[21ms] SD Name: SDU1 
[21ms] SD Date: 3/2023
[22ms] SD Serial: 0x14A11CF7
[23ms] Pre SCSI init boot delay in millis: 600
[624ms] Reading configuration from zuluscsi.ini
[624ms] Active configuration:
[626ms] -- SelectionDelay = 255
[627ms] -- EnableUnitAttention = No
[627ms] -- EnableSCSI2 = Yes
[628ms] -- EnableSelLatch = No
[629ms] -- MapLunsToIDs = No
[630ms] -- EnableParity = Yes
[750ms] Finding HDD images in directory ActiveDrives:
[751ms] -- Opening ActiveDrives/HD60_512 Akai MPC3000.img for id:6 lun:0
[911ms] ---- Read prefetch enabled: 8192 bytes
[911ms] -- Opening ActiveDrives/HD50_512 Akai MPC3000.img for id:5 lun:0
[1073ms] ---- Read prefetch enabled: 8192 bytes
[1073ms] -- Opening ActiveDrives/HD40_512 Akai MPC3000.img for id:4 lun:0
[1235ms] ---- Read prefetch enabled: 8192 bytes
[1236ms] -- Opening ActiveDrives/HD30_512 Akai MPC3000.img for id:3 lun:0
[1397ms] ---- Read prefetch enabled: 8192 bytes
[1399ms] -- Opening ActiveDrives/HD20_512 MPC60-ID00BU2023-650mb.img for id:2 lun:0
[1469ms] ---- Read prefetch enabled: 8192 bytes
[1470ms] -- Opening ActiveDrives/HD10_512 MPC3kBackup2023-ID01.img for id:1 lun:0
[1535ms] ---- Read prefetch enabled: 8192 bytes
[1536ms] -- Opening ActiveDrives/HD00_512 MPC3kBackup2023-ID00.img for id:0 lun:0
[1606ms] ---- Read prefetch enabled: 8192 bytes
[1613ms] -- Platform supports ROM drive up to 3740 kB
[1613ms] ---- ROM drive image not detected
[1614ms] SCSI ID:0 BlockSize:512 Type:0 Quirks:0 ImageSize:665650kB
[1615ms] SCSI ID:1 BlockSize:512 Type:0 Quirks:0 ImageSize:665650kB
[1615ms] SCSI ID:2 BlockSize:512 Type:0 Quirks:0 ImageSize:665650kB
[1616ms] SCSI ID:3 BlockSize:512 Type:0 Quirks:0 ImageSize:798720kB
[1617ms] SCSI ID:4 BlockSize:512 Type:0 Quirks:0 ImageSize:798720kB
[1617ms] SCSI ID:5 BlockSize:512 Type:0 Quirks:0 ImageSize:798720kB
[1618ms] SCSI ID:6 BlockSize:512 Type:0 Quirks:0 ImageSize:798720kB
[1719ms] Initialization complete!

Glammer58, 2023-07-26:
Hi Garth, 
there you are ;) 

> My suggestions would be:
> 
>     * Look at your zululog.txt to verify you are running the latest firmware (2023.07.01).

It´s the latest (see txt log)
 
>     * Just leave termination on (the DIP switch). It's a red herring.

on from the start now

>     * This is a RED board, the RP2040, right?

right

>     * Don't bother with any Facebook group, 99% of the people there don't know what they are talking about
>     * As a cheat, take out the SD card and power up the MPC. Then insert the SD card, then activate it with the MPC. See if that helps.

That works, but I don´t want to do that all the time ;)
When I insert the card while the MPC is looking for the drives it finds the drives after about 2-3 seconds. 

>     * No ID conflicts, make sure you don't have an image (ID) set to 7.

No, IDs from 0 to 6
 
>     * If you got the ZuluSCSI from SamplerZone, the images are straight-MPC3000 format, not Vailixi "hybrid". Make sure you turn that off.

I got my backup images installed (Translator ;) ) and some of your images that had been on the external drive I got from you. (Thanks that folder with all the different formats is pretty helpfull :)  )
As soon, the MPC accesses the images everything works like it should. 

>     * Make sure your ZuluSCSI is powered.

It is via floppypower  ;)

>     * This is actually easier than it looks, don't worry about anything else.

Glammer58, 2023-07-26:
> Did you get the ZuluSCSI from SamplerZone, with the internal cable? If so, was that recent, or months ago? If not, where did you get the internal cable? The reason I ask is that the MPC3000 requires a special cable since the internal port is wired backwards.

I got a normal internal cable with a split adater I bought in Germany. 
Which lines on the internal cable did you flip/change to make it work? Can´t be the whole cable reversed. Would be great, if you could tell me :) 
The Scsi2Sd card works without a problem and with this "insert the card after the MPC is in looking-for-drives-mode" (Thanks for that trick) making it work, your modification on the cable can´t be that drastic. 

Actually it works, when it then works as I mentioned. ;D 
But it doesn´t without those tricks... 

So it seems to be a hardwareissue then.. 

But your external drive doesn´t work neither... 

What about this scsi 5v fuse on the scsi board? I have not checked it, but I guess if this isn´t working no more, nothing connected to the board will work. 
Or does it still work if you connect floppypower and still has some side effects? Like I have?

Glammer58, 2023-07-26:
If it had been able to set the termination via command in the .ini
I had said, if there could be a command to delay the termination for a few seconds after the initialization of the card, it could work without modding the cable. 
As it´s only possible to set it with the dip switches, I will need to mod the internal cable :)
I have no problem to do that :)

Thanks Alex and Garth again for the fast replies  :)

Best regards
Markus
(will be back tomorrow as it´s about 3 in the morning here ;) )

chickeneps, 2023-07-26:
The "SCSI fuse" just means, if blown, the MPC won't supply power via the SCSI bus. So since you are powering via the floppy cable, the fuse is a irrelevant issue.

Since the "cheat" works (a great sign, that means your cable is find too), nothing else is a problem. So that means the sole issue is the System=MPC3000 line, which seems to be read according to your log. Maybe the 600ms preset (invoked by the line) isn't enough for certain MPC's. Try using InitPreDelay=700, and/or fiddle with that number.

Glammer58, 2023-07-26:
> @Glammer58 termination can not be controlled from the ini file. It can only be controlled via the DIP switch.

Got it :)

> It's still not clear to my why you can't leave the DIP switch for termination left on at power up. The state of the DIP switch is ONLY read at power-up, never after power up, so changing the switch [...]

I followed Garth´s advice to insert the SD card after switching on the MPC, set termination on from start and waited to insert the SD card till the MPC is looking for the drives. Then it takes about a second and the MPC3000 is finding the drives :D . Then everything workes like it should. 

Excellent! I can life with that for the moment. But I would prefer to leave the card in the Zulu. 
 
> Generally, there is no need change or set InitPreDelay or InitPostDelay unless you know what you're doing. All that should be necessary is for you to set System=MPC3000 in the [SCSI] section of your zuluscsi.ini file. No other parameters should be necessary.
>

Would it be possible for the coders to extend the optional InitPreDelay time from 600ms to be set for about 5 seconds? Or even 6 ...
Maybe this could help. 
I just checked the time it takes the MPC3000 with Vailixi OS from switching on till it is searching for the drives. It´s about 5 seconds. 
I tried to insert the SD card earlier, but it did not work. It doesn´t work until the unit is looking for the drives.  
Then it works like it should. 

Thanks and best regards
Markus

Glammer58, 2023-07-26:
> The "SCSI fuse" just means, if blown, the MPC won't supply power via the SCSI bus. So since you are powering via the floppy cable, the fuse is a irrelevant issue.

Ah, ok, thanks. I also think the fuse is ok. 
> 
> Since the "cheat" works (a great sign, that means your cable is working too), nothing else is a problem. So that means the sole issue is the System=MPC3000 line, which seems to be read according to your log. Maybe the 600ms preset (invoked by the line) isn't enough for certain MPC's. Try commenting out the System=MPC3000 line and try using InitPreDelay=700, and/or fiddle with that number (higher numbers).

Thanks again for the tip to insert the SD card later. Great that this works. Now I can close the unit and put the screws back in ;D . 
I will try other delaytimes in the .ini file, but I think, the delay time is limited to 600 ms max. 
Problem is as I mentioned in the last comment, that it takes the MPC3000 with Vailixi OS for about 5 seconds till it reaches the screen where it searches for the drives. If the card is inserted earlier, it doesn´t work. 

Hopefully a longer delaytime will help. 

Thanks again and best regards
Markus

Glammer58, 2023-07-26:
Hey guys,

I just tried to put the delay time to 6000 ms in the .ini file .....
It works!!!!!!! :D 👯‍♂️ 

InitPreDelay = 6000

I thought the delaytime is limited, but it is not :D 

So now it works from the start. SD card is inserted on power up and when the screen comes up to look for the drives, it finds it immediately!

Thanks again, problem solved! 👍 🥇 💯

Glammer58, 2023-07-26:
> Thanks for your report, perhaps your MPC3000 requires more "choking". I don't want to speak for Alex, but could you perhaps set DEBUG=1 and comment out System and InitPreDelay, and send in the zululog.txt and/or zuluerr.txt? Perhaps that can shed light on the junk your particular MPC3000 is spitting out.

Comment out System and InitPreDelay? You mean to leave these commands out and check the log without that, or do you want a log with the config I now have?

chickeneps, 2023-07-26:
Comment those lines out temporarily, so the DEBUG will print in the log the SCSI commands that come from the MPC3000 screw up the ZuluSCSI.

Glammer58, 2023-07-26:
This is now a copy of the zululog.txt with Debug without System and Delay config set.

[10ms] Platform: ZuluSCSI RP2040
[10ms] FW Version: 23.07.01-release Jun 30 2023 16:12:07
[11ms] DIP switch settings: debug log 0, termination 1
[11ms] SCSI termination is enabled
[12ms] Flash chip size: 4096 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[20ms] SD card detected, FAT32 volume size: 29988 MB
[21ms] SD MID: 0x00, OID: 0x00 0x00
[21ms] SD Name: SDU1 
[21ms] SD Date: 3/2023
[22ms] SD Serial: 0x14A11CF7
[24ms] Reading configuration from zuluscsi.ini
[25ms] Active configuration:
[27ms] -- SelectionDelay = 255
[27ms] -- EnableUnitAttention = No
[28ms] -- EnableSCSI2 = Yes
[29ms] -- EnableSelLatch = No
[30ms] -- MapLunsToIDs = No
[31ms] -- EnableParity = Yes
[155ms] Finding HDD images in directory ActiveDrives:
[156ms] -- Opening ActiveDrives/HD60_512 Akai MPC3000.img for id:6 lun:0
[315ms] DBG ---- Image file is contiguous, SD card sectors 8817472 to 10414911
[316ms] ---- Read prefetch enabled: 8192 bytes
[316ms] -- Opening ActiveDrives/HD50_512 Akai MPC3000.img for id:5 lun:0
[477ms] DBG ---- Image file is contiguous, SD card sectors 7220032 to 8817471
[478ms] ---- Read prefetch enabled: 8192 bytes
[478ms] -- Opening ActiveDrives/HD40_512 Akai MPC3000.img for id:4 lun:0
[640ms] DBG ---- Image file is contiguous, SD card sectors 5622592 to 7220031
[641ms] ---- Read prefetch enabled: 8192 bytes
[642ms] -- Opening ActiveDrives/HD30_512 Akai MPC3000.img for id:3 lun:0
[803ms] DBG ---- Image file is contiguous, SD card sectors 4025152 to 5622591
[804ms] ---- Read prefetch enabled: 8192 bytes
[805ms] -- Opening ActiveDrives/HD20_512 MPC60-ID00BU2023-650mb.img for id:2 lun:0
[876ms] DBG ---- Image file is contiguous, SD card sectors 2693568 to 4024867
[876ms] ---- Read prefetch enabled: 8192 bytes
[877ms] -- Opening ActiveDrives/HD10_512 MPC3kBackup2023-ID01.img for id:1 lun:0
[942ms] DBG ---- Image file is contiguous, SD card sectors 30912 to 1362211
[943ms] ---- Read prefetch enabled: 8192 bytes
[944ms] -- Opening ActiveDrives/HD00_512 MPC3kBackup2023-ID00.img for id:0 lun:0
[1014ms] DBG ---- Image file is contiguous, SD card sectors 1362240 to 2693539
[1015ms] ---- Read prefetch enabled: 8192 bytes
[1021ms] -- Platform supports ROM drive up to 3740 kB
[1022ms] ---- ROM drive image not detected
[1022ms] SCSI ID:0 BlockSize:512 Type:0 Quirks:0 ImageSize:665650kB
[1023ms] SCSI ID:1 BlockSize:512 Type:0 Quirks:0 ImageSize:665650kB
[1024ms] SCSI ID:2 BlockSize:512 Type:0 Quirks:0 ImageSize:665650kB
[1024ms] SCSI ID:3 BlockSize:512 Type:0 Quirks:0 ImageSize:798720kB
[1025ms] SCSI ID:4 BlockSize:512 Type:0 Quirks:0 ImageSize:798720kB
[1026ms] SCSI ID:5 BlockSize:512 Type:0 Quirks:0 ImageSize:798720kB
[1027ms] SCSI ID:6 BlockSize:512 Type:0 Quirks:0 ImageSize:798720kB
[1127ms] Initialization complete!
[5806ms] DBG ---- SELECTION: 0
[6274ms] DBG BUS RESET
[6274ms] DBG ---- COMMAND: TestUnitReady
[6274ms] DBG ------ OUT: 0x00 0x00 0x00 0x00 0x00 0x00 
[6281ms] DBG -- BUS_FREE
[6573ms] DBG ---- SELECTION: 0
[6774ms] DBG BUS RESET
[6774ms] DBG ---- COMMAND: Unknown
[6774ms] DBG ------ OUT: 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
[6775ms] DBG -- BUS_FREE
[7073ms] DBG ---- SELECTION: 0
[7274ms] DBG BUS RESET
[7274ms] DBG ---- COMMAND: Unknown
[7274ms] DBG ------ OUT: 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
[7275ms] DBG -- BUS_FREE
[7574ms] DBG ---- SELECTION: 0
[7774ms] DBG BUS RESET
[... 123 lines cut ...]
[20074ms] DBG ---- SELECTION: 0
[20274ms] DBG BUS RESET
[20274ms] DBG ---- COMMAND: Unknown
[20274ms] DBG ------ OUT: 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
[20275ms] DBG -- BUS_FREE
[20574ms] DBG ---- SELECTION: 0
[20774ms] DBG BUS RESET
[20774ms] DBG ---- COMMAND: Unknown
[20774ms] DBG ------ OUT: 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
[20775ms] DBG -- BUS_FREE
[21074ms] DBG ---- SELECTION: 0
[21274ms] DBG BUS RESET
[21274ms] DBG ---- COMMAND: Unknown
[21274ms] DBG ------ OUT: 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
[21275ms] DBG -- BUS_FREE
[21574ms] DBG ---- SELECTION: 0
[21774ms] DBG BUS RESET
[21774ms] DBG ---- COMMAND: Unknown
[21774ms] DBG ------ OUT: 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
[21775ms] DBG -- BUS_FREE
[22074ms] DBG ---- SELECTION: 0
[22274ms] DBG BUS RESET
[22274ms] DBG ---- COMMAND: Unknown
[22274ms] DBG ------ OUT: 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
[22275ms] DBG -- BUS_FREE

Glammer58, 2023-07-26:
What I just now tried was to reinsert the Delay config without  the System config. 
Set Delay to 5000 ms again and it works. 
That seems to me, the Delay is more important for my MPC than the System=MPC3000 command.

Glammer58, 2023-07-26:
What also works if I change the PreDelay back to 600ms and put the PostDelay to 5000 

InitPreDelay = 600  # How many milliseconds to delay before the SCSI interface is initialized
InitPostDelay = 5000 # How many milliseconds to delay after the SCSI interface is initialized

aperezbios, 2023-07-26:
@Glammer58 all of the "OUT: 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF" commands are garbage, likely cable noise. 

Setting InitPreDelay=600 is currently the same as System=MPC3000. System=MPC3000 is simply an alias/easier way to set InitPreDelay=600. 

So, setting this:
```
InitPreDelay = 600 # How many milliseconds to delay before the SCSI interface is initialized
InitPostDelay = 5000 # How many milliseconds to delay after the SCSI interface is initialized
```
should have an identical effect to:
```
System=MPC3000 # How many milliseconds to delay before the SCSI interface is initialized
InitPostDelay = 5000 # How many milliseconds to delay after the SCSI interface is initialized
```