ZuluSCSI/ZuluSCSI-firmware #273: Feature Request: Centronics connector instead of 50 pin header

csulaguy, 2023-07-31:
Not sure if it would be a direct swap or not, but it would be really nice to have a centronics connector, instead of using a DB-25 to centronics adapter. Not sure if the pinout is the same or not, or if these connectors are still readily available from parts supply houses?

baconwaifu, 2023-08-14:
the Centronics pinout is "effectively identical" to the pinheader (the numbers are different because centronics uses U-order while headers use Z-order) but most centronics connectors are mechanically incompatible with a .1" header footprint; they use a pin-spacing of 2.14mm instead of 2.54, so would require a new board version for the centronics connector that's not interchangeable with the pinheader version.

I suspect given that the ZuluSCSI is already a low-volume part, and that yet another connector variant would be even lower volume (thus probably not worth the author's time to create yet-another-model), that you'd probably be better off just making a short (~6") 50-pin cable. They sell IDC-type centronics connectors on amazon that you can just clamp on to a normal 50-pin ribbon and should work just fine (you'll probably need a vise to seat it, and make sure you get it the right way around; pin one is the side adjacent to the berg footprint)

PetteriAimonen, 2023-08-17:
I would also suggest that a cable from the 50-pin connector to IDC centronics connector would appear most straightforward.

closed 2023-11-24 by csulaguy