ZuluSCSI/ZuluSCSI-firmware #275: Problems with Kurzweil K2500

chickeneps, 2023-08-02:
Don't know how I missed it, but it seems that Kurzweil K2500 doesn't work at all with ZuluSCSI. I tried RED, BLUE, MINI, FULL, firmwares from before 1.20 and the latest. All the same behavior. (It does work with K2000 3.54 and 3.87. This is K2500 with OS 3.02.)

- K2500 starts up
- Hit DISK, shows FLOPPY
- Increment to show SCSI 0 (there's an image for 0), screen stays unchanged for now
- ZuluSCSI LED turns on for 15sec
- LED turns off and K2500 shows the SCSI 0 screen, but says drive is "asleep"
- Hit K2500 LOAD button; ZuluSCSI LED turns on for 40sec, then screen displays "Problem mounting disk"

Log is attached.
[zululog_K2500_2023_07_01.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/12243622/zululog_K2500_2023_07_01.txt)

chickeneps, 2023-08-02:
Please pause this issue - very probable there's an issue with my K2500

aperezbios, 2023-08-03:
Please test with any other SCSI device if possible. I'm going to close this issue for now, but feel free to re-open it if you discover more information.

closed 2023-08-03 by aperezbios

chickeneps, 2023-08-03:
I did test with other devices, and they didn't work with the K2500, leading me to believe it's the K2500, not the ZuluSCSI. Plus, I checked with some clients, and one so far came forward to say it works for him. But we'll see.

PetteriAimonen, 2023-08-07:
@chickeneps Based on the log, I'd say the hardware has a short-circuit between GND and SCSI DB1 pin (pin number 4 on the 50-pin connector). This could be checked with multimeter.

chickeneps, 2023-08-08:
The K2500 is (externally) DB25. So would that be as much as plugging in a working 25-25 cable, using a multi-meter to black-to-ground and red-to-PIN 1 on the male DB end? (Furthest to my right, top row). I tried that, no connectivity, but if I touch red-to-pin 5, it connects, due to pin 5 being ground.