ZuluSCSI/ZuluSCSI-firmware #288: Constant periodic LED blinking while hooked up to Yamaha RS7000

chickeneps, 2023-08-29:
I have a client where all disk operations are perfect - reading and writing - on a Yamaha RS7000 drum machine (like the Yamaha Motif), but he reports the LED flashes - like once or twice a second - continually, regardless of disk access or not. It sounded like Active Sensing to me, but I tried it on my Yamaha Motif here and I get no blinking. My client is "concerned" and gave me this debug log.

Any thoughts?
[zululog12.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/12460387/zululog12.txt)

PetteriAimonen, 2023-08-29:
It seems to check every drive with TestUnitReady and Inquiry every half a second. I have seen some devices do that for CD drives, but not for harddrives.

There seems to be nothing wrong here. I think this may be primitive support for removable drives without checking if the drive actually claims to be removable. 

If the blinking is disturbing, we could consider an option to limit LED action to only data read/write commands.

PetteriAimonen, 2023-08-29:
The image file is not continous on the exfat filesystem of the SD card.

It does have quite significant performance impact on systems that do a lot of random access, so we do want to note it somehow in the logs. Wording could be better, though I'm not sure I can come up with a better wording :)

aperezbios, 2023-08-29:
> Digressing slightly, since we're on that topic, those warnings like this seem very unnecessary. They clutter up the log, and my client kinda freaked out seeing them.

@chickeneps They are printed for a reason. We could always add an ini parameter to suppress them, but as @PetteriAimonen points out, we need to be able to point it out, ideally with an explanation as to what can be done to correct the issue. if the user decides it's worth doing so. The choice is theirs. If you are supplying SD cards with empty disk images, you will definitely want to ensure that the FAT32/exFAT filesystem they're present on is defragmented, which will may largely avoid the situation, in your particular case.

chickeneps, 2023-08-29:
That's weird, and I'll have to test this, but to fragment a 64GB SD card exFAT, there has to be a lot of writing/deleting/writing - A LOT. I see this commonly after just putting say 7 2GB files plus a couple small ones on, without any deleting/rewriting. I cannot believe that the exFAT is fragmented in any form. And, any SCSI reading/writing NEVER expands the size of the image file, so that can't be it.

My client complained about it using a SD card I supplied him that was brand new and hardly fiddled with.

aperezbios, 2023-08-30:
> The client also said that his previous SCSI2SD v5.5 didn't do that. Why would you think SCSI2SD wouldn't do that but the ZuluSCSI would?

Despite having shared lineage, they aren't the same codebases. The SCSI2SD V6 code that we use as a library is not identical to the SCSI2SD V5 code base, as they had years to diverge

PetteriAimonen, 2023-08-30:
@chickeneps In case you are using Linux, make sure you are not using an outdated fuse-exfat package, but instead the newer exfat support in Linux kernel. The fuse package had a [bug](https://github.com/relan/exfat/pull/101) that despite being fixed years ago, is still present in many distribution packages.

I agree that on a new card, files should become continuous by default because there is so much free space.

SCSI2SD indeed does LED blinking differently. I'm not sure which choice makes more sense, the current ZuluSCSI way makes it easier to see when bus scans etc. are working, but can indeed be quite blinky if the host continuously polls.

dranch, 2023-08-30:
> @chickeneps In case you are using Linux, make sure you are not using an outdated fuse-exfat package, but instead the newer exfat support in Linux kernel. The fuse package had a [...]

Regarding the "fragmented" image issue, I worked with Alex on this.  I found that the Linux instructions at https://github.com/ZuluSCSI/ZuluSCSI-firmware to create a HDD image were not valid where the documentation was previously recommending to use the "fallocate" tool but that would not create a sparse file on exfat file systems.  The doc has been updated to recommend the use of the "truncate" tool but as @chickeneps mentioned, there is is a different bug with the exfat Linux driver when creating sparse files.  For now, a user should either just use dd to create an empty image file (non-sparse) or create new empty images on a Windows or Mac platform for now.

chickeneps, 2023-08-31:
I deleted some of my posts here in regards to ZuluSCSI logs regarding image files not being contiguous, which I initially didn't believe because (on a Mac) I simply copied files onto the SD card and unzipped them. So I thought it was a false positive on ZuluSCSI's part, but as it turns out it's the Mac (10.14 in my case) writing files in a non-contiguous way, at times pretty severely. Copying files onto the SD card was a small improvement, but that takes much more transfer time and fragmentation still exists. Whereas on PC (Win7) zero fragmentation, as you'd expect. I'm going to test on MacOS 13 next to see if it was just a oversight on Apples part, as 10.14 is comparatively and relatively old.

Regardless, this is good information for users since many are Mac users (at least in my market) and unknowing that working with SD cards there may be a lot of fragmentation that are unaware of.

PetteriAimonen, 2023-08-31:
I guess *writing* large files to exfat is not a very important usecase for many platforms, because often you use computer to copy files of a camera memory card.

We could try to cache the block mapping of the fragmented file to reduce performance impact, but it would take considerable amount of effort to implement.

chickeneps, 2023-08-31:
Oh, no need. I believe it's up to the user to manage his disks... but overtime, ultimately this could be a good idea. Similar to what we are talking about - in the old days, the user defragged their hard drive. Now, it's done in the background so nobody worries about it. And besides, this looks like a Mac issue, and maybe not even, if it turns out it's fixed in MacOS 13.

dranch, 2023-08-31:
I won't pretend to understand how complex this fragmentation issue would be to fix but considering we're talking about virtual, non-mechanical drives, it doesn't make much sense to me of why the performance is so bad with a fragmented ZuluSCSI image.  Sure, maybe less prefetch performance with sequential reads but with my limited testing on a PC with a Adaptec 2940 controller, the performance impact can be devastating. I do think there is a major issue here.

chickeneps, 2023-08-31:
Tried writing files and decompressing them on MacOS 13 Ventura; same issues. Lots of unnecessary fragmentation, same as 10.14. Looks like Apple hasn't cared enough about the issue, or if it's even been reported.

PetteriAimonen, 2023-08-31:
@dranch The performance impact is largest for small access sizes, a few kB or smaller. Even though SD cards are solid state, they typically have 1-2 ms of delay when starting a read at a new address. For continuous files, we can just store the starting sector of the image on SD card and immediately know where the data is. For fragmented files, we rely on SdFat library to look up the actual sector, and it can result in multiple separate reads from the SD card on the background.

dranch, 2023-08-31:
I worry about any defragger will wear out common SD card's flash cells.  Maybe a different class of SD card is really required here like "endurance" class cards, industrial cards, SLC-based cards, etc.

aperezbios, 2023-09-01:
@dranch given that defragmenting a disk image is a one-time operation, and does not need to be done on a regular basis (because once the image file is defragmented, it will not re-fragment, once it's being used as a SCSI disk), I don't seen this being a significant concern, over time.

closed 2023-12-04 by aperezbios

mattpstv, 2025-03-26:
was the initial question about "continuous LED blinking when connected" solved, or was it decided to be a non-issue?   

when connected to my Yamaha rs7000, the blue LED light continuously blinks  on the ZuluSCSI (ZuluSCSI 1.1 board, currently running ZuluSCSI v1_1_plus_2025-03-19 firmware but behavior persisted with each firmware update going back to 2022). it's not something i've ever worried about as a user, but my new 18lb ex-feral cat just noticed the blue blinking light last night and seems concerned to the point of terminating it (lol he didn't even get that pun). as a temporary fix, I've covered it up with a few layers of gaffer tape which should last five years or so. just curious. (sorry to comment on a completed issue)

PetteriAimonen, 2025-03-27:
@mattpstv There is this option available in `zuluscsi.ini`:

<pre>
[SCSI]
DisableStatusLED = 1
</pre>

mattpstv, 2025-04-03:
perfect. thank you.