ZuluSCSI/ZuluSCSI-firmware #300: Write errors when using aspi4dos.sys and synchronous mode on AHA-1542CF

lnxsrt, 2023-09-16:
`[0ms] Platform: ZuluSCSI v1.1`
`[0ms] FW Version: 23.09.06-release Sep  7 2023 00:02:54`

Using a AHA-1542CF with 2.11 bios on a 386sx-40 with a single ID0 hard disk image. The AHA-1542CF device is set to FAST scsi and negotiate sync in the SCSI BIOS settings.

When I run DOS 6.22 without loading the aspi4dos.sys driver (3.36), it works perfectly fine, no issues. Negotiates the 10 MB/s SCSI-2 speed and functions wonderfully.

When I run DOS 6.22 and load the aspi4dos.sys driver, it seems to boot fine, but when I try to write to the disk, I get `General failure writing drive C`

Below is the log I captured during this error (two writes at the end), but the log looks fine to me.

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/12641651/zululog.txt)

I tried changing all sorts of settings and the one that causes the issue is when negotiate sync is enabled in the SCSI BIOS. If I disable the negotiate sync in the SCSI BIOS, I also do not get the write error in DOS when aspi4dos.sys is loaded. Not a huge issue as I only need to load aspi4dos.sys when using a CDROM image.

lnxsrt, 2023-09-16:
Okay, did some more hunting using a zuluscsi.ini.

`MaxSyncSpeed = 0` and `MaxSyncSpeed = 5` seem to work fine with aspi4dos.sys.

`MaxSyncSpeed = 10` does not work with aspi4dos.sys (but, does work just fine without aspi4dos.sys).

Not sure what this all means.

PetteriAimonen, 2023-09-18:
To me it sounds like the SCSI BIOS negotiates a synchronous mode, and the later loaded aspi4dos.sys driver doesn't work with that mode. Then when you use any method to disable synchronous mode, it will work again.

MaxSyncSpeed = 5 often almost works even if the host is actually trying to talk asynchronous mode. So I think aspi4dos may be switching the SCSI host card back to async mode, but doesn't inform ZuluSCSI that it did so.

aperezbios, 2023-09-18:
@PetteriAimonen do you have any thoughts on how we might implement a graceful failure back to async, if circumstances such as these occur? It could be something that must explicitly be enabled, at first, and only enabled by default once extensive testing has been completed.

lnxsrt, 2023-09-18:
It seems odd that the final aspi4dos.sys would be doing something that was out-of-spec. Adaptec controllers and software were around a long time with the AHA-1542CF being supported for years. Not saying it isn't possible there is a bug in the driver, just seems unlikely given how many installations there likely were in the 90s.

aperezbios, 2023-09-18:
@lnxsrt It's also possible that the ZuluSCSI code base isn't doing something that was commonly implemented/expected of rotational drives of the era. We simply don't know for sure at this point.

lnxsrt, 2023-09-19:
Just to be sure, I tested version 3.35 (1997) and version 3.34 (1995) of the aspi4dos.sys driver with the same results. I'm definitely willing to help debug this. 

Is it expected that reads would work okay if the aspi4dos.sys driver dropped the host adapter into async mode, but writes would fail?

PetteriAimonen, 2023-09-19:
The difference between synchronous and asynchronous mode is that the transmitting end doesn't wait for receiving end to acknowledge.

For reads from drive, this means that ZuluSCSI doesn't wait for host. For writes, the host doesn't wait for ZuluSCSI.

It could be that ZuluSCSI, being implemented in software instead of dedicated hardware chip, is slower than typical drives of the era. It does meet the negotiated synchronous mode specifications, though.

If you have a fast (at least 50MSps) logic analyzer or oscilloscope available, probing the REQ and ACK lines during a failing transfer could yield some insight.

lnxsrt, 2023-09-19:
I only have seeedstudio DSO Nano at the moment. No where near fast enough.

Digging around the manuals for this card I found a couple of relevant(?) sections.

[image]

This is related to the sync negotiation setting. The docs state that if I leave this in the "No" position, the SCSI device can still initiate the negotiation of synchronous transfers. However, ZuluSCSI does not do that. It only gets into synchronous mode if the host adapter initiates the negotiation.

[image]

aspi4dos.sys has a couple of command-line switches for bus off and on times. Not sure if these would help or not.

PetteriAimonen, 2023-09-20:
I agree that ZuluSCSI does not initiate synchronous mode negotiation, it expects host to do so. It probably wouldn't make a difference which end does the negotiation, when the actual transfer is what I think is failing.

I'm not sure about this bus-on/bus-off time setting. I think it shouldn't matter except for performance, and for performance the best would be maximum bus-on time and minimum bus-off time.