ZuluSCSI/ZuluSCSI-firmware #315: ZuluSCSI RP2040 rev 2023a - how to connect CD-ROM eject switches?

csulaguy, 2023-10-19:
Also got this other new revision. I cannot find any photos or documentation of this board, and again, the pinouts are different, so I'm not sure where I can connect eject switches to this board, or again, if the features have been deprecated without notice.

tenox7, 2024-08-11:
[image]

cathoderaydude, 2024-10-29:
tenox7, thank you for your photo, helped me figure it out. Here's another pic with all three pins labeled for clarity.

For anyone happening across this thread, to save you ten minutes: yes, you can just short these with a paperclip or pair of pliers. To see if it's actually switching and to which image, hook up a micro-USB cable to the USB port and connect to the virtual serial device that appears.

[image]