ZuluSCSI/ZuluSCSI-firmware #325: 23.10.20-release: New Problem with ISO image

ufkel, 2023-10-27:
Introduced with 23.10.20, there is a new problem regarding the ISO image handling.

What happens:
A bootable ISO image no longer gets mounted. Instead, it "does not appear to be a valid Macintosh disk image". With the previous firmware, it gets mounted OK.

**23.10.20:**
[image]

**23.09.06:**
[image]

**Gut feeling:**
The new Macintosh sanity checks are mislead because it is a bootable CDROM with a driver, although the block size is >512. Just a feeling.

ufkel, 2023-10-27:
For completeness – in 2023.10.26, the problem persists:

**2023.10.26:**
[image]

ufkel, 2023-10-29:
Even with `DisableMacSanityCheck = 1`, the problem persists.

2023.10.26:
[image]

The ISO image gets wrongly configured as a disk drive, see the red arrow in the log file above and the red rectangle in the screen shot below.
[image]

With 2023.09.06, everything works OK. The ISO image gets configured as a CDROM drive.

morio, 2023-10-30:
Hi @ufkel 
Hopefully this build fixes the issue 
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/6697222030

ufkel, 2023-10-31:
Hi @morio

Yes, now it does work again - hurray!

Thank you for looking into it.

[image]

aperezbios, 2023-11-01:
The fix for this has been included in the [v2023.10.31](https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/v2023.10.31) release, made earlier today. Closing as fixed.

closed 2023-11-01 by aperezbios