ZuluSCSI/ZuluSCSI-firmware #330: Read(6) bug in Initiator mode

ZigZagJoe, 2023-11-03:
ZuluSCSI currently handles oddly sized drives >1GB incorrectly in initiator mode.

Scenario: IBM DCAS32160 - 4226725 sectors in length. As this is not evenly divisible by 512 (g_initiator_state.max_sector_per_transfer), it trys to use READ(6) rather than READ(10) to read the last 165 sectors. However, READ(6) has a maximum sector of 0x1FFFFF (21 bits), but the code is checking start_sector < 0xFFFFFF (24 bits). This results in failure to read each of these remaining 165 sectors and an incomplete image.

Verified issue is corrected when checking for start_sector < 0x1FFFFF for READ(6) path.

aperezbios, 2023-11-03:
@ZigZagJoe  thanks for the contribution. We should be able to merge this shortly.

ZigZagJoe, 2023-11-03:
Happy to help. As an aside: the Seagate SCSI reference manual I cite there states a 2GB max for READ(6), but with 21 bits for the LBA [max of 0x1FFFFF] * LBA size of 512 bytes that gives an actual max address of 1GB. So that's where that value I actually used came from.

PetteriAimonen, 2023-11-03:
Yeah, this patch looks correct. https://www.staff.uni-mainz.de/tacke/scsi/SCSI2-09.html#9.2.5 agrees on the 21 bit limit for block address.

aperezbios, 2023-11-03:
@ZigZagJoe this has been merged in to the repository, and will be incorporated in the next release. Until then, CI-built nightly/on-demand binaries can be had at https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/latest

closed 2023-11-03 by aperezbios