ZuluSCSI/ZuluSCSI-firmware #333: Misleading comments about right aligned string?

morio, 2023-11-07:
I'm try to centralize all the zuluscsi.ini settings so down the road they can be written to flash and mimic the SCSI2SD settings. 
I've come across a couple of comments about right aligned strings (for SCSI vendor and product string) that look to be unimplemented 

### Comment 1
It doesn't look like right aligned strings are enabled when quirks mode is set to Apple. I can't find any code that references "quirks" that sets rightAlignedStrings.
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/4cba22e7b06b216717d1c726d5f43ee9035f551b/zuluscsi.ini#L47

### Comment 2
It doesn't look like right aligned string are set to -1 by default
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/4cba22e7b06b216717d1c726d5f43ee9035f551b/src/ZuluSCSI_disk.h#L79-L80
It seems it is set to 0
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/4cba22e7b06b216717d1c726d5f43ee9035f551b/src/ZuluSCSI_disk.cpp#L617

Should I delete to comments or implement them?

PetteriAimonen, 2023-11-08:
Probably delete the comments.

I'm not sure what the right aligned strings was about in the first place, it was requested as a feature in early development but I think later it was discovered it wasn't the right thing to do.

morio, 2023-11-13:
Thanks for the feedback

closed 2023-11-13 by morio