ZuluSCSI/ZuluSCSI-firmware #334: K2500X -Not a DOS Disc:pc_dskinit

MistyRockandRoll, 2023-11-13:
Been trying to set up a ZuluSCSI RP2040 with my Kurzweil K2500X, just updated it to firmware 3.02 and running the scsi cable through SCSI Through, everything seems to be working until I try and load anything from it or copy anything to it, shoots me the error of "Not a DOS Disc:pc_dskinit" and then says "Problem mounting disk". Using a k2000 image as I only have a 32GB SD Card, and it can't load the 8GB K2500 Image.
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/13329970/zululog.txt)

MeTalOneOEight, 2023-11-14:
There are two images named HD2.img on the SD card.
One in the root directory and one in the directory SamplerZone.
The one in root directory is write protected and ignored.

Is that twice the same image?

chickeneps, 2023-11-14:
Misty... I assume in zuluscsi.ini you have Dir = "SamplerZone", if so, then move that HD2.img in the root to some other new folder. Then, duplicate the HD2.img in the SamplerZone folder and rename it HD3.img, then format the drive on ID 3 on your Kurzweil, and see how that works out.

aperezbios, 2023-11-25:
@MistyRockandRoll There's been no response from you for. a couple of weeks here. I'm converting this from an issue to a discussion, as there's no clear evidence this is a bug, at this point.