ZuluSCSI/ZuluSCSI-firmware #338: Support unusual sector sizes

wohali, 2023-11-24:
Symbolics Lisp machines require an unusual disk sector size: 1280 bytes.

Currently the firmware [does not support this](https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/ec9257cc9dee1240c7a1555c1ef89b5e2953bfb8/src/ZuluSCSI.cpp#L459-L467).

Can support be added for unusual sector sizes please? Otherwise, the device is not usable in these machines.

aperezbios, 2024-03-19:
@wohali here's a build for you to test whenever you have some time
[ZuluSCSI-firmware-arbitrary-block-sizes-test.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14657657/ZuluSCSI-firmware-arbitrary-block-sizes-test.zip)

wohali, 2024-03-29:
Thanks @aperezbios this looks to resolve the issue.

closed 2024-03-29 by wohali