ZuluSCSI/ZuluSCSI-firmware #340: Issue: Release ZuluSCSI firmware v2023.11.22 breaks implicit device typing

ZigZagJoe, 2023-11-25:
**Issue**: 

Devices are getting initialized with info for fixed drives due to setDefaultDriveInfo running early. Implicit device types - such as a file named NE0.bin is a network device - are only determined later. The drive info strings need to be updated with any drive type info found while scanning for images.

I specifically had this issue with network devices, seems likely the same applies to other image types (ie. CDs).

**Replication:**

- Create NE0.bin
- Boot system. 
- Find SCSI ID #0 info strings reported with the contents of DRIVEINFO_FIXED
- Dayna functionality does not work due to incorrect device info

This is only an issue with the 11-22 release - 10-31 is OK. Suspect is related to 7f4104c.

aperezbios, 2023-11-25:
@ZigZagJoe thanks, as of last night I un-flagged the v2023.11.22 release as latest, and users should continue to use the previous [v2023.10.31](https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/v2023.10.31) until then. We expeect to have the problem resolved no later than Monday morning.

morio, 2023-11-28:
Thanks for the find @ZigZagJoe 
Here is an attempt at a fix
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/7012142248
Could you see if it resolves your issue?

ZigZagJoe, 2023-11-28:
> Thanks for the find @ZigZagJoe Here is an attempt at a fix https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/7012142248 Could you see if it resolves your issue?

Yes, looks good. I just gave it a whirl and I see the expected device info strings for both CD-ROM and network device types. CD-Rom mounts, Dayna driver recognizes the network device and network functionality works too.

aperezbios, 2023-11-28:
This has been fixed with the [v2023.11.27](https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/v2023.11.27) release

closed 2023-11-28 by aperezbios