ZuluSCSI/ZuluSCSI-firmware #345: CD-ROM image loading error ZuluSCSI Pico DaynaPORT 2023-11-27

beezbo, 2023-12-01:
I am finding that there is an issue where it seems like this firmware is trying to load the bin or iso file as a disk image. If I load the same micro sd card with no changes in another Zulu board with an older firmware, the CD-ROM images load properly. Running system 7.1.1 on a 68k mac. Am I missing something?

In the log file:

```
[14565ms] -- Opening 'a_CD.bin' for id: 3
[14614ms] ---- Configuring as disk drive drive
[14615ms] ---- WARNING: This image does not appear to be a valid Macintosh disk image.
```

working version:

```
[114ms] -- Opening 'a_CD.bin' for id:3, specified in zuluscsi.ini
[169ms] ---- Read prefetch enabled: 8192 bytes
[171ms] ---- Found CD-ROM CUE sheet at a_CD.cue
[173ms] ---- Cue sheet loaded with 19 tracks
```

morio, 2023-12-01:
Hi @beezbo , could you provide your zuluscsi.ini file and the full log that isn't working?
Thanks, I have a guess as to what is going but want to make sure I'm recreating the bug.

aperezbios, 2023-12-04:
@beezbo thank you fort this report. We have investigated this, and determined that this is in fact a regression, which occurred after the release of the September 9th release. The only work-around at this point would be to revert to firmware [v2023.09.06](https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/v2023.09.06), which will work for any non-Pico ZuluSCSI RP2040 users, as well as ZuluSCSI V1.1 users. 

Unfortunately, reverting to v2023.09.06 is not an option for the Pico hardware, because the above-mentioned release was made prior to the addition of the ZuluSCSI Pico build target. For all ZuluSCSI Pico users, you will need to wait for us to commit a fix, which should happen in a few days time.

morio, 2023-12-04:
Hi @beezbo, I'm pretty sure I found the bug and have a fix, try this build:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/7093854145

aperezbios, 2023-12-04:
@beezbo for convenience, here are the two Pico-specific firmware blobs, for you to test with. The other link above includes builds for all supported ZuluSCSI platforms, and is quite large, as it includes debug ELF files as well. They're in a small zip file, which is only 632kB in size:  [ZuluSCSI_Pico_DaynaPORT_2023-12-04_703a752.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/13552561/ZuluSCSI_Pico_DaynaPORT_2023-12-04_703a752.zip)

beezbo, 2023-12-09:
@morio @aperezbios Thanks for looking into this and providing a fix!! It seems to be working fine now. Appreciate the quick support.

closed 2023-12-09 by beezbo