ZuluSCSI/ZuluSCSI-firmware #350: Watchdog not functional on Pico/Pico Dayna (may also affect RP2040)

ZigZagJoe, 2023-12-04:
The watchdog functionality has a few issues on the Pico platform. May also affect RP2040 platforms. 

There's a critical issue that causes the watchdog to not function: hardcoded alarm of 3 needs to be the variable alarm_num dynamically selected in platform_reset_watchdog()
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/83b42e5d1e6dc258a7973a8f6ef0a22ef13bb88a/lib/ZuluSCSI_platform_RP2040/ZuluSCSI_platform.cpp#L682

That at least gets the watchdog cycling every second as it ought and pumping the usb log on a crash. However, on the pico platform either of the emergency actions appear to hard lock up the RP2040. The stack trace crashes the pico in the watchdog restart firmware case; I did not test why SCSI reset was crashing things.

morio, 2023-12-22:
Here is a basic fix, works for the RP2040 target:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/commit/24a05c1d04b6ae3b95e8453019a6209ba7166810

ZigZagJoe, 2023-12-22:
That should also work for the pico as that's the exact same as what I tested to fix it on mine. 

However I think you'll want to evaluate if the emergency actions do actually work by deliberately hanging it somewhere... I was finding the pico would simply hang when either trying the SCSI reset or the device reset.

morio, 2023-12-27:
@ZigZagJoe 
I think this fixes the hanging issue
https://github.com/ZuluSCSI/ZuluSCSI-firmware/tree/fix-watchdog-timeout

This is attempt to get hard faults reported with the DaynaPORT build
https://github.com/ZuluSCSI/ZuluSCSI-firmware/tree/fix-hardfault-handler
This will report a hard fault. Currently it only works for me when using the debugger and when I pause the debugger after it hits this line
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/8777060c06e0078d88ce8dd30c246d42a3ca518f/lib/ZuluSCSI_platform_RP2040/ZuluSCSI_platform.cpp#L524

It freezes on this function doing a sleep
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/8777060c06e0078d88ce8dd30c246d42a3ca518f/lib/ZuluSCSI_platform_RP2040/ZuluSCSI_platform.cpp#L428

After pausing the debugger and then doing a continue the code writes the crash log to the `zuluerr.txt`. Otherwise it just freeze at the `SD.begin(SD_CONFIG_CRASH);`

To test this I do a divide by zero around here
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/8777060c06e0078d88ce8dd30c246d42a3ca518f/src/ZuluSCSI.cpp#L808

morio, 2024-07-03:
Resolved last December, closing issue

closed 2024-07-03 by morio