ZuluSCSI/ZuluSCSI-firmware #364: Feature request: Use USB port as remote management console

hdeller, 2024-01-19:
I'm using ZuluSCSI as hard disc replacement in historic HP-PARISC machines. It runs great.

This feature request is about adding some possibility to manually switch CD-ROM drives at runtime.
The idea:
Currently the USB port is only used to log the log file in real-time.
Instead a simple "management console" could be implemented, in which ZuluSCSI will keep to use the USB port as serial console, where people could run commands like e.g.
- eject CD1
- list CD1   -> list all iso files which could be used, prefixed by a number 1...x
- insert CD1 <number_from_list_CD1>
and so on.
There are probably more possibilities.

PoC-dev, 2026-08-27:
Isn't this already implemented?

aperezbios, 2026-09-26:
> Isn't this already implemented?

It is now, but was certainly not at the time when @hdeller requested it in January of 2024.

closed 2026-09-26 by aperezbios