ZuluSCSI/ZuluSCSI-firmware #367: Feature request: manage mounted hard drive images via USB port (similar to #364)

geneve9640, 2024-01-21:
I find myself removing the SD card periodically to change HD images and assignments.  A simple USB console feature would be a nice way to mount/dismount the hard drive images. This request is similar to idea #364 for CDs but intended for hard drive images:

mount image10_512.hda   - mount hard drive image to SCSI ID1 lun0  (based on filename style) 
or mount image.hda ID20 512 - generically named image mounted to ID2 LUN0 512 byte sector
list images - list available image files in the image folder
dismount ID1 - dismount the hard drive image associated with ID1

aperezbios, 2024-01-21:
Hard drive images are, by their very definition, non-removable, so I am not really sure that this would be something that would be particularly straightforward to achieve. Most operating systems would likely not handle a change like this with much grace. That said, if you want to take a stab at it, we're happy to consider any potential pull requests.

geneve9640, 2024-01-21:
Thank you for the consideration. :)  I should have been clear: I wasn't proposing any changes while the device is actively in use, this was meant as a way to update the associations via USB versus removing the SD card.  Editing the .ini file via USB would also get the same results. As far as I could tell, accessing the .ini isn't possible via the USB port.

aperezbios, 2024-01-22:
@geneve9640 which variant of ZuluSCSI board are you using?

geneve9640, 2024-02-04:
> @geneve9640 which variant of ZuluSCSI board are you using?

Sorry for the late response. The ZuluSCSI board rev is 2022d.  I recently received a 2023a that I have not yet put into a system.

aperezbios, 2024-02-04:
In that case, can you try flashing this RP2040 firmware to your board https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v2023.12.14-USBMassStor-BETA/ZuluSCSI_RP2040_MSC_2023-12-21_f424be9.bin

You should be able to simply copy the .bin file above to your SD card, power on your ZuluSCSI, and it should auto-update, and then remove the .bin file. The LED will flash rapidly for around one second, then the new firmware will execute.

The .uf2 version of this firmware is available at the URL below. In order to use it, you must hold down the BOOTLDR button while plugging your board in to a computer with USB, then simply copy the .uf2 file to the board.
https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v2023.12.14-USBMassStor-BETA/ZuluSCSI_RP2040_MSC_2023-12-21_f424be9.uf2

Either firmware update mechanism will accomplish the same end result. The contributor of this RP2040-specific USB Mass Storage firmware is @ZigZagJoe.