ZuluSCSI/ZuluSCSI-firmware #384: RP2040 Mini works, RP2040 does not

ufkel, 2024-02-17:
So here is a strange issue. Booting a Macintosh from the RP2040 Mini works OK all the time. Shut-down, unplug the Mini and plug in the RP2040 (non-Mini, Rev 2023a). Using the very same SD Card in the RP2040 (non-Mini), the Macintosh does not boot, but hangs all the time. 
Apparently, the (non-Mini) RP2040 seems to crash (see log file zululog_BIG).

The RP2040 is attached to the very same external SCSI DB25 cable as the RP2040 Mini, together with a DB25 to 50pin adapter.
Both ZuluSCSI devices are running the same firmware 24.02.06-release.

Here are the log files for the RP2040 Mini and the RP2040 (non-Mini)
[zululog_MINI.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14319291/zululog_MINI.txt)
[zululog_BIG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14319292/zululog_BIG.txt)

aperezbios, 2024-02-17:
@ufkel thanks for the report. Which model of Mac are you using when this happens? Knowing that would help us to attempt to reproduce the problem. We won't be able to look at this in-depth until mid-next-week, but rest assured, we'll dig in once we have the time to do so.

ufkel, 2024-02-18:
@aperezbios thanks for reaching out. I am using a Macintosh Quadra 700.

aperezbios, 2024-09-27:
@ufkel can you let us know if your ZuluSCSI RP2040 (Full Size) is still behaving this way with the latest firmware?

ufkel, 2024-10-15:
@aperezbios Yes, it still behaves the same. I will collect same logs and come back again.

ufkel, 2024-10-15:
@aperezbios With 2024.08.22-release, the ZuluSCSI RP2040 (Full Size) is still running into a WATCHDOG TIMEOUT. 
Here are the new log files for the RP2040 Mini and the RP2040 (non-Mini):
[zululog_MINI.txt](https://github.com/user-attachments/files/17374491/zululog_MINI.txt)
[zululog_BIG.txt](https://github.com/user-attachments/files/17374496/zululog_BIG.txt)

aperezbios, 2024-10-16:
@ufkel Thanks. The log file you provided in your comment yesterday still shows that your full-size ZuluSCSI RP2040 is running firmware from February, so it looks like you may not have been successful in updating the firmware on that board. Can you please try again, and send us the log file once you've updated. Please check the log file to ensure it's running the current release.

From the log file: ```[12ms] FW Version: 24.02.06-release```

ufkel, 2024-10-17:
@aperezbios Thanks. I'm not sure what you are talking about: Both log file I have provided do contain the following line
`[12ms] FW Version: 24.08.22-release Aug 22 2024 23:23:49
`

Just to be sure, here is the full-size ZuluSCSI RP2040 log file again:
[zululog_BIG2.txt](https://github.com/user-attachments/files/17407024/zululog_BIG2.txt)

ufkel, 2024-10-17:
@aperezbios Before you ask :wink: – here are both log files for the new release
`FW Version: 24.10.16-release Oct 16 2024 17:34:25`

The symptoms are the same, though. The RP2040 MINI works fine, the RP2040 full-size does not. Both test have been done with the very same SD card.

[zululog_MINI10.txt](https://github.com/user-attachments/files/17414265/zululog_MINI10.txt)
[zululog_BIG10.txt](https://github.com/user-attachments/files/17414268/zululog_BIG10.txt)

PetteriAimonen, 2024-10-18:
After analyzing the log files, there are a few common symptoms:

    SCSI DMA state: WRITE       <-- Always WRITE
    Current buffer: 0x00000200/0x00000200, next 0x00000000 bytes   <-- Always end of last buffer
    PIO Data SM: tx_fifo 8, rx_fifo 0, pc 27, instr 0x0000200A   <-- tx_fifo always full
    GPIO states: 0x3BF9EDBF   <-- Pin9 / REQ always 0 (asserted), Pin10 / ACK always 1 (deasserted)
    DMA CH A: ctrl: 0x01000011 count: 0x000000AD    <-- always hundreds or thousands of bytes remaining

Usually this means that the host believes it has already received all of the requested bytes, but ZuluSCSI has not yet sent them. ZuluSCSI is waiting the host to ACK the next byte, which it will never do.

The exact command at which the transfer fails seems to vary.

I think the only possible explanation is some signal integrity issue. The mini and full-size boards use different, though similar, ferrites on the SCSI connectors. In addition the DB25-to-50pin adapter may cause small differences in the SCSI signal loading.

Normally SCSI is robust enough that none of these differences should matter. However in this case they appear to.

If it is possible, you could try another DB25-to-50 pin adapter, or try the adapter with another SCSI device.

aperezbios, 2024-10-18:
@ufkel I agree that the most likely culprit is the DB25 to IDC50 adapter. Is this adapter something you purchased fully-assembled, or soldered together yourself?

ufkel, 2024-10-19:
> @ufkel I also suspect that the most likely culprit is the DB25 to IDC50 adapter. Is this adapter something you purchased fully-assembled, or soldered together yourself?

I bought it fully-assembled. Long time ago, from the SCSI2SD master himself :innocent:
[image]

ufkel, 2024-11-27:
Finally...
It took some time, but now I can confirm you are right. The DB25 to IDC50 adapter is responsible for the RP2040 full-size problems.
Thank you so much for pointing into the right direction. We can close this issue.

closed 2024-11-27 by ufkel

aperezbios, 2024-11-27:
@ufkel thank you for confirming your finding, and closing this. I was just reviewing tickets yesterday and almost reached out.