ZuluSCSI/ZuluSCSI-firmware #392: RP2040 not working using Amiga GVP Impact Serries II - A500-HD+

danhall18, 2024-03-08:
Hi, hope you can help / identify the following isssue!

On an Amiga 500 with an A500-HD+ side car SCSI expansion, plugging into the expansion bus. 

The RP2040 is identified by the Amiga and attempts to boot. Following boot and also during HD prep (when saving config to the drive) an error pops up: gvpscsi.device: unit 000 Unexpected Status $49/$46 

I can confirm the system works with a real spinning disk (Quantum Fireball SE 2GB). I also saw a similar issue where a BluesScsi adpter had the issue and according to the video could be fixed by adding a small delay. 

https://www.youtube.com/watch?v=RyzWCMf_OXo&t=447s

Please see attached images, zululogs + zulu config

Thanks

[image]
[image]
[image]

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14535806/zululog.txt)
[zuluscsi.ini.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14535810/zuluscsi.ini.txt)

aperezbios, 2024-03-08:
@danhall18 thanks for the report. From perspective of the ZuluSCSI/contents of the Zululog, everything is working as expected, and there are no oservable errors in the log. The ini file you shared should not be necessary at all, and because it's all commented out, it isn't actually doing anything, except for the one line that disables parity. Can you explain why you chose to disable it?

aperezbios, 2024-03-08:
@danhall18 thanks. Which version of AmigaOS are you using? 1.3?

aperezbios, 2024-03-08:
@danhall18 can you determine the version of the GVP ROM on the sidecar? It would be helpful to know if it's the latest version, or an early rev.

danhall18, 2024-03-08:
4.15

[image]

aperezbios, 2024-03-09:
@danhall18. Thanks. As it's not clear what's going on here, it would be helpful to have a copy of a debug zululog, where you boot the machine, and then attempt to scan for and configure the drive. Can you confirm that you're getting the error message right after you click "save changes to drive"

danhall18, 2024-03-10:
Hi thanks for continuing to look into this. This is really odd behavior, heres what just happened and without even trying to configure the drive using HD tool box.

Very easy to re-create with these steps:

1. Created an image using: fsutil file createnew HD0.img 2150629376
2. With no other files on the sd card I power on the Amiga.
3. Amiga hangs with a white screen and does not even reach a KS screen. 
4. I then power everything off and slide the debug dip switch on the zulu to on
5. I power on the Amiga and it no longer hangs but proceeds to try and boot
6. The error dialog box is displayed Unexpected Status $49/$46 (Screen shot attached)

I also noticed that the presence of zuluscsi.ini on the card provoked this behavior, which was why I thought disabling Parity helped. This has also made the problem even more confusing, as it makes it impossible to try different drive settings.

Thanks again and really hope this helps.

[zululog - before debug switched on.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14550017/zululog.-.before.debug.switched.on.txt)

[zululog - after debug switched on.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14550021/zululog.-.after.debug.switched.on.txt)

[image]

apolkosnik, 2024-03-10:
Do you have 68040+ on that machine?

danhall18, 2024-03-10:
No it is a stock rev 6. With only ram upgrades. 2mb in the trap door and 4mb in the GVP side car

danhall18, 2024-03-13:
Just to add a bit more info. I managed to successfully image the original (working) Quantum disk using initiator with no errors. I then booted from the image and got the same unexpected status message.

[image]

danhall18, 2024-03-13:
Hi, don't know if this is relevant for this issue. I noticed that the recent commits for Daynaport device and the RobSmithDev video:

https://youtu.be/XJIrfQ9C0J4?si=q_dLw3aKkiXLG2P8

seem to work around an issue by redirecting a command. I note the video displays the exact same unexpected status error!

aperezbios, 2024-09-06:
@danhall18 did you ever get this working? Either way, can you let us know so we can update this issue? Thanks!

danhall18, 2024-09-06:
Hi, no I ended up using a Blue Scsi in the end. I believe the issue with the Zulu is due to some hardware differences, in particular by the Blue Scsi having Bus Tranceivers. I could be wrong and there maybe other factors at play.

aperezbios, 2024-09-06:
@danhall ZuluSCSI also uses bus transceivers, and always has. It seems you have your information backwards. BlueSCSI v1 did not use them, but since the BlueSCSI v2 is a rebranded clone of ZuluSCSI RP2040, which uses 99% of the same firmware source code, which we wrote and paid for.

BlueSCSI v2 is a knockoff. T would seem that you've been sold a pack of lies. If you still have your ZuluSCSI, try the latest firmware.

closed 2024-09-06 by aperezbios

danhall18, 2024-09-06:
@aperezbios Apologies I did not intend to spread misinformation. I did say I could be wrong in my previous comment and given you have confirmed Zulu does have Bus Tranceivers there must be other factors at play. Thank you for looking into the issue, I will try the latest firmware at some point.