ZuluSCSI/ZuluSCSI-firmware #393: Initiator mode issue with corrupted log file

lausvi, 2024-03-10:
I did a dump of a 36GB SCSI drive with the initiator mode. When it finished after 4 hours the drive spun down, and I unpowered the board (ZuluSCSI Compact Homebrew) and took the SD card out. The logfile on the card appeared like this:

[image]

The disk image file appears fine (although I cannot be sure). The SD card (64GB) was just formatted (ExFAT) before the disk was dumped.

Are there any rules on when the SD card can be removed safely? I also had an issue (before the formating of the card) where the card wouldn't mount on my Mac until I ran Disk Utility and let it repair the disk.

Also I managed to overwrite another disk dump's log file by powering the board accidentally again. Would it be possible to add an option to save the dump logs into separate log files (same file name with the dumps so easily matched and non-overwritten) perhaps?

Also; in the Compact, is there an option to run the live serial debug over USB to see the progress?

aperezbios, 2024-03-10:
@lausvi Interesting, thanks for the report. please provide a non-corrupted version of your zululog, so we can see what version of firmware you're using, etc. 
USB serial console output is enabled by default, so there's nothing special you need to do to see the output other than to plug the board in to a computer via USB, once it's powered on. There's no write cache or anything like that, so once the drive imagine is done, you can safely power everything off.

lausvi, 2024-03-10:
Ok great, I'll try the serial console next time I dump a disk!

Here is a a fresh log file (same board and SD card, after a fresh format):

`[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 24.03.07-release Mar  7 2024 20:03:17
[13ms] DIP switch settings: debug log 0, termination 1
[13ms] SCSI termination is enabled
[14ms] Flash chip size: 2048 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[21ms] SD card detected, FAT64 volume size: 59608 MB
[22ms] SD MID: 0x41, OID: 0x34 0x32
[22ms] SD Name: SD64G
[22ms] SD Date: 2/2017
[22ms] SD Serial: 0x002919EA
[25ms] Config file zuluscsi.ini not found, using defaults
[28ms] Active configuration:
[29ms] -- SelectionDelay = 255
[29ms] -- EnableUnitAttention = No
[29ms] -- EnableSCSI2 = Yes
[29ms] -- EnableSelLatch = No
[30ms] -- MapLunsToIDs = No
[30ms] -- EnableParity = Yes
[46ms] Finding images in directory /:
[47ms] -- Platform supports ROM drive up to 1692 kB
[48ms] ---- ROM drive image not detected
[48ms] No images found, enabling RAW fallback partition
[50ms] ---- Limiting RAW image mapping to SD card sector count: 122142720
[51ms] ---- Configuring as disk drive drive
[51ms] ---- Read prefetch enabled: 8192 bytes
[1552ms] Initialization complete!
`

lausvi, 2024-03-11:
Hmm, I guess this could've been just a failing SD card after all! 

I tried to use the same card and a hard disk file to install AIX on my RS/6000 and the install failed and the zuluscsi.log shows several lines that sound to me like the SD card itself might be the issue:

> [261094ms] SDIO checksum error in reception: block 0 calculated 0x66D3E0469B71BF38 expected 0x4D422F7320287065
> [261095ms] SdioCard::readSector(0x00010510) failed: 6

> [303149ms] SDIO checksum error in reception: block 0 calculated 0xC4062B38DB34917F expected 0x6F74696174656420
> [303150ms] SdioCard::readSector(0x00010513) failed: 6
> [303151ms] SCSI ID 1 negotiated asynchronous mode (period 4x0 ns, offset 0 bytes)
> [303152ms] SCSI ID 1 negotiated synchronous mode 10 MB/s (period 4x25 ns, offset 15 bytes)
> [303159ms] SDIO card reports unknown write status 0xFFFFFFFF
> [303159ms] SdioCard::writeSectors(0x31322F08,...,16) failed: 8
> [303163ms] SDIO SD card error on line 255, error code 2
> [303163ms] SD card write failed: 0x02
> [303164ms] SCSI ID 1 negotiated asynchronous mode (period 4x0 ns, offset 0 bytes)
> [303165ms] SCSI ID 1 negotiated synchronous mode 10 MB/s (period 4x25 ns, offset 15 bytes)
> [303172ms] SDIO card reports unknown write status 0xFFFFFFFF
> [303172ms] SdioCard::writeSectors(0x31322F10,...,16) failed: 8
> [303176ms] SDIO SD card error on line 255, error code 2
> [303176ms] SD card write failed: 0x02

> [1115681ms] SD card write failed: 0x00
> [1115683ms] SdioCard::readSector(0x0001059A) failed: 6

aperezbios, 2024-03-11:
Thanks for the follow-up, @lausvi A failing SD card would certainly explain a few things. If you could try with a new/different SD card, and it works, that would be great, so we can close out this issue. At your convenience, of course :)

lausvi, 2024-03-16:
Borrowed another 64GB SD card and repeated the dump. This time no problems, so I presume the issue on the first attempt was indeed the failing SD card. 

The log file was 12.6MB after the 4 hour dump. The two dump files are identical in size, but the md5 hashes differ, I am going to presume the first dump is bad due the errors writing on the SD card. (There were couple of retries during the dump on both times, but they all seemed to resolve at first retry).

This time I also plugged in the USB cable and had a peek there from time to time to see the progress, very handy!

I noticed this little glitch during the dump and it shows in the log as well: after 59% the counter reset back to 0% (and the dump finished at 40%).

> [8341931ms] SCSI read succeeded, sectors done: 42949120 / 71687372 speed 2702 kB/s - 59%
> [8342028ms] SCSI read succeeded, sectors done: 42949632 / 71687372 speed 2730 kB/s - 59%
> [8342124ms] SCSI read succeeded, sectors done: 42950144 / 71687372 speed 2759 kB/s - 0%
> [8342223ms] SCSI read succeeded, sectors done: 42950656 / 71687372 speed 2674 kB/s - 0%

and at the end:

> [13839086ms] SCSI read succeeded, sectors done: 71687168 / 71687372 speed 2674 kB/s - 40%
> [13839126ms] SCSI read succeeded, sectors done: 71687372 / 71687372 speed 2678 kB/s - 40%
> [13839128ms] Finished imaging drive with id 1
> [13839128ms] Marking SCSI ID, 1, as imaged, wont ask it again.

aperezbios, 2024-03-16:
@lausvi excellent, thanks for following up, and reporting the percentage error. We'll get it fixed.

aperezbios, 2024-03-20:
c

closed 2024-03-20 by aperezbios