ZuluSCSI/ZuluSCSI-firmware #394: Zuluscsi V1.2 in X68000 XVI SCSI Model hangs

ymos168, 2024-03-12:
I have a ZS V1.2 on the latest firmware installed in a X68000 XVI. It hangs with Watchdog Timeouts. Also, for Some reason it looks like Im not able to EnableParity according to the log even tho I enable it in the .ini. Any help is much appreciated.

4GB card with single .hda image

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14571395/zululog.txt)

```
[SCSI]
# Settings that apply to all SCSI IDs

# Select a system preset to apply default settings
# Known systems: "Mac", "MacPlus", "MPC3000", "MegaSTE", "X68000"
#System="Mac"

#Debug = 0   # Same effect as DIPSW2, enables verbose log messages
#SelectionDelay = 255   # Millisecond delay after selection, 255 = automatic, 0 = no delay
#Dir = "/"   # Optionally look for image files in subdirectory
#Dir2 = "/images"  # Multiple directories can be specified Dir1...Dir9
#DisableStatusLED 1 # 0: Use status LED, 1: Disable status LED

# NOTE: PhyMode is only relevant for ZuluSCSI V1.1 at this time.
#PhyMode = 0   # 0: Best available  1: PIO  2: DMA_TIMER  3: GREENPAK_PIO   4: GREENPAK_DMA
# The PhyMode parameter has no effect on ZuluSCSI RP2040-based platforms, as there is only one PHY mode.

# Settings that can be needed for compatibility with some hosts
#Quirks = 0   # 0: Standard, 1: Apple, 2: OMTI, 4: Xebec, 8: VMS, 16: X68000, 32: EWSD
#EnableUnitAttention = 0 # Post UNIT_ATTENTION status on power-on or SD card hotplug
EnableSCSI2 = 1 # Enable faster speeds of SCSI2
#EnableSelLatch = 0 # For Philips P2000C and other devices that release SEL signal before BSY
EnableParity = 1 # Enable parity checks on platforms that support it (RP2040)
#MapLunsToIDs = 0 # For Philips P2000C simulate multiple LUNs
#MaxSyncSpeed = 10 # Set to 5 or 10 to enable synchronous SCSI mode, 0 to disable
#InitPreDelay = 0  # How many milliseconds to delay before the SCSI interface is initialized
#InitPostDelay = 0 # How many milliseconds to delay after the SCSI interface is initialized

# ROM settings
#DisableROMDrive = 1 # Disable the ROM drive if it has been loaded to flash
#ROMDriveSCSIID = 7 # Override ROM drive's SCSI ID

#Initiator settings
#InitiatorID = 7 # SCSI ID, 0-7, when the device is in initiator mode, default is 7
#InitiatorMaxRetry = 5 #  number of retries on failed reads 0-255, default is 5
#InitiatorImageHandling = 0 # 0: skip exisitng images, 1: create new image with incrementing suffix, 2: overwrite exising image

#EnableCDAudio = 0 # Enable CD audio - an external I2S DAC on the v1.2 is required

# Settings that can be specified either per-device or for all devices.

# Select a device preset to apply default settings
# Known systems: "ST32430N"
#Device = "ST32430N"
#Vendor = "QUANTUM"
#Product = "FIREBALL1"
#Version = "1.0"
#Serial = "0123456789ABCDEF"
#Type = 0     # 0: Fixed, 1: Removable, 2: Optical, 3: Floppy, 4: Mag-optical, 5: Tape, 6: Network
#VendorExtensions = 0 # Bit flags for specific extensions per device type
#  CDROM - 1: Plextor's d8h vendor command
#TypeModifier = 0  # Affects only INQUIRY response
#SectorsPerTrack = 63
#HeadsPerCylinder = 255
#RightAlignStrings = 0 # Right-align SCSI vendor / product strings
#PrefetchBytes = 8192 # Maximum number of bytes to prefetch after a read request, 0 to disable
#ReinsertCDOnInquiry = 1 # Reinsert any ejected CD-ROM image on Inquiry command
#ReinsertAfterEject = 1 # Reinsert next CD image after eject, if multiple images configured.
#EjectButton = 0 # Enable eject by button 1 or 2, or set 0 to disable
#CDAVolume = 63 # Change CD Audio default volume. Maximum 255.
#DisableMacSanityCheck = 0 # Disable sanity warnings for Mac disk drives. Default is 0 - enable checks

# SCSI DaynaPORT settings
#WiFiSSID = "Wifi SSID string"
#WiFiPassword = "WiFi Password"
#WiFiMACAddress = "01:23:45:67:89"

# Settings can be overridden for individual devices.
#[SCSI2]
#Product = "Disk with ID2"

#[SCSI5]
#Product = "CD-ROM Drive"
#Type = 2

# If IMG0..IMG9 are specified, they are cycled after each CD eject command.
#IMG0 = FirstCD.iso
#IMG1 = SecondCD.bin

# Raw sector range from SD card can be passed through
# Format is RAW:first_sector:last_sector where sector numbers can be decimal or hex.
# If end sector is beyond end of SD card, it will be adjusted automatically.
# [SCSI4]
# IMG0 = RAW:0x00000000:0xFFFFFFFF # Whole SD card
```

8GB card with multiple .hda images

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14571420/zululog.txt)

```
[SCSI]
# Settings that apply to all SCSI IDs

# Select a system preset to apply default settings
# Known systems: "Mac", "MacPlus", "MPC3000", "MegaSTE", "X68000"
#System="Mac"

#Debug = 0   # Same effect as DIPSW2, enables verbose log messages
#SelectionDelay = 255   # Millisecond delay after selection, 255 = automatic, 0 = no delay
#Dir = "/"   # Optionally look for image files in subdirectory
#Dir2 = "/images"  # Multiple directories can be specified Dir1...Dir9
#DisableStatusLED 1 # 0: Use status LED, 1: Disable status LED

# NOTE: PhyMode is only relevant for ZuluSCSI V1.1 at this time.
#PhyMode = 0   # 0: Best available  1: PIO  2: DMA_TIMER  3: GREENPAK_PIO   4: GREENPAK_DMA
# The PhyMode parameter has no effect on ZuluSCSI RP2040-based platforms, as there is only one PHY mode.

# Settings that can be needed for compatibility with some hosts
#Quirks = 0   # 0: Standard, 1: Apple, 2: OMTI, 4: Xebec, 8: VMS, 16: X68000, 32: EWSD
#EnableUnitAttention = 0 # Post UNIT_ATTENTION status on power-on or SD card hotplug
EnableSCSI2 = 1 # Enable faster speeds of SCSI2
#EnableSelLatch = 0 # For Philips P2000C and other devices that release SEL signal before BSY
EnableParity = 1 # Enable parity checks on platforms that support it (RP2040)
#MapLunsToIDs = 0 # For Philips P2000C simulate multiple LUNs
#MaxSyncSpeed = 10 # Set to 5 or 10 to enable synchronous SCSI mode, 0 to disable
#InitPreDelay = 0  # How many milliseconds to delay before the SCSI interface is initialized
#InitPostDelay = 0 # How many milliseconds to delay after the SCSI interface is initialized

# ROM settings
#DisableROMDrive = 1 # Disable the ROM drive if it has been loaded to flash
#ROMDriveSCSIID = 7 # Override ROM drive's SCSI ID

#Initiator settings
#InitiatorID = 7 # SCSI ID, 0-7, when the device is in initiator mode, default is 7
#InitiatorMaxRetry = 5 #  number of retries on failed reads 0-255, default is 5
#InitiatorImageHandling = 0 # 0: skip exisitng images, 1: create new image with incrementing suffix, 2: overwrite exising image

#EnableCDAudio = 0 # Enable CD audio - an external I2S DAC on the v1.2 is required

# Settings that can be specified either per-device or for all devices.

# Select a device preset to apply default settings
# Known systems: "ST32430N"
#Device = "ST32430N"
#Vendor = "QUANTUM"
#Product = "FIREBALL1"
#Version = "1.0"
#Serial = "0123456789ABCDEF"
#Type = 0     # 0: Fixed, 1: Removable, 2: Optical, 3: Floppy, 4: Mag-optical, 5: Tape, 6: Network
#VendorExtensions = 0 # Bit flags for specific extensions per device type
#  CDROM - 1: Plextor's d8h vendor command
#TypeModifier = 0  # Affects only INQUIRY response
#SectorsPerTrack = 63
#HeadsPerCylinder = 255
#RightAlignStrings = 0 # Right-align SCSI vendor / product strings
#PrefetchBytes = 8192 # Maximum number of bytes to prefetch after a read request, 0 to disable
#ReinsertCDOnInquiry = 1 # Reinsert any ejected CD-ROM image on Inquiry command
#ReinsertAfterEject = 1 # Reinsert next CD image after eject, if multiple images configured.
#EjectButton = 0 # Enable eject by button 1 or 2, or set 0 to disable
#CDAVolume = 63 # Change CD Audio default volume. Maximum 255.
#DisableMacSanityCheck = 0 # Disable sanity warnings for Mac disk drives. Default is 0 - enable checks

# SCSI DaynaPORT settings
#WiFiSSID = "Wifi SSID string"
#WiFiPassword = "WiFi Password"
#WiFiMACAddress = "01:23:45:67:89"

# Settings can be overridden for individual devices.
#[SCSI2]
#Product = "Disk with ID2"

#[SCSI5]
#Product = "CD-ROM Drive"
#Type = 2

# If IMG0..IMG9 are specified, they are cycled after each CD eject command.
#IMG0 = FirstCD.iso
#IMG1 = SecondCD.bin

# Raw sector range from SD card can be passed through
# Format is RAW:first_sector:last_sector where sector numbers can be decimal or hex.
# If end sector is beyond end of SD card, it will be adjusted automatically.
# [SCSI4]
# IMG0 = RAW:0x00000000:0xFFFFFFFF # Whole SD card
```

aperezbios, 2024-03-12:
@ymos168  thanks for the report, and the debug logs, without having to ask for them :) This comment is simply to acknowledge your report, and let you know we're looking in to the root cause.

aperezbios, 2024-03-12:
@ymos168 can you try setting System = X68000 in your ini file, and let us know if this changes any behavior? I suspect it will.

 Also, all of the commented out lines have no effect, so the presence of your existing ini file has no effect, and is equivalent to there being no INI file at all.

ymos168, 2024-03-13:
@aperezbios 

Test With System = X68000
4GB Panasonic SDHC
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588740/zululog.txt)
[zuluscsi.ini.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588752/zuluscsi.ini.txt)

8GB Sandisk SDHC
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588760/zululog.txt)
[zuluscsi.ini.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588761/zuluscsi.ini.txt)

8GB Sandisk Ulltra
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588785/zululog.txt)
[zuluscsi.ini.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588787/zuluscsi.ini.txt)

Test with just  EnableSCSI2 and EnableParity

4GB Panasonic SDHC
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588855/zululog.txt)
[zuluscsi.ini.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588858/zuluscsi.ini.txt)

8GB Sandisk SDHC
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588860/zululog.txt)
[zuluscsi.ini.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588862/zuluscsi.ini.txt)

8GB Sandisk Ultra
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588872/zululog.txt)
[zuluscsi.ini.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14588875/zuluscsi.ini.txt)

ymos168, 2024-03-20:
I gave up on this one and replaced it with ZuluSCSI RP2040. I still have a issue but looks like the log is producing usable data. If there's something wrong with my machine maybe the logs can help. i will close this and open a new thread for the ZuluSCSI RP2040. thanks

closed 2024-03-20 by ymos168