ZuluSCSI/ZuluSCSI-firmware #405: ZuluSCSI RP2040 vs Sharp X68000 XVI

ymos168, 2024-03-20:
I Replaced the zs v1.2 with zs RP2040. XVI still not booting from it but get better info from log. hopefully this can help determine if I need to start looking for a fault on my XVI or use different settings, card etc..  Although I have examined the cable and checked continuity to traces going to the SCSI chip there might be a fault.

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14672521/zululog.txt)
`[SCSI]
System="X68000"
EnableParity = 0`
-----
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14672540/zululog.txt)
`[SCSI]
EnableSCSI2 = 1 
EnableParity = 1`
------
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14672556/zululog.txt)
Default ini
------
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14672568/zululog.txt)
`[SCSI]

System="X68000"`
------
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14672613/zululog.txt)

`[SCSI]

EnableParity = 0`

morio, 2024-03-20:
After looking at the specs of the X68000 XVI it has an actual SCSI controller. The System = X68000 is for older X68000s that have a SASI controller so that probably won't work.

It should work with just default setting (no zuluscsi.ini).

How is the ZuluSCSI installed on the system? Internally or externally?
I found this posting possibly about the XVI

https://gamesx.com/wiki/doku.php?id=x68000:black_screen_on_xvi_hd_and_potentially_other_hd_computers

If it is connected internally and those blue resistor packs are there, you could try turning termination off, on the ZuluSCSI.

If the ZuluSCSI is connected the the external SCSI port, you could try removing those blue resistor packs and turning termination on, on the ZuluSCSI.

ymos168, 2024-03-25:
im getting err.txt from 2 different cards
8GB Sandisk SDHC
[zuluerr.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14746617/zuluerr.txt)
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14746619/zululog.txt)

4GB Panasonic (Works fine in my ACEHD with ZS 1.1)
[zuluerr.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14746634/zuluerr.txt)
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14746636/zululog.txt)

ymos168, 2024-03-25:
@morio Thank you for the reply. I have the ZS connected internally. I saw that page too and also tried with and without the blue resistors. Thank you for pointing that out. I just tried with term off on ZS and no dice. 

It’s an odd issue  because the machine works perfectly with floppies with no issues at all (cotton even plays flawless with just the 2mb ram). The only thing I can think of is a possible DMA issue(???), scsi ic issue or in between (???). I will double check the cable again just in case. I do have a replacement scsi controller ic to try but I will recap the I/0 and main board first and go from there. The PSU is new and voltages are correct. 

At this point I believe I may have some kind of hardware issue with the XVI and will update and close this if I do or don’t figure out the problem.

thank you all for your help.

Vinni8154, 2024-04-26:
Fools

aperezbios, 2024-07-16:
@ymos168 since we haven't had an update from you in over three months. Feel free to add comments later, if you have anything to add.

closed 2024-07-16 by aperezbios