ZuluSCSI/ZuluSCSI-firmware #421: Suddenly getting a Watchdog Timeout on Microchannel BusLogic BT-646s/Storage Dimensions 2FAST SDC3211F

PhantomEight, 2024-05-10:
Not quite sure what's going on, was suddenly getting weird detection of extra LUNs when booting with the second to last firmware update, so I upgraded to 24.04.01.  It worked for a bit and now it hangs on boot when detecting devices.   If I pull the SD card it detects the physical CDROM and boots no problem from a floppy.  Not sure if it's a hardware error or what.   

[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 24.04.01-release Mar 31 2024 01:01:16
[13ms] DIP switch settings: debug log 0, termination 0
[13ms] NOTE: SCSI termination is disabled
[14ms] Flash chip size: 2048 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[52ms] SD card detected, FAT64 volume size: 60874 MB
[52ms] SD MID: 0x03, OID: 0x53 0x44
[53ms] SD Name: SR64G
[53ms] SD Date: 7/2023
[53ms] SD Serial: 0x8C4635E3
[75ms] Reading configuration from zuluscsi.ini
[96ms] Active configuration:
[96ms] -- SelectionDelay = 255
[96ms] -- EnableUnitAttention = No
[96ms] -- EnableSCSI2 = Yes
[97ms] -- EnableSelLatch = No
[97ms] -- MapLunsToIDs = No
[97ms] -- EnableParity = Yes
[296ms] Finding images in directory /:
[315ms] -- Opening /HD0_Buslogic _Main.hda for id:0 lun:0
[315ms] DBG ---- Image file is contiguous, SD card sectors 67072 to 16844287
[316ms] ---- Configuring as disk drive drive
[316ms] ---- Read prefetch enabled: 8192 bytes
[334ms] -- Opening /CD6_Fallout 1.1.iso for id:6 lun:0
[335ms] DBG ---- Image file is contiguous, SD card sectors 16844288 to 18080767
[335ms] ---- Configuring as CD-ROM drive
[336ms] ---- Read prefetch enabled: 8192 bytes
[354ms] -- Opening /HD2_Buslogic_Games.hda for id:2 lun:0
[356ms] DBG ---- Image file is contiguous, SD card sectors 18081024 to 22275327
[356ms] ---- Configuring as disk drive drive
[357ms] ---- Read prefetch enabled: 8192 bytes
[365ms] -- Platform supports ROM drive up to 1692 kB
[365ms] ---- ROM drive image not detected
[365ms] SCSI ID: 0, BlockSize: 512, Type: 0, Quirks: 0, Size: 8388608kB
[366ms] SCSI ID: 2, BlockSize: 512, Type: 0, Quirks: 0, Size: 2097152kB
[367ms] SCSI ID: 6, BlockSize: 2048, Type: 2, Quirks: 0, Size: 618240kB
[486ms] Initialization complete!
[28517ms] DBG BUS RESET
[28524ms] DBG -- BUS_BUSY
[28524ms] DBG -- BUS_FREE
[31729ms] DBG ---- SELECTION: 0
[31730ms] DBG ---- MESSAGE_OUT
[46486ms] --------------
[46486ms] WATCHDOG TIMEOUT, attempting bus reset
[46487ms] Platform: ZuluSCSI RP2040
[46487ms] FW Version: 24.04.01-release Mar 31 2024 01:01:16
[46488ms] GPIO states: out 0x3BFED5FF oe 0x178F3A00 in 0x1BFAD5FF
[46488ms] scsiDev.cdb: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[46489ms] scsiDev.phase: 2
[46489ms] SCSI DMA state: IDLE
[46490ms] Current buffer: 0x00000000/0x00000000, next 0x00000000 bytes
[46491ms] SyncOffset: 0 SyncPeriod 0
[46491ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[46492ms] PIO Data SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[46492ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[46493ms] DMA CH A: ctrl: 0x00000000 count: 0x00000000
[46494ms] DMA CH B: ctrl: 0x00000000 count: 0x00000000
[46494ms] DMA CH C: ctrl: 0x00000000 count: 0x00000000
[46495ms] DMA CH D: ctrl: 0x00000000 count: 0x00000000
[46495ms] GPIO states: 0x1BFBD5FF
[46496ms] STACK 0x2003FDE8:    0x10032E74 0x1BFAD5FF 0x2003FE1C 0x20014D20
[46496ms] STACK 0x2003FDF8:    0x10032E74 0x00000000 0x2003FE68 0x00000000
[46497ms] STACK 0x2003FE08:    0x2002D894 0x0000000C 0x00000000 0x00000000
[46498ms] STACK 0x2003FE18:    0x00000000 0x00000000 0x00000000 0x2000FCC5
[46499ms] STACK 0x2003FE28:    0x00010188 0x00000010 0x00000400 0x2001D6F0
[46499ms] STACK 0x2003FE38:    0x00000000 0x000101BE 0x2001D6EC 0xFFFFFFF9
[46500ms] STACK 0x2003FE48:    0x00010188 0x1BFFD5FF 0x00000400 0xD0000000
[46501ms] STACK 0x2003FE58:    0x10020185 0x2000999D 0x2000BA38 0x21000000
[46502ms] DBG ------ OUT: 0x00 
[46502ms] DBG -- BUS_FREE

aperezbios, 2024-05-11:
@PhantomEight Interesting. what's the OS and/or platform of the machine you're using with it? Do you know what the _previous_ firmware version that you had on your ZuluSCSI previously was?

If you could try stepping back through the prior releases of firmware, it may help us narrow down where things changed.

So, if you're able, please try (in this order) rolling back to:
* https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v2024.03.07/ZuluSCSI_RP2040_2024-03-07.bin
* https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v2024.02.06/ZuluSCSI_RP2040_2024-02-06.bin
* https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v2023.12.13/ZuluSCSI_RP2040_2023-12-13.bin

PhantomEight, 2024-05-13:
@aperezbios I didn't get a chance to look at this again until late this evening and of course I boot the system and it runs fine.  Perhaps it's some kind of intermittent cable or hardware error.  

I think previous version I was running was v2024.02.06, based on date mods of the update.bin's I have in my download folder. 

When it was hanging, it was prior to passing control to the operating system.  During boot up, the card's BIOS does a detection routine and with the SD card inserted it would hang.  I was definately cold booting to try and clear what might be wrong.  Will the RP2040 ever be in a sitaution where it needs to be powered off for more than a few seconds to clear RAM?

The system is an clapped out IBM PS/2 Model 80, upgraded with a Reply TurboProcessor planar that has a Pentium Overdrive 83Mhz, 128kb L2 cache, 64MB of RAM, and the [BusLogic BT-646s (Rebranded as a Storage Dimensions 2FAST SDC3211F).](https://www.ardent-tool.com/SCSI/Buslogic_BT646.html), a FAST SCSI card capable of 10Mb/sec max theoretical which is why I picked the ZuluSCSI RP2040.   It will do about 6 to 7MB/sec and I run a mixture of DOS 6.22/WFW3.11 and Windows 95 using System Commander as a boot loader.  There's also a 3COM NIC, an IBM NIC and a ChipChat sound card... but those should matter.   The physical CD drive is a Nakamichi MJ-5.16 which shows up at 1 ID and 5 LUNs.  

It's currently hooked up like this, Card -> CD Drive -> ZuluSCSI -> Terminator 

Anyway it runs just fine now... I'll attach the log with debugging turned on.  This is me booting System Commander, loading DOS, and playing the first 3 levels of DOOM II with [FastDoom](https://github.com/viti95/FastDoom) v0.9.8

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/15288303/zululog.txt)

If it fouls up again, I'll try to rollback.

aperezbios, 2024-05-16:
@PhantomEight thanks for the update. I'm going to close this issue out, as it doesn't seem there's anything obvious that can be done from our end, at least at the moment.

closed 2024-05-16 by aperezbios

PhantomEight, 2024-05-27:
@aperezbios Haven't really gotten the bus time out on card initialization lately... but if I am not getting the initialziation errors, then I occasionally run into an issue where the images on the card are detected properly, but System Commander in DOS throws an error about the the boot sector.

[image]

This is about how the log ends...  Never really seen that stuff at the end before.  Does that have any significance?  Are those parity error.  Wondering if it's a bad SD card or the image is bad. 

```
[34773ms] DBG ------ OUT: 0x28 0x00 0x00 0x00 0x00 0x5B 
[34774ms] DBG ------ OUT: 0x00 0x00 0x01 0x00 
[34774ms] DBG ------ Read 1x512 starting at 91
[34775ms] DBG ---- DATA_IN, syncOffset 15 syncPeriod 25
[34776ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 39619
[34776ms] DBG ---- STATUS: 0 GOOD
[34779ms] DBG ---- MESSAGE_IN
[34779ms] DBG ------ IN: 0x00 
[34779ms] DBG -- BUS_FREE
[34780ms] DBG ---- SELECTION: 0
[34780ms] DBG ---- MESSAGE_OUT
[34781ms] DBG ------ OUT: 0x80 
[34781ms] DBG ---- COMMAND: Read10
[34781ms] DBG ------ OUT: 0x28 0x00 0x00 0x00 0x00 0x5C 
[34782ms] DBG ------ OUT: 0x00 0x00 0x01 0x00 
[34782ms] DBG ------ Read 1x512 starting at 92
[34783ms] DBG ---- DATA_IN, syncOffset 15 syncPeriod 25
[34783ms] DBG ------ Found 1 sectors in prefetch cache
[39785ms] scsi_accel_rp2040_stopWrite() timeout
[39785ms] SCSI DMA state: WRITE_DONE
[39785ms] Current buffer: 0x00000000/0x00000000, next 0x00000000 bytes
[39786ms] SyncOffset: 15 SyncPeriod 25
[39786ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 28, instr 0x000080A0
[39787ms] PIO Data SM: tx_fifo 0, rx_fifo 3, pc 17, instr 0x00007309
[39788ms] PIO Sync SM: tx_fifo 4, rx_fifo 0, pc 21, instr 0x0000200A
[39788ms] DMA CH A: ctrl: 0x00000011 count: 0x00000000
[39789ms] DMA CH B: ctrl: 0x01020809 count: 0x00000001
[39789ms] DMA CH C: ctrl: 0x00008805 count: 0x00000000
[39790ms] DMA CH D: ctrl: 0x01011809 count: 0xFFFFFF80
[39791ms] GPIO states: 0x3BF9EEE7
[39794ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 33418
[39794ms] DBG -- BUS_FREE
```

PhantomEight, 2024-05-27:
You know what... this has to be a hardware issue.  Taking off the side cover and touching the cable has made the issue go away.  Really really odd.  Starting to wonder if I should rid of the terminator and turn on the terminator switch on the RP2040