ZuluSCSI/ZuluSCSI-firmware #422: Wrong analysis of TOC

Kurist2010, 2024-05-12:
When you load the bin+cue corresponding to the cue sheet below, the start position and total length of track 2 will be notified with a shift of the pregap (2 seconds). The file size of image.bin is 605908128 bytes.

image.cue
```
FILE "image.bin" BINARY
  TRACK 01 AUDIO
    INDEX 01 00:00:00
  TRACK 02 MODE1/2352
    PREGAP 00:02:00
    INDEX 01 01:06:19
```

For CD-ROM drives (this is correct)
```
Lead out - Red Book 57:18.64 - High Sierra 03EEE4       
Track 01 - Red Book 00:02.00 - High Sierra 000000 - CD/DA
Track 02 - Red Book 01:10.19 - High Sierra 0013FF - DATA
```

For v2024.04.01 (this is wrong)
```
Lead out - Red Book 57:16.63 - High Sierra 03EE4D       
Track 01 - Red Book 00:02.00 - High Sierra 000000 - CD/DA
Track 02 - Red Book 01:08.19 - High Sierra 001369 - DATA
```

morio, 2024-05-13:
@Kurist2010 the current implementation of bin/cue doesn't support pregap. What are you using to grab the the track information? It would be helpful to use to test implementation of pregap support.

Kurist2010, 2024-05-14:
If you use CDP, a free software for MS-DOS of Japan's NEC PC-9801, you can display the TOC relatively easily. It can be obtained from the following, but separate hardware is required to operate it.

[See here. This is a Japanese site.](https://www.ne.jp/asahi/kenji/pda/freedos/cdp.htm)

PetteriAimonen, 2024-09-25:
I'm looking into this, so far I have managed to reproduce the issue using a CD-R burned from the .cue.

cdrtools `readcd -fulltoc` shows for physical CD-ROM drive:

<pre>
Using dev=4,0,0.
Read  speed:  7056 kB/s (CD  40x, DVD  5x, BD  1x).
Write speed:  7056 kB/s (CD  40x, DVD  5x, BD  1x).
TOC len: 59. First Session: 1 Last Session: 1.
01 10 00 A0 00 00 00 00 01 00 00 
01 14 00 A1 00 00 00 00 02 00 00 
01 14 00 A2 00 00 00 00 39 12 40 
01 10 00 01 00 00 00 00 00 02 00 
01 14 00 02 00 00 00 00 01 0A 13 
00 00 00 00 00 00 00 00 00 00 00 
Lead out 1: 257764
</pre>

With ZuluSCSI 2024-08-22:

<pre>
Using dev=0,1,0.
TOC len: 59. First Session: 1 Last Session: 1.
resid: 1
01 10 00 A0 00 00 00 00 01 00 00 
01 14 00 A1 00 00 00 00 02 00 00 
01 14 00 A2 00 00 00 00 39 10 40 
01 10 00 01 00 00 00 00 00 02 00 
01 14 00 02 00 00 00 00 01 08 13 
00 00 00 00 00 00 00 00 00 00 00 
Lead out 1: 257614
</pre>

with the 0x0A vs. 0x08 corresponding to the error.

The data reading from the drive accordingly starts from byte offset 10178560 instead of 10485760. Data itself appears correct, just in wrong position.

PetteriAimonen, 2024-09-26:
This should work now, test build available here: https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/11049237148/artifacts/1981362565

Kurist2010, 2024-09-26:
Test builds worked as expected.

PR #459 (2024-09-26): CD-ROM: Support PREGAP in .cue files
Previously `PREGAP` specifier in .cue files was ignored.
This pull request adds support for it.

Also includes fix for incorrect image length if sector size is not divisible by 512. Previously for CD files in 2352 sector format the reported data length could be one sector too short.

Tests done:
* The .cue file provided in #422 with `readcd -fulltoc` and `cdrdao read-cd` now produces identical results as with real Lite-On CD drive and CD-R burned from the same files.
* `beos-5.0.3-professional-gobe.cue` with `cdrdao read-cd` gives identical .bin and .cue as [... cut]
closed 2024-09-26 by aperezbios