ZuluSCSI/ZuluSCSI-firmware #429: Mac Performa 460 (LCIIIish) - image creaged with truncate not recognized

cmigliorini, 2024-06-09:
I finally succeeded to create an image from my Ubuntu 22 box to be accepted by my Mac 460 and ZuluSCSI 2040, but neither of the following methods worked=
- the `truncate` method indicated in the documentation
- `truncate` + `mkfs.hfs` 
- `truncate` + `fdisk` + `mkfs.hfs` 

The only method I found working was to use [Disk Jockey Junior](https://diskjockey.onegeekarmy.eu/djjr/). Is it just me?

I'm willing to propose a PR to improve documentation.

aperezbios, 2024-06-10:
@cmigliorini thanks for the report. Real 68k/PPC Macintosh computers can not boot from any SCSI drive that does not contain a valid Apple Partition Map. The boot ROMs in Classic Macs have no awareness of DOS/MBR-style partitioning, so you would have needed to create a partition map with hfdisk, not fdisk. Hence the warning that "This image does not appear to be a valid Macintosh disk image"

I say "any SCSI drive" because Classic Macs booted from the **floppy** interface _can_ boot from a straight HFS disk/disk image, but not the other way around. When you use Disk Jockey/DJJR, it's creating the missing APM for you, which is why it works.

Does this make sense? I'm open to any suggestions for improvement of the documentation.

cmigliorini, 2024-06-11:
@aperezbios thanks so much for the detailed response. What I did exactly is
- I not only created the hard disk image: I also installed a CD image (System 7.6), and the system properly booted from the CD, which was what I wanted to achieve
- but the installation program did not recognize any hard drive

So you're telling me it would have worked if I had used hfdisk (I should have thought of that, still) to partition my `truncate`d file? I'll give it a try.

At any rate, if this is the solution it'd be of great help that the documentation reflects this. Let me know if I can help.

aperezbios, 2024-06-11:
@cmigliorini There are two separate issues/things to know here. One is that an image must contain a valid Apple Partition Map, to be able to be recognized by the ROM a classic Macintosh. 

The other is that, by default, all of the original Apple-provided disk formatting utilities will ignore any non-Apple-firmware SCSI hard drive, by design. If you set System=Mac in the zuluscsi.ini file, the ZuluSCSI firmware will report back with device/vendor strings that Apple's un-patched utility (either HD SC Setup, or Drive Setup) should be happy with, at which point it will let you initialize the drive as if it were an OEM Apple HDD. 

Alternatively, you can patch, or use a pre-patched version, of HD SC Setup or Drive Setup utilities, which removes this artificial limitation from the utility, allowing them to initialize any third-party HDD. There's a link to some more info about this at https://en.wikipedia.org/wiki/Apple_HD_SC_Setup