ZuluSCSI/ZuluSCSI-firmware #438: BlockSize setting in zuluscsi.ini doesn’t work

delan, 2024-07-19:
I went to the [manual](https://github.com/ZuluSCSI/ZuluSCSI-firmware/wiki/ZuluSCSI-Manual/ac534fef0c3928ea008edadb27bf86e61f8f2fbf) from the top of [this page](https://zuluscsi.com/), then followed the link to [this example config](https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/b03b4c3f6617ee17d0530220ee002e0e3c9d4364/zuluscsi.ini), which said…

```
# Settings that can be specified either per-device or for all devices.
#BlockSize = 0 # Set the drive's blocksize, defaults to 2048 for CDs and 512 for all other drives
```

…but when I tried that in my zuluscsi.ini…

```
[SCSI]
BlockSize = 512  # does not work
[SCSI6]
Product = "ceedee"  # works (but not shown below)
BlockSize = 512  # does not work
```

…the block size was still 2048:

```
[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 24.04.01-release Mar 31 2024 01:01:16
[13ms] DIP switch settings: debug log 1, termination 0
[13ms] NOTE: SCSI termination is disabled
[14ms] Flash chip size: 4096 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[20ms] SD card detected, FAT32 volume size: 30544 MB
[20ms] SD MID: 0x1B, OID: 0x53 0x4D
[21ms] SD Name: JB1Q5
[21ms] SD Date: 2/2023
[21ms] SD Serial: 0xB91A5735
[23ms] Reading configuration from zuluscsi.ini
[23ms] Active configuration:
[23ms] -- SelectionDelay = 255
[24ms] -- EnableUnitAttention = No
[24ms] -- EnableSCSI2 = Yes
[24ms] -- EnableSelLatch = No
[25ms] -- MapLunsToIDs = No
[25ms] -- EnableParity = Yes
[32ms] Finding images in directory /:
[33ms] -- Opening /CD6.iso for id:6 lun:0
[121ms] DBG ---- Image file is contiguous, SD card sectors 32704 to 915823
[122ms] ---- Configuring as CD-ROM drive
[122ms] ---- Read prefetch enabled: 8192 bytes
[123ms] -- Platform supports ROM drive up to 3740 kB
[123ms] ---- ROM drive image not detected
[124ms] SCSI ID: 6, BlockSize: 2048, Type: 2, Quirks: 0, Size: 441560kB
[369ms] Initialization complete!
```

[image]

Then I saw #403 and the [README](https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/b03b4c3f6617ee17d0530220ee002e0e3c9d4364/README.md), so I renamed `CD6.iso` to `CD6_512.iso` which did the trick:

```
[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 24.04.01-release Mar 31 2024 01:01:16
[13ms] DIP switch settings: debug log 1, termination 0
[13ms] NOTE: SCSI termination is disabled
[14ms] Flash chip size: 4096 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[20ms] SD card detected, FAT32 volume size: 30544 MB
[20ms] SD MID: 0x1B, OID: 0x53 0x4D
[20ms] SD Name: JB1Q5
[21ms] SD Date: 2/2023
[21ms] SD Serial: 0xB91A5735
[23ms] Reading configuration from zuluscsi.ini
[23ms] Active configuration:
[23ms] -- SelectionDelay = 255
[24ms] -- EnableUnitAttention = No
[24ms] -- EnableSCSI2 = Yes
[24ms] -- EnableSelLatch = No
[25ms] -- MapLunsToIDs = No
[25ms] -- EnableParity = Yes
[33ms] Finding images in directory /:
[33ms] -- Using custom block size, 512 from filename: CD6_512.iso
[35ms] -- Opening /CD6_512.iso for id:6 lun:0
[122ms] DBG ---- Image file is contiguous, SD card sectors 32704 to 915823
[123ms] ---- Configuring as CD-ROM drive
[123ms] ---- Read prefetch enabled: 8192 bytes
[124ms] -- Platform supports ROM drive up to 3740 kB
[124ms] ---- ROM drive image not detected
[124ms] SCSI ID: 6, BlockSize: 512, Type: 2, Quirks: 0, Size: 441560kB
[368ms] Initialization complete!
```

[image]

Anyway thanks for making ZuluSCSI, it works great on my SPARCstation 5!

aperezbios, 2024-07-19:
@delan thanks for the report, if you could try with the latest release build, that would be a helpful data point.

 You're running several revisions back from the current release:
```FW Version: 24.04.01-release Mar 31 2024 01:01:16```

Just download this firmware image and copy it to your SD card:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v2024.07.16/ZuluSCSI_RP2040_2024-07-16.bin

...and then put it back in your ZuluSCSI, and power on the unit. The ZuluSCSI LED will blink rapidly for about a second, then the .bin file will be removed, and your ZuluSCSI will re-set itself, and be running the new firmware.

delan, 2024-07-20:
I updated to 24.07.16-release and indeed the setting works now. Thanks!

```
$ ls -l /mnt/sd
total 2520208
-rwxr-xr-x 1 root root  452157440 Jul 19 23:21 CD6.iso
-rwxr-xr-x 1 root root 2128486400 Jul 19 23:52 HD1.img
-rwxr-xr-x 1 root root       1493 Jan  1  2024 zululog.txt
-rwxr-xr-x 1 root root        173 Jul 20 18:37 zuluscsi.ini
```

```
$ cat /mnt/sd/zuluscsi.ini
# SUN2.1G
# 2126929920/19/80/512
# cyl 2733 alt 2 hd 19 sec 80
# (2733+2)*19*80*512 = 2128486400
[SCSI1]
HeadsPerCylinder = 19
SectorsPerTrack = 80

[SCSI6]
BlockSize = 512
```

```
[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 24.07.16-release Jul 16 2024 23:29:35
[13ms] DIP switch settings: debug log 0, termination 0
[13ms] NOTE: SCSI termination is disabled
[14ms] Flash chip size: 4096 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[21ms] SD card detected, FAT32 volume size: 30544 MB
[21ms] SD MID: 0x1B, OID: 0x53 0x4D
[21ms] SD Name: JB1Q5
[21ms] SD Date: 2/2023
[22ms] SD Serial: 0xB91A5735
[24ms] Reading configuration from zuluscsi.ini
[26ms] Active configuration:
[26ms] -- SelectionDelay = 255
[27ms] -- EnableUnitAttention = No
[27ms] -- EnableSCSI2 = Yes
[27ms] -- EnableSelLatch = No
[27ms] -- MapLunsToIDs = No
[28ms] -- EnableParity = Yes
[42ms] Finding images in directory /:
[43ms] -- Opening /HD1.img for id:1 lun:0
[468ms] ---- Configuring as disk drive drive
[469ms] ---- Read prefetch enabled: 8192 bytes
[470ms] -- Opening /CD6.iso for id:6 lun:0
[561ms] ---- Configuring as CD-ROM drive
[561ms] ---- Read prefetch enabled: 8192 bytes
[562ms] -- Platform supports ROM drive up to 3740 kB
[563ms] ---- ROM drive image not detected
[563ms] SCSI ID: 1, BlockSize: 512, Type: 0, Quirks: 0, Size: 2078600kB
[564ms] SCSI ID: 6, BlockSize: 512, Type: 2, Quirks: 0, Size: 441560kB, Removable
[565ms] Found 1 removable device, to set an eject button see EjectButton in the, 'zuluscsi.ini', or the http://zuluscsi.com/manual
[806ms] Initialization complete!
[69570ms] Toolbox enabled = 0
```

closed 2024-07-20 by delan