ZuluSCSI/ZuluSCSI-firmware #445: Correct zuluscsi.ini settings for Mac SE?

davidfrerichs, 2024-08-04:
Hello, I have multiple known good disk images for Mac SE as verified by loading them in Mini vMac SE.

However, when I put them onto the ZuluSCSI RP2040, it will not recognize the disk images.

I get the single fast blink then the question mark icon on the Mac SE.

The only active settings in my zuluscsi.ini file are:
System="Mac"
Quirks = 2

I have tried System="MacPlus" as well, but that did not help.

Here is the log:
[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 24.05.17-release May 17 2024 16:24:20
[13ms] DIP switch settings: debug log 0, termination 1
[13ms] SCSI termination is enabled
[14ms] Flash chip size: 4096 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[32ms] SD card detected, FAT32 volume size: 7574 MB
[33ms] SD MID: 0x03, OID: 0x53 0x44
[33ms] SD Name: SA08G
[33ms] SD Date: 8/2022
[34ms] SD Serial: 0xBF0BAF43
[440ms] Reading configuration from zuluscsi.ini
[832ms] Active configuration (using system preset "Mac"):
[833ms] -- SelectionDelay = 255
[833ms] -- EnableUnitAttention = No
[833ms] -- EnableSCSI2 = Yes
[834ms] -- EnableSelLatch = No
[834ms] -- MapLunsToIDs = No
[834ms] -- EnableParity = Yes
[3246ms] Finding images in directory /:
[3508ms] -- Opening /HD0.dsk for id:0 lun:0
[3514ms] ---- Configuring as disk drive drive
[3515ms] ---- Read prefetch enabled: 8192 bytes
[3776ms] -- Opening /HD1.dsk for id:1 lun:0
[3782ms] ---- Configuring as disk drive drive
[3782ms] ---- Read prefetch enabled: 8192 bytes
[4043ms] -- Opening /HD2.dsk for id:2 lun:0
[4076ms] ---- Configuring as disk drive drive
[4076ms] ---- Read prefetch enabled: 8192 bytes
[4203ms] -- Platform supports ROM drive up to 3740 kB
[4203ms] ---- ROM drive image not detected
[4204ms] SCSI ID: 0, BlockSize: 512, Type: 0, Quirks: 2, Size: 20480kB
[4204ms] SCSI ID: 1, BlockSize: 512, Type: 0, Quirks: 2, Size: 20480kB
[4205ms] SCSI ID: 2, BlockSize: 512, Type: 0, Quirks: 2, Size: 131072kB

davidfrerichs, 2024-08-04:
Mac SE only has SCSI-1, so went back to the "MacPlus" hint as the starting point. From there, I tried all four combinations of SelectionDelay = 0 | 255, EnabledSelLatch = 0 | 1.

None of them worked. 

Booting from the Macintosh Utilities floppy (6.0.8) and running the Disk Utility shows no HD. Verbose logging shows no indication that the machine even tries to read from it. 

[image]

aperezbios, 2024-08-04:
@davidfrerichs Just because they work in mini vMac does _not_ mean that they will work with a SCSI emulator in a real machine, or written to a "real" rotational hard drive. Mini vMac does not have SCSI emulation, and therefore, the way the emulator boots images is as if they're very large floppy volumes and are then exposed to the OS via the virtualized floppy controller. For historical reasons, when booting via the true floppy interface on real Macs, no Apple Partition Map is required, but it is absolutely required when booting any block device via SCSI on any real Mac hardware.

The key feedback that the firmware is giving you, when it says "WARNING: This image does not appear to be a valid Macintosh disk image." is what lets us know there's a missing APM on the volume you're attempting to boot from. This is a hard requirement of Apple's Toolbox ROM, and cannot be modified. 

Also, System=MacPlus does not need to be used for any 68k Mac that is not the actual Mac Plus. Please use System=Mac instead. It works fine, even with the SE.

aperezbios, 2024-08-04:
@davidfrerichs please try with the images in Download #3 at https://macintoshgarden.org/apps/piscsi-rascsi

Any of those should work fine.

aperezbios, 2024-08-04:
Also, setting System="MacPlus" also enables Apple quirks mode, so you're effectively setting it twice, which won't hurt anything, but is also not necessary.

davidfrerichs, 2024-08-04:
@aperezbios Thank you for the guidance. I have downloaded the 6.0.8 bootable image and configured it as HD0, but I am still getting the Question Mark icon. 

[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 24.05.17-release May 17 2024 16:24:20
[13ms] DIP switch settings: debug log 1, termination 1
[13ms] SCSI termination is enabled
[14ms] Flash chip size: 4096 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[33ms] SD card detected, FAT32 volume size: 7574 MB
[33ms] SD MID: 0x03, OID: 0x53 0x44
[33ms] SD Name: SA08G
[33ms] SD Date: 8/2022
[34ms] SD Serial: 0xBF0BAF43
[491ms] Reading configuration from zuluscsi.ini
[932ms] Active configuration (using system preset "Mac"):
[933ms] -- SelectionDelay = 255
[933ms] -- EnableUnitAttention = No
[933ms] -- EnableSCSI2 = Yes
[934ms] -- EnableSelLatch = No
[934ms] -- MapLunsToIDs = No
[934ms] -- EnableParity = Yes
[3693ms] Finding images in directory /:
[3985ms] -- Opening /HD0.hda for id:0 lun:0
[3996ms] DBG ---- Image file is contiguous, SD card sectors 448876 to 530795
[3996ms] ---- Configuring as disk drive drive
[3997ms] DBG ---- Valid Macintosh disk image detected.
[3998ms] ---- Read prefetch enabled: 8192 bytes
[4144ms] -- Platform supports ROM drive up to 3740 kB
[4145ms] ---- ROM drive image not detected
[4145ms] SCSI ID: 0, BlockSize: 512, Type: 0, Quirks: 1, Size: 40960kB
[4280ms] Initialization complete!

davidfrerichs, 2024-08-04:
For completeness, I also tried the MacPlus hint. The Mac SE is still unable to find the [6.0.8 drive](https://macintoshgarden.org/sites/macintoshgarden.org/files/apps/RaSCSI-Boot-6.0.8.hda__1.zip).

[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 24.05.17-release May 17 2024 16:24:20
[13ms] DIP switch settings: debug log 1, termination 1
[13ms] SCSI termination is enabled
[14ms] Flash chip size: 4096 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[33ms] SD card detected, FAT32 volume size: 7574 MB
[33ms] SD MID: 0x03, OID: 0x53 0x44
[33ms] SD Name: SA08G
[34ms] SD Date: 8/2022
[34ms] SD Serial: 0xBF0BAF43
[498ms] Reading configuration from zuluscsi.ini
[948ms] Active configuration (using system preset "MacPlus"):
[949ms] -- SelectionDelay = 0
[949ms] -- EnableUnitAttention = No
[949ms] -- EnableSCSI2 = No
[950ms] -- EnableSelLatch = Yes
[950ms] -- MapLunsToIDs = No
[950ms] -- EnableParity = Yes
[3754ms] Finding images in directory /:
[4051ms] -- Opening /HD0.hda for id:0 lun:0
[4062ms] DBG ---- Image file is contiguous, SD card sectors 448876 to 530795
[4063ms] ---- Configuring as disk drive drive
[4064ms] DBG ---- Valid Macintosh disk image detected.
[4064ms] ---- Read prefetch enabled: 8192 bytes
[4213ms] -- Platform supports ROM drive up to 3740 kB
[4214ms] ---- ROM drive image not detected
[4214ms] SCSI ID: 0, BlockSize: 512, Type: 0, Quirks: 1, Size: 40960kB
[4348ms] Initialization complete!

aperezbios, 2024-08-04:
Hmm, how confident are you that the 50 pin ribbon cable is okay? Given that the ZuluSCSI is in debug mode, but isn't logging bus selects, it would seem to suggest that there may be an issue with either the cable or the 5380/53C80 SCSI controller chip itself. Something seems broken at the electrical-interface level.

davidfrerichs, 2024-08-05:
a good thought, but I don't think the cable is the issue. that ribbon cable is the same one that was working with the physical drive as recently as two days ago, so unless I have damaged it in between, that should not be the problem.

davidfrerichs, 2024-08-05:
@aperezbios OK, here is the funny thing. On a whim, I removed the BERG power connector and let the ZuluSCSI RP2040 just be powered off the SCSI. Suddenly, it worked!

closed 2024-08-05 by davidfrerichs