ZuluSCSI/ZuluSCSI-firmware #446: pleaes update button pinout for 2023 boards

tenox7, 2024-08-11:
could you update the pinout for the buttons for 2023a boards?

https://github.com/ZuluSCSI/ZuluSCSI-firmware/tree/main/docs

I think these are outdated!

thanks!

posi74de, 2025-01-04:
Did anyone found out which pins are required for 2023a rev? :-(

aperezbios, 2025-01-06:
@posi74de which physical form factor are you referring to/requesting information for?

On Rev 2023a Compact and Compact Homebrew boards, the footprint is at J305

On Rev 2023a Full-size ZuluSCSI boards, the connector at J303 has the same pinout, with more readily solderable through holes of the same signals at J307.

posi74de, 2025-01-06:
Sorry for the unclear report - I was refering to docs/ZuluSCSI_RP2040_buttons.jpg for the full-size board:
Rev 2022d on the image above has the buttons marked on J303 which is a 2x6
Rev 2023a has a J303 but it is 1x7 so I wonder how this matches? I there a schematic of the board available?

aperezbios, 2025-01-06:
@posi74de no worries. One correction: On the full-size Rev 2023a, board J303, is not 1x7, however **J305** _is_ 1x7. Those are the SDIO signals for the SD card. I agree that is rather confusing.

J305/J307 is just to the upper-left of J305, immediately to the right of the four surface-mount capacitors. J307 is what you would want to use. J305 is a surface-mount "Qwiic" connector, which is a four-pin JST connector. Earlier production runs of the full-size ZuluSCSI RP2040 did not have this part assembled, but all later ones do. J307 carries the same signals as J305, but with one extra "interrupt" pin. 

[image]
[image]

Here is the relevant portion of the schematic, taken from the Rev 2023a full-size board:

[image]