ZuluSCSI/ZuluSCSI-firmware #448: ZuluScsi 1.2 Not reliable on Amiga 3000

bit-image, 2024-08-12:
After 100+ hours messing with ZuluScsi 1.2 obtained from Rabbit Hole Computing. random watchdog timeouts persist. My memory card (32gb) was purchased from Rabbit Hole Computing at the same time. The site would have you believe that this product is 100% flawless in the Amiga platform and working but my experience is not the case. Direct mode is even less reliable than RAW mode, wish I would have purchased the cheaper 1.1 card. Is there any Amiga specific settings I should set? I have 2 3.9gb files, 3 termination resisters on the motherboard and ZuluScsi is also terminated . Here is my ini and logs. Any help would be appreciated. Thanks

[SCSI]
MaxSyncSpeed=0
EnableSCSI2=1
;PhyMode=2 # 0: Best available 1: PIO 2: DMA_TIMER
;EnableUnitAttention = 1
;EnableSelLatch = 1

[26ms] Platform: ZuluSCSI v1.1+
[26ms] FW Version: 24.07.03-release Jul  3 2024 19:00:55
[26ms] Board Version: ZuluSCSI v1.2
[27ms] DIPSW4 is ON: Enabling SCSI termination
[27ms] DIPSW3 is ON: Enabling debug messages
[28ms] DIPSW2 is ON: Disabling direct/raw mode
[28ms] DIPSW1 is OFF: Disabling Apple quirks mode by default
[42ms] SD card detected, FAT32 volume size: 30424 MB
[42ms] SD MID: 0x03, OID: 0x53 0x44
[43ms] SD Name: SD32G
[43ms] SD Date: 11/2023
[43ms] SD Serial: 0x8D159594
[45ms] Reading configuration from zuluscsi.ini
[47ms] Active configuration:
[47ms] -- SelectionDelay = 255
[48ms] -- EnableUnitAttention = No
[48ms] -- EnableSCSI2 = Yes
[48ms] -- EnableSelLatch = No
[49ms] -- MapLunsToIDs = No
[97ms] Finding images in directory /:
[98ms] -- Opening /HD1.bin for id:1 lun:0
[838ms] DBG ---- Image file is contiguous, SD card sectors 15409088 to 23601087
[839ms] ---- Configuring as disk drive drive
[839ms] ---- Read prefetch enabled: 8192 bytes
[840ms] -- Opening /HD0.bin for id:0 lun:0
[1579ms] DBG ---- Image file is contiguous, SD card sectors 23601088 to 31793087
[1580ms] ---- Configuring as disk drive drive
[1581ms] ---- Read prefetch enabled: 8192 bytes
[1582ms] SCSI ID: 0, BlockSize: 512, Type: 0, Quirks: 0, Size: 4096000kB
[1583ms] SCSI ID: 1, BlockSize: 512, Type: 0, Quirks: 0, Size: 4096000kB
[1683ms] SCSI PHY operating mode: DMA_TIMER
[1699ms] Initialization complete!
[1880ms] DBG -- BUS_BUSY
[1881ms] DBG -- BUS_FREE
[2137ms] DBG -- BUS_BUSY
[2137ms] DBG -- BUS_FREE
[2393ms] DBG -- BUS_BUSY
[2393ms] DBG -- BUS_FREE
[2649ms] DBG -- BUS_BUSY
[2650ms] DBG -- BUS_FREE
[2906ms] DBG -- BUS_BUSY
[2906ms] DBG -- BUS_FREE
[3162ms] DBG -- BUS_BUSY
[3162ms] DBG ---- SELECTION: 0
[3164ms] DBG ---- MESSAGE_OUT
[3164ms] DBG ------ OUT: 0xC0 
[3164ms] DBG ---- COMMAND: Inquiry
[3164ms] DBG ------ OUT: 0x12 0x00 0x00 0x00 0xFE 0x00 
[3165ms] Toolbox enabled = 0
[3165ms] DBG ---- DATA_IN
[3166ms] DBG ---- Total IN: 254 OUT: 0 CHECKSUM: 58090
[3166ms] DBG ---- STATUS: 0 GOOD
[3171ms] DBG ---- MESSAGE_IN
[3172ms] DBG ------ IN: 0x00 
[3172ms] DBG -- BUS_FREE
[3173ms] DBG -- BUS_BUSY

aperezbios, 2024-08-13:
@bit-image ZuluSCSI V1.1 is no longer manufactured, so buying it wouldn't be an option, though ZuluSCSI RP2040 would. If you purchsed the board

Your initial report doesn't include any information about the version of AmigaOS in use, or anything else that would help us replicate the problem you're having. Please describe your OS version. We also need you to attach a log (preferably with debug enabled) of an actual watchdog timeout event, so we can use the information to understand what may be happening.

Have you tried a different SCSI ribbon cable, or do you know the one you're using works fine with any other SCSI hard drive? A flaky ribbon cable can cause all sorts of "random" issues.

Can you also explain why you've disabled synchronous SCSI mode, by setting MaxSyncSpeed=0? Does it make things more reliable?

bit-image, 2024-08-14:
Thanks Alex for your help again. Enclosed is a log of the actual timeout event. I am able to get this working 95% of the way. Unfortunately, every now and then I receive either read errors or the scsi indicator led will stop responding and the machine locks up. If I do receive a read error, I am able to kick okay and the drive will begin working as if nothing happened. No read errors actually exist.

I am using Amiga OS3.2.2. I have tested with a 10” and a 16” cable, issue is the same. I have the onboard terminator selected on the ZuluScsi as well as a 3 terminator resistors in place on the moth
[zululog.txt](https://github.com/user-attachments/files/16606578/zululog.txt)
erboard.  I have also tested my machine’s scsi port with a Fujitsu 1.2GB hard drive and have no problems. MaxSyncSpeed=0 is set because you recommended that I set this in the config when you were working with me previously. I find that the scsi card does not work at all unless MaxSyncSpeed=0 is set.

bit-image, 2024-08-14:
[zululog.txt](https://github.com/user-attachments/files/16606582/zululog.txt)

aperezbios, 2024-08-14:
@bit-image thanks, we will review this tomorrow. Can we also get a log from you without MaxSyncSpeed set to zero? That would be most helpful.

bit-image, 2024-08-14:
[SCSI]
;MaxSyncSpeed=0
EnableSCSI2=1

bit-image, 2024-08-15:
Could I maybe simply return this and exchange it for a RP2040?

aperezbios, 2024-08-15:
@bit-image it depends on which vendor you purchased it from. If that's what you'd prefer, please e-mail us directly, with your order number.

bit-image, 2024-08-16:
I purchased it directly from you at Rabbit Hole Computing order # 12113 placed 06/04/2024 at 07:19AM. I will email you at https://store.rabbitholecomputing.com/ thanks

bit-image, 2024-08-20:
I received fantastic support from Rabbit Hole Computing today. Alex sent me a replacement RP2040 board and it is working perfectly now. Thanks very much Alex! A+++

closed 2024-08-20 by aperezbios