ZuluSCSI/ZuluSCSI-firmware #451: X68000 parity setting should be off for SASI models

Pawlicker, 2024-08-18:
So the Sharp X68000 has two different variations of the hard disk controller bus: the Super/XVI/Compact/X68030 use the SCSI bus. However the older X68000 revisions (CZ-600C/Pro/Expert/Ace) all use the SASI bus and there are some differences. Notably, not only does the zuluscsi need to be powered externally but more importantly, **parity has to be set to off.** 

When trying to set mine up I was constantly getting signal low errors from the SXSI driver, however after reading the debug log, I found out that there were parity errors. After reading that people had to turn it off for other SCSI emulators, this fixed the issue. 

This should be added in the documentation or ini file, given that parity is entirely dependent on if the system is SASI or SCSI. 
[image]
[image]

aperezbios, 2024-08-18:
Thanks for the report, @Pawlicker, I will give some thought to how we can best document this. i think an alternate System= preset would probably be the best option, but we would need to figure out how to name the two X68000 systems in a manner that would ideally not require a lot of documentation reference. Does the community (or did Sharp) have any way of referring to those first and second-generation X68000's?

Pawlicker, 2024-08-18:
> Thanks for the report, @Pawlicker, I will give some thought to how we can best document this. i think an alternate System= preset would probably be the best option, but we would need to figure out [...]

In the case of using hard disks/similar, the X68000s are grouped solely by HDD bus (SASI and SCSI). The older ones are called SASI models and later ones are SCSI models in this context.

They're grouped this way as SASI models need a few workarounds to use larger SCSI disks, and have distinct quirks and limitations that do not apply to the later models using SCSI.

aperezbios, 2024-08-18:
@Pawlicker in that case, breaking out the System presets to "X68000-SCSI" and "X6800-SASI" would probably be sufficient. Agreed?

Pawlicker, 2024-08-18:
Yep, it would be.