ZuluSCSI/ZuluSCSI-firmware #458: SD Card IO errors when imaging a large drive in initiator mode (ZuluSCSI RP2040)

agillen, 2024-09-18:
I've successfully imaged several smaller drives (2-10GB) in initiator mode, but when trying to image a larger drive (70GB Seagate Cheetah) the process consistently fails with SD card write errors. Notably, once this occurs _nothing_ - including logs - are written to the card.

I've used nearly a dozen different ExFAT-formatted SD cards - new, older, faster, slower, tested with f3, etc. - all produce the result described below (i.e. doesn't _seem_ to be an SD card issue).

First, my `zuluscsi.ini`:

<details>
  <summary>zuluscsi.ini</summary>

  ```SCSI]
   Quirks = 0
   EnableSCSI2 = 1
   MaxSyncSpeed = 0
  ```
</details>

The logs start normally, identifying the drive and beginning the transfer:

<details>
  <summary>zululog.txt (Start)</summary>

```
[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 24.08.22-release Aug 22 2024 23:23:49
[13ms] DIP switch settings: debug log 1, termination 1
[13ms] SCSI termination is enabled
[14ms] Flash chip size: 4096 kB
[14ms] SCSI initiator mode selected by DIP switch, expecting SCSI disks on the bus
[20ms] SD card detected, FAT64 volume size: 118048 MB
[20ms] SD MID: 0x27, OID: 0x50 0x48
[20ms] SD Name: MTG1b
[21ms] SD Date: 10/2022
[21ms] SD Serial: 0x0148E3F8
[296ms] InitiatorID set to ID 0x07
[801ms] Initialization complete!
[1801ms] DBG -- BUS_BUSY
[1801ms] DBG ---- SELECTION
[1802ms] DBG ------ SELECTING 0 with initiator ID 7
[2055ms] DBG ------ Target 0 did not respond
[2055ms] DBG -- BUS_FREE
[2055ms] DBG Failed to connect to SCSI ID 0
[3058ms] DBG -- BUS_BUSY
[3058ms] DBG ---- SELECTION
[3058ms] DBG ------ SELECTING 1 with initiator ID 7
[3311ms] DBG ------ Target 1 did not respond
[3312ms] DBG -- BUS_FREE
[3312ms] DBG Failed to connect to SCSI ID 1
[4314ms] DBG -- BUS_BUSY
[4314ms] DBG ---- SELECTION
[4314ms] DBG ------ SELECTING 2 with initiator ID 7
[4568ms] DBG ------ Target 2 did not respond
[4568ms] DBG -- BUS_FREE
[4568ms] DBG Failed to connect to SCSI ID 2
[5571ms] DBG -- BUS_BUSY
[5571ms] DBG ---- SELECTION
[5571ms] DBG ------ SELECTING 3 with initiator ID 7
[5824ms] DBG ------ Target 3 did not respond
[5824ms] DBG -- BUS_FREE
[5825ms] DBG Failed to connect to SCSI ID 3
[6827ms] DBG -- BUS_BUSY
[6827ms] DBG ---- SELECTION
[6827ms] DBG ------ SELECTING 4 with initiator ID 7
[6828ms] DBG ---- COMMAND: TestUnitReady
[6828ms] DBG ------ OUT: 0x00 0x00 0x00 0x00 0x00 0x00 
[6829ms] DBG ---- STATUS
[6829ms] DBG ------ IN: 0x02 
[6829ms] DBG ------ STATUS: 0x02
[6830ms] DBG ---- MESSAGE_IN
[6830ms] DBG ------ IN: 0x00 
[6830ms] DBG -- BUS_FREE
[... 244 lines cut ...]

```
[8985817ms] DBG -- BUS_BUSY
[8985817ms] DBG ---- SELECTION
[8985817ms] DBG ------ SELECTING 4 with initiator ID 7
[8985818ms] DBG ---- COMMAND: Read10
[8985818ms] DBG ------ OUT: 0x28 0x00 0x02 0x06 0xF6 0xB5 0x00 0x00 0x01 0x00 
[8985819ms] DBG ---- DATA_IN
[8985820ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 60972
[8985820ms] DBG ---- STATUS
[8985823ms] DBG Timeout waiting for response in rp2040_sdio_command_R1(16), PIO PC: 14 RXF: 0 TXF: 0
[8985823ms] SDIO SD card error on line 354, error code 2
[8985824ms] scsiInitiatorReadDataToFile: SD card write failed
[8985824ms] DBG ------ IN: 0x00 
[8985825ms] DBG ------ STATUS: 0x00
[8985825ms] DBG ---- MESSAGE_IN
[8985825ms] DBG ------ IN: 0x00 
[8985826ms] DBG -- BUS_FREE
[8985826ms] Failed to transfer 1 sectors starting at 34010805
[8985827ms] Retrying.. 1/5
```

I'd be inclined to try increasing the timeout given the error, but considering that _nothing_ is written to the SD card once this starts (not even logs) I suspect that's futile.

Any suggestions?

aperezbios, 2024-10-07:
Thanks for the report, @agillen, and apologies for the lack of response. 

Can you specify the exact model number of the Cheetah? I may have something identical or similar that we can test against on our end.

agillen, 2024-10-10:
Thanks for the reply, @aperezbios  - it's a 73GB Cheetah 10K.7 model # ST373207LC. Pulled from a working SGI O2.

Forgot to note in the original issue, but for completeness: this drive has an 80-pin SCA connector, so I'm using a generic SCA-to-50 pin adapter with the ZuluSCSI.

dranch, 2024-10-10:
If I remember correctly, SCA-attached SCSI drives are LVD (low voltage differential) signals which is NOT compatible with classic SCSI-II.

aperezbios, 2024-10-10:
> If I remember correctly, SCA-attached SCSI drives are LVD (low voltage differential) signals which is NOT compatible with classic SCSI-II.

SCA drives can still fall back to single-ended SCSI, in nearly all cases. I've never encountered an SCA drive that can't do SE.