ZuluSCSI/ZuluSCSI-firmware #471: Amiga Nexus SCSI controller finds 7 LUNs but no drives

reinauer, 2024-11-15:
```
[12ms] Platform: ZuluSCSI Pico
[12ms] FW Version: 24.10.16-release Oct 16 2024 17:34:59
[12ms] DIP switch settings: debug log 0, termination 1
[12ms] SCSI termination is enabled
[12ms] Flash chip size: 16384 kB
[12ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[33ms] SD card detected, FAT64 volume size: 118589 MB
[33ms] SD MID: 0x9F, OID: 0x54 0x49
[33ms] SD Name: SPCC 
[33ms] SD Date: 10/2024
[33ms] SD Serial: 0x440503B1
[35ms] Config file zuluscsi.ini not found, using defaults
[36ms] Active configuration:
[36ms] -- SelectionDelay = 255
[36ms] -- EnableUnitAttention = No
[36ms] -- EnableSCSI2 = Yes
[36ms] -- EnableSelLatch = No
[36ms] -- MapLunsToIDs = No
[36ms] -- EnableParity = Yes
[46ms] Finding images in directory /:
[47ms] -- Opening /hd0_AmigaOS.hda for id:0 lun:0
[47ms] ---- Configuring as disk drive drive
[47ms] ---- Read prefetch enabled: 8192 bytes
[48ms] -- Opening /cd1_amigaos32.iso for id:1 lun:0
[49ms] ---- Configuring as CD-ROM drive
[49ms] ---- Read prefetch enabled: 8192 bytes
[49ms] -- Platform supports ROM drive up to 16028 kB
[49ms] ---- ROM drive image not detected
[49ms] SCSI ID: 0, BlockSize: 512, Type: 0, Quirks: 0, Size: 7806960kB
[49ms] SCSI ID: 1, BlockSize: 2048, Type: 2, Quirks: 0, Size: 76074kB, Removable
[49ms] Found 1 removable device, to set an eject button see EjectButton in the, 'zuluscsi.ini', or the http://zuluscsi.com/manual
[157ms] Initialization complete!
[124581ms] Toolbox enabled = 0

[574529ms] DMA CH A: ctrl: 0x01020021 count: 0x00000006
[574529ms] DMA CH B: ctrl: 0xC11F9007 count: 0x00000001
[574529ms] DMA CH C: ctrl: 0x00029009 count: 0x00000000
[574529ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[574529ms] GPIO states: 0x156DFBFF
[579533ms] scsi_accel_rp2040_finishRead timeout
[579533ms] SCSI DMA state: READ
[579533ms] Current buffer: 0x00000006/0x00000006, next 0x00000000 bytes
[579533ms] SyncOffset: 0 SyncPeriod 0
[579533ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 3, instr 0x000060C8
[579533ms] PIO Data SM: tx_fifo 3, rx_fifo 1, pc 14, instr 0x0000201A
[579533ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[579533ms] DMA CH A: ctrl: 0x01020021 count: 0x00000006
[579533ms] DMA CH B: ctrl: 0xC11F9007 count: 0x00000001
[579533ms] DMA CH C: ctrl: 0x00029009 count: 0x00000000
[579533ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[579533ms] GPIO states: 0x156DFBFF
[584537ms] scsi_accel_rp2040_finishRead timeout
[584537ms] SCSI DMA state: READ
[584537ms] Current buffer: 0x00000006/0x00000006, next 0x00000000 bytes
[584537ms] SyncOffset: 0 SyncPeriod 0
[584537ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 3, instr 0x000060C8
[584537ms] PIO Data SM: tx_fifo 3, rx_fifo 1, pc 14, instr 0x0000201A
[584537ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[584537ms] DMA CH A: ctrl: 0x01020021 count: 0x00000006
[584537ms] DMA CH B: ctrl: 0xC11F9007 count: 0x00000001
[584537ms] DMA CH C: ctrl: 0x00029009 count: 0x00000000
[584537ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[584537ms] GPIO states: 0x156DFFFF
[589541ms] scsi_accel_rp2040_finishRead timeout
[589541ms] SCSI DMA state: READ
[589541ms] Current buffer: 0x00000006/0x00000006, next 0x00000000 bytes
[589541ms] SyncOffset: 0 SyncPeriod 0
[589541ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 3, instr 0x000060C8
[589541ms] PIO Data SM: tx_fifo 3, rx_fifo 1, pc 14, instr 0x0000201A
[... 190 lines cut ...]
[702634ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 3, instr 0x000060C8
[702634ms] PIO Data SM: tx_fifo 3, rx_fifo 1, pc 14, instr 0x0000201A
[702634ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[702634ms] DMA CH A: ctrl: 0x01020021 count: 0x00000006
[702634ms] DMA CH B: ctrl: 0xC11F9007 count: 0x00000001
[702634ms] DMA CH C: ctrl: 0x00029009 count: 0x00000000
[702634ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[702634ms] GPIO states: 0x156DFBFF
[707638ms] scsi_accel_rp2040_finishRead timeout
[707638ms] SCSI DMA state: READ
[707638ms] Current buffer: 0x00000006/0x00000006, next 0x00000000 bytes
[707638ms] SyncOffset: 0 SyncPeriod 0
[707638ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 3, instr 0x000060C8
[707638ms] PIO Data SM: tx_fifo 3, rx_fifo 1, pc 14, instr 0x0000201A
[707638ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[707638ms] DMA CH A: ctrl: 0x01020021 count: 0x00000006
[707638ms] DMA CH B: ctrl: 0xC11F9007 count: 0x00000001
[707638ms] DMA CH C: ctrl: 0x00029009 count: 0x00000000
[707638ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[707638ms] GPIO states: 0x156DFBFF
[747464ms] scsi_accel_rp2040_finishRead timeout
[747464ms] SCSI DMA state: READ
[747464ms] Current buffer: 0x00000006/0x00000006, next 

```

reinauer, 2024-11-15:
Interestingly, there is a zuluscsi.ini file on that SD card that says:
[SCSI]
System="Generic"
EnableSCSI2 = 0 # Enable faster speeds of SCSI2
EnableParity = 1 # Enable parity checks on platforms that support it (RP2040)
PrefetchBytes=0
[SCSI0]
SectorsPerTrack=32
HeadsPerCylinder=1

aperezbios, 2024-11-17:
@reinauer can you try this again but with no ini file, and debug enabled? Thanks.

reinauer, 2024-11-17:
Another log, this was with ini file (and this time it looks like it took)

```
[12ms] Platform: ZuluSCSI Pico
[12ms] FW Version: 24.10.16-release Oct 16 2024 17:34:59
[12ms] DIP switch settings: debug log 1, termination 1
[12ms] SCSI termination is enabled
[12ms] Flash chip size: 16384 kB
[12ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[19ms] SD card detected, FAT64 volume size: 118589 MB
[19ms] SD MID: 0x9F, OID: 0x54 0x49
[19ms] SD Name: SPCC 
[19ms] SD Date: 10/2024
[19ms] SD Serial: 0x440503B1
[21ms] Reading configuration from zuluscsi.ini
[22ms] Active configuration (using system preset "Mac"):
[22ms] -- SelectionDelay = 255
[22ms] -- EnableUnitAttention = No
[22ms] -- EnableSCSI2 = No
[22ms] -- EnableSelLatch = No
[22ms] -- MapLunsToIDs = No
[22ms] -- EnableParity = Yes
[83ms] Finding images in directory /:
[84ms] -- Opening /cd10_amigaos32.iso for id:1 lun:0
[84ms] DBG ---- Image file is contiguous, SD card sectors 15680256 to 15832403
[85ms] ---- Configuring as CD-ROM drive
[85ms] ---- Read prefetch disabled
[86ms] -- Opening /hd00_amigadisk.hda for id:0 lun:0
[87ms] ---- WARNING: file /hd00_amigadisk.hda is not contiguous. This will increase read latency.
[87ms] ---- Configuring as disk drive drive
[88ms] DBG ---- Apple magic not found.
[88ms] DBG ---- Block size not 512
[88ms] ---- WARNING: This image does not appear to be a valid Macintosh disk image.
[88ms] ---- Read prefetch disabled
[89ms] -- Platform supports ROM drive up to 16028 kB
[89ms] ---- ROM drive image not detected
[89ms] SCSI ID: 0, BlockSize: 512, Type: 0, Quirks: 1, Size: 2097152kB
[89ms] SCSI ID: 1, BlockSize: 2048, Type: 2, Quirks: 1, Size: 76074kB, Removable
[89ms] Found 1 removable device, to set an eject button see EjectButton in the, 'zuluscsi.ini', or the http://zuluscsi.com/manual
[198ms] Initialization complete!
[33494ms] DBG ---- SELECTION: 0
[33495ms] DBG ---- COMMAND: Read6
[33495ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00 
[33495ms] Toolbox enabled = 0
[33495ms] DBG ------ Read 1x512 starting at 0
[33495ms] DBG ---- DATA_IN
[33496ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 0
[33496ms] DBG ---- STATUS: 0 GOOD
[33501ms] DBG ---- MESSAGE_IN
[33501ms] DBG ------ IN: 0x00 
[33501ms] DBG -- BUS_FREE
[33501ms] DBG ---- SELECTION: 1
[33503ms] DBG ---- COMMAND: Read6
[33503ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00 
[33503ms] DBG ------ Read 1x2048 starting at 0
[33503ms] DBG ---- DATA_IN
[48198ms] --------------
[48198ms] WATCHDOG TIMEOUT, attempting bus reset
[48198ms] Platform: ZuluSCSI Pico
[48199ms] FW Version: 24.10.16-release Oct 16 2024 17:34:59
[48199ms] GPIO states: out 0x143FFCFF oe 0x085F8FFF in 0x153DFCFF
[48199ms] scsiDev.cdb: 0x08 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[48199ms] scsiDev.phase: 4
[48199ms] SCSI DMA state: WRITE
[48199ms] Current buffer: 0x00000800/0x00000800, next 0x00000000 bytes
[48199ms] SyncOffset: 0 SyncPeriod 0
[48199ms] PIO Parity SM: tx_fifo 4, rx_fifo 4, pc 31, instr 0x00004037
[48199ms] PIO Data SM: tx_fifo 8, rx_fifo 0, pc 27, instr 0x0000201A
[48199ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[48199ms] DMA CH A: ctrl: 0x01000011 count: 0x000005E8
[48199ms] DMA CH B: ctrl: 0x00020809 count: 0x00000000
[48199ms] DMA CH C: ctrl: 0x01008805 count: 0x00000001
[48199ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[48199ms] GPIO states: 0x153DF8FF
[48199ms] STACK 0x2003FEA0:    0x10030357 0x153DFCFF 0x00000000 0x00000000
[48199ms] STACK 0x2003FEB0:    0x10030357 0x00000000 0x2003FF20 0x00000000
[48199ms] STACK 0x2003FEC0:    0x20031714 0x0000000C 0x00000000 0x00000000
[48199ms] STACK 0x2003FED0:    0x00000000 0x00000000 0x00000000 0x20013EE9
[48199ms] STACK 0x2003FEE0:    0x20018B8C 0x00000010 0xD0000160 0x2003FF3C
[48199ms] STACK 0x2003FEF0:    0x00000000 0x20018B84 0x20021570 0xFFFFFFF9
[48199ms] STACK 0x2003FF00:    0x20018B8C 0x00000000 0xD0000160 0x00000000
[48199ms] STACK 0x2003FF10:    0x2001602D 0x2001605B 0x20015CC6 0x61000000
[48206ms] DBG ---- Total IN: 2048 OUT: 0 CHECKSUM: 0
[48206ms] DBG -- BUS_FREE
[48484ms] DBG ---- SELECTION: 0
[53486ms] scsi_accel_rp2040_finishRead timeout
[53486ms] SCSI DMA state: READ
[53486ms] Current buffer: 0x00000006/0x00000006, next 0x00000000 bytes
[53486ms] SyncOffset: 0 SyncPeriod 0
[53486ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 3, instr 0x000060C8
[53486ms] PIO Data SM: tx_fifo 0, rx_fifo 4, pc 16, instr 0x00005056
[53486ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[53486ms] DMA CH A: ctrl: 0x01020021 count: 0x000005E7
[53486ms] DMA CH B: ctrl: 0xC11F9007 count: 0x00000001
[53486ms] DMA CH C: ctrl: 0x00029009 count: 0x00000000
[53486ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[53486ms] GPIO states: 0x157BFAFF
[53486ms] DBG ---- COMMAND: Read6
[53486ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00 
[53494ms] DBG -- BUS_FREE
```

I have now renamed hd0_amigadisk.hda to hd00_ and that seems to have solved the issue with the device showing up on every lun but the device still shows up as "Unknown" in the UI of the Nexus tool

Will produce a log without ini file.

PetteriAimonen, 2024-11-19:
@reinauer It looks like the issue is with the CD drive. The Amiga seems to read only 520 bytes of the 2048 byte sector.

You could try naming the CD image as "CD10_512.iso". Some platforms expect CD drives that are factory-configured for 512 byte sectors, though I'm not sure if this applies to Amiga.

reinauer, 2024-12-26:
I think the  driver for this device might not support SCSI CDROMs.

closed 2024-12-26 by reinauer