ZuluSCSI/ZuluSCSI-firmware #480: 2024.12.03 prevents booting from ZuluSCSI v1.1

Jakeysp, 2024-12-07:
**Problem**
I have a ZuluSCSI v1.1, which hangs on boot after ZuluSCSIv1_1_plus_2024-12-03.bin is loaded. The system being used is an Apple Macintosh LC III.

**Symptoms**
- Upon boot with an SD card with a working image inserted, the ZuluSCSI displays a solid orange LED for ~15 seconds then goes out.
- The Mac appears to hang, i.e. no Happy Mac or missing disk floppy icon is detected.
- Reviewing zululog.txt, I can see that it locates the valid image, gets as far as setting SCSI PHY operating mode, but then doesn't get to the "Initialization complete!" message.
- If I remove the SD card and reboot, the missing disk behaviour is as expected. The Mac does not hang, and displays a missing disk floppy icon.

**Troubleshooting Attempted**
- I have a zuluscsi.ini file containing only System=Mac to set quirks, I removed it but it made no difference. Plus, it works on 2024.11.01.
- I removed my dsk file and allowed the ZuluSCSI to create its own using a text file as per your guide. No difference, still hangs.
- I rolled back to 2024.11.01, functionality was restored.
- I reattempted updating to 2024.12.03, issue returned.
- The ZuluSCSI is the only device on the SCSI chain, so nothing to do there.

I've attached the zululog from 2024.12.03, the only difference between this and one from 2024.11.01 is that the following two lines are missing from the newer firmware's output:

`[908ms] Initialization complete!`
`[9906ms] Toolbox enabled = 0`
[zululog.txt](https://github.com/user-attachments/files/18047938/zululog.txt)

aperezbios, 2024-12-07:
@Jakeysp thanks for the report. it would be helpful to have a debug log from the ZuluSCSI V1.1 when it enters this state. I will attempt to replicate on this end with an LC III and V1.1 running the same firmware.

aperezbios, 2024-12-09:
@Jakeysp I'm able to reproduce what I believe is the same behavior, and can confirm that there appears to be a regression. For now, the appropriate course of action is to downgrade to the previous release,[ v2024.11.1](https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/v2024.11.1) if you are using a ZuluSCSI V1.0 (mini only), V1.1, or V1.2. 

We will investigate and resolve this regression in the coming days, at which point we will release a V1.x-specific hotfix.

[zululog-maxSyncSpeed0.txt](https://github.com/user-attachments/files/18054752/zululog-maxSyncSpeed0.txt)
[zululog-noMaxSyncSpeed.txt](https://github.com/user-attachments/files/18054753/zululog-noMaxSyncSpeed.txt)

Jakeysp, 2024-12-09:
Thanks! I think I also updated the firmware on my V1.0 mini and couldn't replicate it. I've downgraded the V1.1 and am having no issues now.

Jakeysp, 2024-12-10:
Confirmed fixed, thanks guys!

aperezbios, 2024-12-10:
@Jakeysp I've now cut a V1.1/V1.2-specific hotfix release, which can be downloaded at https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v2024.12.09-v1.1plus/ZuluSCSIv1_1_plus_2024-12-09.bin

closed 2024-12-10 by aperezbios