ZuluSCSI/ZuluSCSI-firmware #489: Support for Emulating DVD ISO Images on ZuluSCSI RP2040

LucasLangnaese, 2024-12-20:
I am currently using the ZuluSCSI RP2040 to emulate SCSI devices for a Sun SPARCstation 20. While the device works flawlessly with CD ISO images, I am wondering if it is also possible to emulate DVD ISO images.

Specifically, I would like to know:

    Does the current firmware support the emulation of DVD ISO files?
    If not, are there any plans to implement this functionality in the future?
    Are there any technical limitations or workarounds that could enable DVD ISO emulation, particularly for use with a Sun SPARCstation 20?

I am trying to boot or access a DVD ISO on this legacy system, so any insights or suggestions would be highly appreciated.

Thank you for your great work on this project!

Best regards,
Lucas

aperezbios, 2024-12-20:
@LucasLangnaese since the beginning, ZuluSCSI has always allowed you to expose image files, including ISOs, of any arbitrary size, so this "just works" right now, in every case I've encountered to date. Whatever the size of the ISO you're attempting to use is the reported size of the volume on the SCSI side. The physical capacity limitations simply don't apply in this sort of context.

MeTalOneOEight, 2024-12-23:
Which filesystem is in the image?
A DVD usually uses UDF which maybe the firmware or operating system on the Sun machine does not know because of it's age.
Have you checked that angle?

closed 2024-12-24 by LucasLangnaese