ZuluSCSI/ZuluSCSI-firmware #492: Roland S550 ID 0 Incompatibility

chickeneps, 2024-12-28:
Using any ZuluSCSI (regular), if there is a drive assigned to ID 0, the Roland S550 (and W30) stalls upon booting and negotiating with the S550.

In contrast, when a there is no drive assigned to ID 0, the negotiation sails through perfectly.

The S550 is fixed on self-ID as 7, so it can't be worked around by assigning the S550 to 7; and of course the workaround is not assigning a drive to 0, but then you only have 6 ID's to work with, and they are valuable, since the S550 maximum drive size is 80mb.

aperezbios, 2024-12-29:
@chickeneps Noted, however this is not something that can be fixed in firmware on our side.

chickeneps, 2024-12-31:
My apologies, you can close this issue. This did exist in a earlier firmware and was reported privately, there was no updates on it. I just tried it with the latest firmware and it seems to work. Probably fixed unintentionally somewhere in the past. Please close this issue.