ZuluSCSI/ZuluSCSI-firmware #493: Change LED active HIGH to LOW. Is it possible?

walkero-gr, 2024-12-29:
I am using a ZuluSCSI Compact RP2040 in an Amiga 1200 system and I would like to connect the led to the case led, and be able to see HDD activity from there. 

I tried to use the HDD Led adapter from Alinea (https://www.amiga-shop.net/en/Amiga-Hardware/Amiga-classic-hardware/HDD-LED-adapter-for-Amiga-1200::1301.html) and connect it with the LED connection on the ZuluSCSI, but it seems the Amiga hdd led is always on and gets off when there is activity. This means that it works the opposite way that it should. It is ON when there is no HDD activity, and OFF when there is HDD activity.

Is there a way to change the ZuluSCSI led to work in the opposite way? I checked the latest config INI file, and I haven't seen a variable there to change the behaviour.

Thank you for your help.

morio, 2025-01-09:
@walkero-gr try this build https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/12697424446?pr=497
In your `zuluscsi.ini` file put `InvertStatusLED = 1 ` under `[SCSI]`
```
[SCSI]
InvertStatusLed = 1
```

walkero-gr, 2025-01-10:
@morio 
Thank you so much for your reply. I will have a look at it asap and will let you know if it worked.

aperezbios, 2025-01-29:
@walkero-gr  this is now incorporated in to the latest firmware release. Since we haven't heard back from you, I'm going to close this issue as complete. You're still welcome to comment here if need be.

closed 2025-01-29 by aperezbios