ZuluSCSI/ZuluSCSI-firmware #494: Initiator Parity Issue

chickeneps, 2024-12-29:
Hooking up to a 8GB Seagate. I know it works since it images on a computer. Here's the log.
[zululog.txt](https://github.com/user-attachments/files/18270007/zululog.txt)

chickeneps, 2024-12-29:
I tried EnableParity on and off, same thing really.

aperezbios, 2024-12-31:
@chickeneps this firmware is rather old, can you please re-try with the latest v2024.12.24 release?

Also, please provide the exact make and model of Seagate 8GB HDD, so I can attempt to replicate the issue. I may have something identical or similar on-hand. A photo of the label on the HDD would eliminate all doubt.

chickeneps, 2024-12-31:
Seagate Medalist ST39140N. Tried with 12/24, same thing. Log attached.
[zululog.txt](https://github.com/user-attachments/files/18283147/zululog.txt)

aperezbios, 2025-01-16:
@chickeneps I received this drive in the mail today, but it is is severely ill. I am also unable to succesfully image it from a Linux PC. The first ~250ish megabytes dump fine, but then the drive returns MEDIUM_ERROR. I don't think there's an error on the ZuluSCSI side, given this behavior.

closed 2025-01-16 by aperezbios

chickeneps, 2025-01-16:
Well, good, the most important thing is that the Initiator operation now works with this drive model.