ZuluSCSI/ZuluSCSI-firmware #499: ZuluSCSI Pico Slim suddenly stopped recognizing micro SD cards

Tamh, 2025-01-17:
Las week I bought a ZuluSCSI Pico Slim directly from Rabbit Hole Computing, and I started testing it about 2 days ago. It worked pretty well with my Mac SE FDHD, and I was quite happy about it.

I was playing with it today, backing up a couple of floppy disks I had lying around to a HD emulated in the ZuluSCSI, when suddenly my Mac started lagging.
Finally it threw me for a restart with a message saying that it couldn't find Finder in the booting HD (which was an image in the SD card in the ZuluSCSI), and then it stopped me with the question mark disk (I had already deleted the System Folder in the included HD).

I powered off and on the Mac, then the ZuluSCSI started blinking five times, a sign that the SD card wasn't recognized. That was odd, because it had worked perfectly for the past 2 days.
I formatted another SD card, added some files, and again blinking five times.
I connected it over micro USB in my laptop and after connecting to it using Tera Term, this appears:
```
[12ms] Platform: ZuluSCSI Pico DaynaPORT
[12ms] FW Version: 25.01.09-release Jan  9 2025 20:20:47
[12ms] DIP switch settings: debug log 0, termination 1
[12ms] SCSI termination is enabled
[12ms] Flash chip size: 2048 kB
[12ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
```

This is the last zululog that was written to the SD card:
```
[12ms] Platform: ZuluSCSI Pico DaynaPORT
[12ms] FW Version: 25.01.09-release Jan  9 2025 20:20:47
[12ms] DIP switch settings: debug log 0, termination 1
[12ms] SCSI termination is enabled
[12ms] Flash chip size: 2048 kB
[12ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[35ms] SD card detected, FAT32 volume size: 29807 MB
[35ms] SD MID: 0x9F, OID: 0x54 0x49
[35ms] SD Name: SD32G
[35ms] SD Date: 9/2022
[35ms] SD Serial: 0x54580008
[550ms] Reading configuration from zuluscsi.ini
[1049ms] Active configuration (using system preset "Mac"):
[1049ms] -- SelectionDelay = 255
[1049ms] -- EnableUnitAttention = No
[1049ms] -- EnableSCSI2 = Yes
[1049ms] -- EnableSelLatch = No
[1049ms] -- MapLunsToIDs = No
[1049ms] -- EnableParity = Yes
[3785ms] Finding images in directory /:
[4082ms] -- Opening /NE6.img for id:6 lun:0
[4085ms] ---- Configuring as network based on image name
[4381ms] -- Opening /HD3 Stuff Copy.hda for id:3 lun:0
[4395ms] ---- Configuring as disk drive drive
[4395ms] ---- Read prefetch enabled: 4608 bytes
[4691ms] -- Opening /HD5 Saved HD.hda for id:5 lun:0
[4825ms] ---- Configuring as disk drive drive
[4826ms] ---- Read prefetch enabled: 4608 bytes
[5121ms] -- Opening /HD20 System.hda for id:2 lun:0
[5134ms] ---- WARNING: file /HD20 System.hda is not contiguous. This will increase read latency.
[5134ms] ---- Configuring as disk drive drive
[5135ms] ---- Read prefetch enabled: 4608 bytes
[5431ms] -- Opening /HD1 BlueSCSI Bootstrap.hda for id:1 lun:0
[5498ms] ---- Configuring as disk drive drive
[5499ms] ---- Read prefetch enabled: 4608 bytes
[5641ms] -- Platform supports ROM drive up to 1468 kB
[5641ms] ---- ROM drive image not detected
[5641ms] SCSI ID: 1, BlockSize: 512, Type: 0, Quirks: 1, Size: 524336kB
[5641ms] SCSI ID: 2, BlockSize: 512, Type: 0, Quirks: 1, Size: 524352kB
[5641ms] SCSI ID: 3, BlockSize: 512, Type: 0, Quirks: 1, Size: 41024kB
[5641ms] SCSI ID: 5, BlockSize: 512, Type: 0, Quirks: 1, Size: 1126448kB
[5641ms] SCSI ID: 6, Type: 6, Quirks: 1
[6507ms]  
[6507ms] === Network Initialization ===
[7373ms] Wi-Fi MAC: 00:80:19:04:03:50
[7373ms] Connecting to Wi-Fi SSID "sthng" with WPA/WPA2 PSK
[7569ms] Clock set to: 133000000Hz
[7569ms] Initialization complete!
[8964ms] Toolbox enabled = 0
[16136ms] Successfully connected to Wi-Fi SSID "sthng"
[656543ms] Successfully connected to Wi-Fi SSID "sthng"
```

Nothing else shows, I can read the SD cards without issue in my laptop, and now I'm worried I have a quite light paperweight in my hands.

Tamh, 2025-01-17:
As an experiment, I rolled back to firmware v2024.12.12, and I'm getting this, repeatedly:
```
[12ms] Platform: ZuluSCSI Pico DaynaPORT
[12ms] FW Version: 24.12.12-release Dec 12 2024 20:03:44
[12ms] DIP switch settings: debug log 0, termination 1
[12ms] SCSI termination is enabled
[12ms] Flash chip size: 2048 kB
[12ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[36ms] SDIO SD card error on line 468, error code 2
[39ms] SDIO SD card error on line 468, error code 2
[42ms] SDIO SD card error on line 468, error code 2
[45ms] SDIO SD card error on line 468, error code 2
[48ms] SDIO SD card error on line 468, error code 2
[51ms] SDIO SD card error on line 468, error code 2
[51ms] SD card init failed, sdErrorCode: 2 sdErrorData: 0
[22474ms] SDIO SD card error on line 468, error code 3
[22474ms] SDIO SD card error on line 468, error code 3
[22474ms] SDIO SD card error on line 468, error code 3
[22474ms] SDIO SD card error on line 468, error code 3
[22474ms] SDIO SD card error on line 468, error code 3
[22474ms] SDIO SD card error on line 468, error code 3
[22478ms] SDIO SD card error on line 468, error code 3
[22478ms] SDIO SD card error on line 468, error code 3
[22478ms] SDIO SD card error on line 468, error code 3
[22478ms] SDIO SD card error on line 468, error code 3
[22478ms] SDIO SD card error on line 468, error code 3
[22478ms] SDIO SD card error on line 468, error code 3
[22481ms] SDIO SD card error on line 468, error code 3

```
In this case I inserted my SD card at 22 seconds after boot.

Testing with an older version (24.11.1):
```
[36ms] SDIO SD card error on line 468, error code 2
[39ms] SDIO SD card error on line 468, error code 2
[42ms] SDIO SD card error on line 468, error code 2
[45ms] SDIO SD card error on line 468, error code 2
[48ms] SDIO SD card error on line 468, error code 2
[51ms] SDIO SD card error on line 468, error code 2
[51ms] SD card init failed, sdErrorCode: 2 sdErrorData: 0
[31746ms] SDIO SD card error on line 468, error code 3
[31746ms] SDIO SD card error on line 468, error code 3
[31746ms] SDIO SD card error on line 468, error code 3
[31746ms] SDIO SD card error on line 468, error code 3
[31746ms] SDIO SD card error on line 468, error code 3
```
I inserted my SD card at 31 seconds after boot.

aperezbios, 2025-01-17:
@Tamh the annoyingly-verbose messages are expected with the previous release, as it's something we cleaned up recently.  I'm guessing from your "light paperweight" comment that no back-ups of the images you made were made. Is this correct?

Where did the SD card come from? There's no mention of that, and it's important. Was it purchased from us, or just something you had lying around? Are you able to copy off any image files that may still exist on the SD card you were using? If so, can you compress and send the one you were trying to boot from to us via e-mail, at support@zuluscsi.com? Thanks!

aperezbios, 2025-01-17:
@Tamh It certainly sounds like there could be an issue (premature failure?) with the SD card socket itself. Please contact us via e-mail at support@zuluscsi.com with your order number, as we have no way of cross-referencing your identity from here on GitHub. We'll get you sorted in the morning, one way or another.