ZuluSCSI/ZuluSCSI-firmware #504: Copying from CD-ROM with Initiator mode trigger watchdog timeout

dlundqvist, 2025-01-29:
Hi,

I'm trying to copy a few Amiga CD:s I have, one of them works fine. But two of them cause ZuluSCSI to trigger watchdog timeout:

[zululog.txt](https://github.com/user-attachments/files/18592942/zululog.txt)

USB serial:

[screenlog.txt](https://github.com/user-attachments/files/18593348/screenlog.txt)

Here is one where it works (same CD-ROM):

[zululog.txt](https://github.com/user-attachments/files/18593008/zululog.txt)

All CD:s work fine in the same CD-ROM from AmigaOS. I've done a full copy of each and they all read fine.

Same setup as in #503.

aperezbios, 2025-01-29:
@dlundqvist thanks for the logs. The watchdog timeout happens right after the ZuluSCSI sends StartStopUnit to the CD-ROM drive. Did you use the exact same CD-ROM drive to successfully dump the images, or was it a different drive? What's your cable set-up like? How long is the ribbon cable, etc?

We also still need to know what the track contents of the CD-ROM are, which you would need to collect another way.  Imgburn in windows can display this, once you insert the CD. We need to know if the CD contains only data, or if it's a mixture of data and audio, etc. Please provide this information. If you have

PetteriAimonen, 2025-01-31:
I wonder if this might be an issue with the USB serial port. Do you have USB cable connected when it happens? If yes, can you try without?

dlundqvist, 2025-01-31:
It also happens without USB connected, that's how I triggered it the first times.

dlundqvist, 2025-01-31:
I've dumped all three CD:s with `readom` on an USB CD-ROM, so there's nothing wrong with the discs.

aperezbios, 2025-01-31:
> It also happens without USB connected, that's how I triggered it the first times.

@dlundqvist Then we're going to need a crashlog from when it does this without USB connected, if that's possible.

dlundqvist, 2025-01-31:
Sure. How can I grab that? Just the log file without USB attached?-- daniel31 jan.

aperezbios, 2025-01-31:
> Sure. How can I grab that? Just the log file without USB attached?

Yep, that's it. It needs to crash.

dlundqvist, 2025-01-31:
Here it is:

[zululog.txt](https://github.com/user-attachments/files/18624704/zululog.txt)

PetteriAimonen, 2025-02-03:
Ok, it seems the watchdog timeout is just incidental. Actual problem appears to be that the `START STOP UNIT` command gets stuck in the command phase for longer than 15 seconds. Then ZuluSCSI resets the bus, and the CD drive restarts its initialization process.

In the log file where reading works, `START STOP UNIT` takes 3 seconds. It is not out of the question that an old CD drive might take longer than 15 sec to read the disc table of contents, especially if it has any problems such as dust on the lens.

Can you try this test build which increases the timeout to 60 seconds:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13108542232

dlundqvist, 2025-02-03:
Makes sense. I can absolutely try the test build, but not until end of week as I'm currently out of town. I could also try to clean the lens and use the normal release, to see if that makes a difference.

dlundqvist, 2025-02-08:
So, got time to test the test build. It doesn't hit the watchdog but, also imaging does not start: 

[zululog.txt](https://github.com/user-attachments/files/18719054/zululog.txt)

PetteriAimonen, 2025-02-10:
Hmm, now it is answering the START STOP UNIT after 29 seconds with "NOT READY".
The sense key is weird, 04h/08h indicating `LOGICAL UNIT NOT READY, LONG WRITE IN PROGRESS`.
But it definitely shouldn't be writing anything.

If you can continue testing, here is a build that tries various tricks to try to bring the drive out of this state.
The log file would then indicate if any of them help:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13236494531

dlundqvist, 2025-02-10:
Now it actually imaged the two disks it could not image before:

[zululog.txt](https://github.com/user-attachments/files/18732578/zululog.txt)

The image files are not same size as when I imaged using my USB CD-ROM, but hashes of files inside the ISO filesystem is exactle the same for both images. Interesting.

I also noticed that ZuluSCSI used Read10 for the first disc and Read6 for the second. The first one was imaged much faster because of that. Is it about file system size or something else?

PetteriAimonen, 2025-02-10:
Looks like what it needed was simply STARTSTOP with argument stop, and then start again.
I'll prepare a patch that will include these improvements.

The determination of CD-ROM image sizes is a bit complex. I think many CD imaging programs leave out the leadout area because it doesn't contain actual data. ZuluSCSI just uses the same GET CAPACITY response as with any SCSI device, and I think that can be larger. But I have not studied this in depth.

The read speed dropping seems to be due to the read error that happened at 135176ms:

<pre>
[128781ms] DBG -- BUS_BUSY
[128781ms] DBG ---- SELECTION
[128781ms] DBG ------ SELECTING 3 with initiator ID 7
[128782ms] DBG ---- COMMAND: Read10
[128782ms] DBG ------ OUT: 0x28 0x00 0x00 0x00 0x3A 0x00 0x00 0x02 0x00 0x00 
[128784ms] DBG ---- DATA_IN
[132477ms] DBG scsi_accel_host_read: aborting because target switched transfer phase (IO: 0, CD: 0, MSG: 0)
[132481ms] scsiHostRead: received 22528 bytes, expected 49152
[132481ms] Read failed at byte 802816
[132482ms] DBG ---- Total IN: 825344 OUT: 0 CHECKSUM: 22533
[132482ms] DBG -- BUS_FREE
[132488ms] SCSI read from sector 14848 was incomplete: expected 1048576 got 851968 bytes
[132488ms] DBG scsiInitiatorReadDataToFile: Incomplete transfer
[132489ms] Failed to transfer 512 sectors starting at 14848
[132490ms] Retrying.. 1/5
[132896ms] DBG -- BUS_BUSY
[132896ms] DBG ---- SELECTION
[132896ms] DBG ------ SELECTING 3 with initiator ID 7
[132897ms] DBG ---- COMMAND: Read10
[132897ms] DBG ------ OUT: 0x28 0x00 0x00 0x00 0x3A 0x00 0x00 0x02 0x00 0x00 
[132898ms] DBG ---- DATA_IN
[134766ms] DBG scsi_accel_host_read: aborting because target switched transfer phase (IO: 1, CD: 1, MSG: 0)
[134767ms] scsiHostRead: received 2040 bytes, expected 2048
[134768ms] Read failed at byte 1046528
[134769ms] DBG ---- Total IN: 1048568 OUT: 0 CHECKSUM: 37163
[134769ms] DBG ---- STATUS
[134774ms] DBG ------ IN: 0x00 
[134774ms] DBG ------ STATUS: 0x00
[134774ms] DBG ---- MESSAGE_IN
[134774ms] DBG ------ IN: 0x00 
[134775ms] DBG -- BUS_FREE
[134775ms] DBG scsiInitiatorReadDataToFile: Incomplete transfer
[134775ms] Failed to transfer 512 sectors starting at 14848
[134776ms] Retrying.. 2/5
[135176ms] Multiple failures, retrying sector-by-sector
</pre>

Whatever happened at that point was successfully retried using the sector-by-sector method. By there is indeed a bug that does not reset this state for the next disk, instead continuing in the sector-by-sector mode until reset. It would make sense to return to normal reading once past the failing spot.

PetteriAimonen, 2025-02-11:
@dlundqvist If you can check if this finished version of the fix works for you, it would be great:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13260431849

dlundqvist, 2025-02-11:
Not sure what I'm doing wrong but I can't get the image from latest run to work, when booting the board I get this:

```
[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 25.02.05-release Feb 11 2025 10:41:23
[13ms] DIP switch settings: debug log 1, termination 1
[13ms] SCSI termination is enabled
[14ms] Flash chip size: 4096 kB
[14ms] SCSI initiator mode selected by DIP switch, expecting SCSI disks on the bus
```
This gives me watchdog timeout as before. Neither https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13260431849 nor https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13261005896 seem to give me a development image.

While using image from previously linked run,https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13236494531, I get:
```
[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 25.02.01-devel Feb 10 2025 08:26:49
[13ms] DIP switch settings: debug log 1, termination 1
[13ms] SCSI termination is enabled
[14ms] Flash chip size: 4096 kB
[14ms] SCSI initiator mode selected by DIP switch, expecting SCSI disks on the bus
```

PetteriAimonen, 2025-02-11:
@dlundqvist Hmm, nothing after those initial messages? That's weird.

dlundqvist, 2025-02-11:
I did, it was to show that I don't get a development image when downloading from those actions. But I must have made a mistake downloading, I'll try again later tonight.

dlundqvist, 2025-02-11:
It works with https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13236494531 but not with today's release:

[zululog.txt](https://github.com/user-attachments/files/18757883/zululog.txt)

I don't see the new log message `Attempting to recover from CD-ROM Logical Unit Not Ready ASC = 0x04 ASCQ = 0x08` in the release image.

PetteriAimonen, 2025-02-11:
Yeah, I took out that specific debug message when I generalized the fix.

Looks like I failed to account for that the first read operation also takes long, even though it was visible from the earlier logs. I'll decide tomorrow whether I'll increase the watchdog timeout altogether or just for reads.

PetteriAimonen, 2025-02-14:
@dlundqvist After reviewing the logs I think it is indeed just a timeout issue that is remaining. This build will increase it for all commands:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13323692498

(Don't worry about the version number, it is not updated automatically currently. The build date can be used to verify that the firmware has upgraded.)

dlundqvist, 2025-02-15:
Works much better but still some issues. This is imaging CygnusEd disc five times:

[zululog.txt](https://github.com/user-attachments/files/18810902/zululog.txt)

It finished five times, but only image 1, 3 and 4 was correct. Image 2 could be mounted but differs in contents from what it should, and image 5 was not a valid ISO file even.

And this is with Roadshow disc:

[zululog-Roadshow.txt](https://github.com/user-attachments/files/18811002/zululog-Roadshow.txt)

Again imaging five times. Image 1, 3 and 5 came out correct. Image 2 and 4 was not valid ISO files.

Imaging the third disc, AmigaOS, worked five out if five times. Also much faster even though is it's about twice as large.

I'm not sure the remaining problems have anything to do with ZuluSCSI. Might very well be the combination of drive and discs.

Thanks for working on this though.

dlundqvist, 2025-02-16:
I think what's been done so far is a big improvement and I'm happy with this. Thanks.

closed 2025-05-05 by aperezbios