ZuluSCSI/ZuluSCSI-firmware #519: Apple IBM H3171-S2 hard drive incompatible with initiator mode

dougg3, 2025-02-16:
**Hardware:** ZuluSCSI Compact Homebrew
**Firmware:** 25.02.11-release Feb 11 2025 17:57:30
**Hard Drive:** IBM H3171-S2 with Apple sticker, 160 MB hard drive

Previously with firmware 23.12.13-release, I was able to use initiator mode to dump this drive. Here's the zululog.txt output:

```
[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 23.12.13-release Dec 13 2023 00:34:03
[13ms] DIP switch settings: debug log 0, termination 1
[13ms] SCSI termination is enabled
[14ms] Flash chip size: 2048 kB
[14ms] SCSI initiator mode selected by DIP switch, expecting SCSI disks on the bus
[28ms] SD card detected, FAT64 volume size: 59312 MB
[28ms] SD MID: 0x27, OID: 0x50 0x48
[29ms] SD Name: SD64G
[29ms] SD Date: 10/2022
[29ms] SD Serial: 0xDA359ADA
[784ms] Initialization complete!
[1808ms] RequestSense response: 0x70 0x00 0x02 0x00 0x00 0x00 0x00 0x18 0x00 0x00 0x00 0x00 0x04 0x01 0x00 0x00 0x00 0x00 
[1809ms] Target 0 reports NOT_READY, running STARTSTOPUNIT
[2008ms] RequestSense response: 0x70 0x00 0x02 0x00 0x00 0x00 0x00 0x18 0x00 0x00 0x00 0x00 0x04 0x01 0x00 0x00 0x00 0x00 
[2009ms] START STOP UNIT on target 0 failed, sense key 0x02
[2208ms] RequestSense response: 0x70 0x00 0x02 0x00 0x00 0x00 0x00 0x18 0x00 0x00 0x00 0x00 0x04 0x01 0x00 0x00 0x00 0x00 
[2209ms] Target 0 reports NOT_READY, running STARTSTOPUNIT
[2408ms] RequestSense response: 0x70 0x00 0x02 0x00 0x00 0x00 0x00 0x18 0x00 0x00 0x00 0x00 0x04 0x01 0x00 0x00 0x00 0x00 
[2409ms] START STOP UNIT on target 0 failed, sense key 0x02
[12330ms] SCSI ID 0 capacity 333924 sectors x 512 bytes
[12330ms] [SCSI0]
[12331ms]   Vendor = "IBM     "
[12331ms]   Product = "H3171-S2        "
[12331ms]   Version = "S52H"
[12332ms]   Type = 0
[12332ms] Drive total size is 163 MiB
[12373ms] Preallocating image file
[12386ms] Starting to copy drive data to HD00_imaged.hda
[12571ms] SCSI read succeeded, sectors done: 512 / 333924 speed 1440 kB/s
[12702ms] SCSI read succeeded, sectors done: 1024 / 333924 speed 2016 kB/s
[12828ms] SCSI read succeeded, sectors done: 1536 / 333924 speed 2097 kB/s
...
[116909ms] SCSI read succeeded, sectors done: 331776 / 333924 speed 1230 kB/s
[117141ms] SCSI read succeeded, sectors done: 332288 / 333924 speed 1129 kB/s
[117342ms] SCSI read succeeded, sectors done: 332800 / 333924 speed 1310 kB/s
[117542ms] SCSI read succeeded, sectors done: 333312 / 333924 speed 1317 kB/s
[117776ms] SCSI read succeeded, sectors done: 333824 / 333924 speed 1125 kB/s
[201199ms] SCSI read succeeded, sectors done: 333924 / 333924 speed 0 kB/s
```

It did seem to get hung up at the end and never spun down the drive when it was finished, as it does with other drives I've tested. But at least it would always dump all the contents.

Now, with the new 25.02.11-release firmware, initiator mode doesn't work on this drive at all, or at least it's far too slow to be usable. zululog.txt:

```
[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 25.02.11-release Feb 11 2025 17:57:30
[13ms] DIP switch settings: debug log 0, termination 1
[13ms] SCSI termination is enabled
[14ms] Flash chip size: 2048 kB
[14ms] SCSI initiator mode selected by DIP switch, expecting SCSI disks on the bus
[31ms] SD card detected, FAT64 volume size: 59312 MB
[32ms] SD MID: 0x27, OID: 0x50 0x48
[32ms] SD Name: SD64G
[32ms] SD Date: 10/2022
[32ms] SD Serial: 0xDA359ADA
[36ms] Speed grade set to Default, skipping reclocking
[288ms] InitiatorID set to ID 0x07
[293ms] Clock set to: 125000000Hz
[293ms] Initialization complete!
[1528ms] -- START STOP UNIT on target 0 failed with status 2 and sense_key 0x02 (NOT_READY)
[2931ms] -- START STOP UNIT on target 0 failed with status 2 and sense_key 0x02 (NOT_READY)
[... 169 lines cut ...]
[74ms] Presenting SD card as USB storage device
[74ms] SCSI Initiator: activating USB MSC mode
[327ms] InitiatorID set to ID 0x07
[750ms] -- START STOP UNIT on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[2502ms] SCSI Initiator: found 0 SCSI drives
[5304ms] -- START STOP UNIT on target 0 failed with status 2 and sense_key 0x02 (NOT_READY)
[11367ms] Found SCSI drive with ID 0: IBM      H3171-S2         capacity 163 MB
[14068ms] -- SCSI Initiator read on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[14229ms] -- READ CAPACITY on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[14233ms] -- SCSI Initiator read on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[14237ms] -- READ CAPACITY on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[14253ms] -- READ CAPACITY on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[15003ms] USB MSC: 35 commands, 0 kB/s
[20004ms] USB MSC: 5 commands, 0 kB/s
[25005ms] USB MSC: 5 commands, 0 kB/s
[30006ms] USB MSC: 5 commands, 0 kB/s
[35007ms] USB MSC: 5 commands, 0 kB/s
[40008ms] USB MSC: 5 commands, 0 kB/s
[45009ms] USB MSC: 5 commands, 0 kB/s
[50010ms] USB MSC: 5 commands, 0 kB/s
[55011ms] USB MSC: 5 commands, 0 kB/s
...
```

Happy to perform any testing or debugging you might need! Thanks!

dougg3, 2025-02-17:
Thanks! Here's a log with debug enabled -- realized I should have done that from the start.

[zululog.txt](https://github.com/user-attachments/files/18817766/zululog.txt)

dougg3, 2025-02-17:
And here's a log of success using the old working version with debug enabled. Notice how the final `SCSI read succeeded` appears several minutes late. The drive still hadn't spun down by the time I shut it off, but the dump was good.

[zululog.txt](https://github.com/user-attachments/files/18817920/zululog.txt)

PetteriAimonen, 2025-02-17:
@dougg3 Thanks for the logs!

It appears the target drive is somehow re-entering the COMMAND phase after the command has already been completed.
The most obvious difference in the logs is READ6 vs. READ10 command, which is affected by the drive being detected as SCSI-1 device. The old firmware only used READ10.

Can you try this firmware build:

dougg3, 2025-02-18:
Thanks @PetteriAimonen! I also noticed the READ6 vs READ10 difference in the logs, seems like a good clue.

I just ran a test and it looks like the problem is also present in 2024.03.07. Here's a debug log of a couple of minutes of action:

[zululog.txt](https://github.com/user-attachments/files/18834154/zululog.txt)

Also, I tested the version directly before it (2024.02.06), and it works great. Looks like you pinpointed the problematic version perfectly. Here's a success log from 2024.02.06:

[zululog-2024.02.06.txt](https://github.com/user-attachments/files/18834313/zululog-2024.02.06.txt)

BTW, 2024.02.06 even successfully spun down the disk when it finished imaging, which didn't seem to work properly on this drive in 2023.12.13! It's like the watchdog timeout at the end is something new that fixed that little annoyance.

dougg3, 2025-02-18:
@PetteriAimonen I did a little more tinkering with this after you helped reveal the version where the problem started. First I reverted 1e0aa9641a11806eda27bf4008755c84096e0ef3, but that didn't fix it. Then I reverted 57210a6d22a8fe0e9787a9ded2f6ad548704933e, and that fixed the problem -- now initiator mode dumps the drive successfully. Reverting required some conflict resolution, but nothing crazy.

That's a pretty big commit, so I'm not sure exactly which part of it is causing the problem I'm seeing.

BTW, even after that revert, USB MSC initiator doesn't work with this drive. Seems to fail on READ_CAPACITY. It's something specific to this drive, though -- the Quantum drive still works fine.

```
[74ms] SCSI Initiator: activating USB MSC mode
[705ms] -- START STOP UNIT on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[2457ms] SCSI Initiator: found 0 SCSI drives
[5258ms] -- START STOP UNIT on target 0 failed with status 2 and sense_key 0x02 (NOT_READY)
[13750ms] Found SCSI drive with ID 0: IBM      H3171-S2         capacity 163 MB
[17000ms] -- SCSI Initiator read on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[17004ms] -- READ CAPACITY on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[17009ms] -- SCSI Initiator read on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[17013ms] -- READ CAPACITY on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[17025ms] -- READ CAPACITY on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[17367ms] USB MSC: 54 commands, 0 kB/s
[22368ms] USB MSC: 37 commands, 0 kB/s
[27369ms] USB MSC: 5 commands, 0 kB/s
[32370ms] USB MSC: 5 commands, 0 kB/s
[37371ms] USB MSC: 5 commands, 0 kB/s
[42372ms] USB MSC: 5 commands, 0 kB/s
[47378ms] USB MSC: 5 commands, 0 kB/s
```

Working log of USB MSC initiator after the revert with Quantum drive:

```
[74ms] SCSI Initiator: activating USB MSC mode
[10010ms] Found SCSI drive with ID 0: QUANTUM  CTS160S          capacity 163 MB
[11512ms] SCSI Initiator: found 1 SCSI drives
[31546ms] USB MSC: 733 commands, 66 kB/s
[36547ms] USB MSC: 125 commands, 10 kB/s
[41548ms] USB MSC: 5 commands, 0 kB/s
[46549ms] USB MSC: 5 commands, 0 kB/s
[51550ms] USB MSC: 5 commands, 0 kB/s
[56551ms] USB MSC: 5 commands, 0 kB/s
[61552ms] USB MSC: 5 commands, 0 kB/s
[66553ms] USB MSC: 5 commands, 0 kB/s
[71554ms] USB MSC: 5 commands, 0 kB/s
[76555ms] USB MSC: 5 commands, 0 kB/s
[81556ms] USB MSC: 5 commands, 0 kB/s
[86557ms] USB MSC: 5 commands, 0 kB/s
[91558ms] USB MSC: 5 commands, 0 kB/s
[96559ms] USB MSC: 5 commands, 0 kB/s
[101560ms] USB MSC: 5 commands, 0 kB/s
[106561ms] USB MSC: 5 commands, 0 kB/s
[111562ms] USB MSC: 5 commands, 0 kB/s
[116563ms] USB MSC: 113 commands, 9 kB/s
[121564ms] USB MSC: 196 commands, 33 kB/s
[126586ms] USB MSC: 167 commands, 53 kB/s
[131587ms] USB MSC: 265 commands, 84 kB/s
[136588ms] USB MSC: 5 commands, 0 kB/s
```

PetteriAimonen, 2025-02-18:
Thanks for the information. I think there is nothing specifically wrong with that commit, it is just that it reads the SCSI version from the drive and uses the READ6 command on SCSI-1 drives.

For some reason READ6 is not working properly on that drive. I will try to dig into some old standards, maybe there is some SCSI-1 specific thing we are not currently implementing correctly. It looks like the command execution is ok, it is just that the device is not releasing the bus at the end - it is waiting for ZuluSCSI to do something more.

PetteriAimonen, 2025-02-18:
@dougg3 Here is a build that tries poking ACK after the device gets stuck in this state. A debug log from it could give me some clues to what is happening:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13392726743

PetteriAimonen, 2025-02-19:
The speed difference is partially due to the unnecessary delays in the recovery hack, and partially due to transferring only 256 sectors at a time vs. the 512 sectors for SCSI-2.

Weirdly this is only happening with the READ6 command, and the phase sequence is always COMMAND->STATUS->MESSAGE_IN. But even searching back to SASI (predating SCSI-1) standards I couldn't find anything that would explain this behavior.

The online information on H3171-S2 suggests it should support SCSI-2, but the inquiry response reports SCSI-1. I think this is somehow specially configured version of the drive, perhaps for compatibility with some specific system.

There are two possible approaches to resolving this:

1) Implement the ACK cycling after command complete, with smaller delays. This would still cause the READ6 smaller sector count performance degradation.
2) Always try READ10 and fall back to READ6 if target reports error.

I chose to go with 1), as it has less risk of causing regressions with other drives.
The performance difference of READ6 vs. READ10 for SCSI-1 drives appears to be about 25%, which is not that critical when the drives are small enough that whole imaging is just a few minutes.

Can you still try this final version of the patch before we merge it?

https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13409411464

dougg3, 2025-02-20:
Thanks for all your quick patches to test! That makes sense that it would have to be a little slower, and what you are saying about option 1 makes sense to me too (definitely safer in terms of not causing regressions). I guess a third option could be a quirks list or something like that. But anyway...the ACK cycling thing you implemented works like a charm. Dumped the drive perfectly on the first try.

For good measure, I attached successful debug logs of initiator mode with both the H3171-S2 and my Quantum ProDrive ELS I've also been testing with.

[zululog-H3171-S2.txt](https://github.com/user-attachments/files/18880114/zululog-H3171-S2.txt)
[zululog-CTS160S.txt](https://github.com/user-attachments/files/18880115/zululog-CTS160S.txt)

As far as USB MSC initiator mode goes, this new firmware doesn't work with the H3171-S2. Windows gets stuck asking to insert a removable disk over and over again. The previous test firmware you gave me did work okay with it, except writes didn't actually go through.

```
[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 25.02.11-release Feb 19 2025 09:33:48
[13ms] DIP switch settings: debug log 0, termination 1
[13ms] SCSI termination is enabled
[14ms] Flash chip size: 2048 kB
[14ms] SCSI initiator mode selected by DIP switch, expecting SCSI disks on the bus
[53ms] No SD card detected, imaging to SD card not possible
[73ms] Continuing without SD card
[73ms] Clock set to: 125000000Hz
[73ms] Initialization complete!
[73ms] Entering USB MSC initiator mode
[74ms] Presenting SD card as USB storage device
[74ms] SCSI Initiator: activating USB MSC mode
[327ms] InitiatorID set to ID 0x07
[2417ms] SCSI Initiator: found 0 SCSI drives
[5218ms] -- START STOP UNIT on target 0 failed with status 2 and sense_key 0x02 (NOT_READY)
[11366ms] Found SCSI drive with ID 0: IBM      H3171-S2         capacity 163 MB
[15003ms] USB MSC: 102 commands, 12 kB/s
[16691ms] Prefetch of sector 0x00000008 failed: status 2
[20006ms] USB MSC: 642 commands, 11 kB/s
[25007ms] USB MSC: 593 commands, 0 kB/s
[30008ms] USB MSC: 666 commands, 0 kB/s
[35009ms] USB MSC: 595 commands, 0 kB/s
[40010ms] USB MSC: 623 commands, 0 kB/s
[45011ms] USB MSC: 683 commands, 0 kB/s
[50012ms] USB MSC: 623 commands, 0 kB/s
```

If I hack the code in ZuluSCSI_msc_initiator.cpp to always do a READ(10), the problem goes away.

Is writing with USB MSC initiator mode tested at all? I am finding that reads work fine on both drives (H3171-S2 after a READ(10) hack) but writes don't. It acts like it's writing successfully, but the data doesn't actually end up changed on the drive. I'm wondering if there's a deeper problem since I'm seeing the same behavior across multiple drives. I can open a different issue for that though, since it's kind of a completely different topic -- your patch here definitely fixes normal initiator mode with the H3171-S2!

PetteriAimonen, 2025-02-20:
Uh, yeah, seems like I have forgotten to test USB MSC initiator writing at all, because wrong function was being called :)
Here is a test build that includes the prior fix and also fixes write support:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13430803054

But as to why USB MSC doesn't work with H3171-S2, can you turn on the debug switch and then capture debug log either through USB serial port, or by having this in zuluscsi.ini (in which case the log is saved to SD card):

<pre>
[SCSI]
Debug = 1
InitiatorMSC = 1
</pre>

dougg3, 2025-02-21:
Haha, that would be a problem wouldn't it? :-) Thanks for the quick fix on that! Here are a few debug logs, using the zuluscsi.ini contents you posted. In summary, you definitely fixed writes! Woohoo!

On the first debug attempt, it didn't work. Everything just got hung up in HxD when I tried to open the disk and I eventually had to unplug the ZuluSCSI to un-hang it. Not the same symptom I saw without debugging though. Hopefully the debug dump is still useful.

[zululog1.txt](https://github.com/user-attachments/files/18896667/zululog1.txt)

On the second debug attempt, it actually worked. Everything was sluggish, but it actually did work. I wrote a bunch of data (what I've been trying to do this whole time for a fun little project) and it actually did write successfully! Very exciting because now I have a way to put the test data onto this drive that I've been trying to put on it! Then afterward I tried to eject the disk safely in Windows and the eject eventually timed out. I ended up just unplugging the USB eventually. I verified afterward that the write was definitely 100% successful.

[zululog2.txt](https://github.com/user-attachments/files/18896671/zululog2.txt)

Then I tried with debugging disabled (no SD card inserted) and it went back to the original behavior of constantly asking me to insert a disk. I think maybe the debugging being enabled is slowing things down enough to change the behavior or something? BTW, Windows is very annoying when it's stuck asking me to insert a disk...it takes over all keyboard focus and really tries to prevent me from doing anything else!

I unplugged everything and powered it all off and on again, and tried again with debugging disabled and it acted like it was going to work for a while, but eventually failed when I opened the disk with HxD and started getting stuck again in the loop of asking to insert a disk over and over again. The bad behavior seems to coincide with `[81878ms] Prefetch of sector 0x00000001 failed: status 2`. From that point on, Windows always thinks it needs a disk inserted, over and over.

Unfortunately I don't know how to get a debug log of this because when debugging is enabled this problem doesn't seem to occur, lol. Here's the serial console of the second attempt, I know it's probably not that useful but I'll post it anway.

```
[12ms] Platform: ZuluSCSI RP2040
[12ms] FW Version: 25.02.11-release Feb 20 2025 08:19:47
[13ms] DIP switch settings: debug log 0, termination 1
[13ms] SCSI termination is enabled
[14ms] Flash chip size: 2048 kB
[14ms] SCSI initiator mode selected by DIP switch, expecting SCSI disks on the bus
[53ms] No SD card detected, imaging to SD card not possible
[73ms] Continuing without SD card
[73ms] Clock set to: 125000000Hz
[73ms] Initialization complete!
[73ms] Entering USB MSC initiator mode
[74ms] Presenting SD card as USB storage device
[74ms] SCSI Initiator: activating USB MSC mode
[327ms] InitiatorID set to ID 0x07
[694ms] -- START STOP UNIT on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)
[2447ms] SCSI Initiator: found 0 SCSI drives
[5249ms] -- START STOP UNIT on target 0 failed with status 2 and sense_key 0x02 (NOT_READY)
[11377ms] Found SCSI drive with ID 0: IBM      H3171-S2         capacity 163 MB
[40029ms] USB MSC: 146 commands, 17 kB/s
[45040ms] USB MSC: 399 commands, 34 kB/s
[50041ms] USB MSC: 323 commands, 26 kB/s
[55042ms] USB MSC: 41 commands, 3 kB/s
[60043ms] USB MSC: 5 commands, 0 kB/s
[65099ms] USB MSC: 92 commands, 7 kB/s
[70100ms] USB MSC: 26 commands, 1 kB/s
[75101ms] USB MSC: 5 commands, 0 kB/s
[80104ms] USB MSC: 162 commands, 15 kB/s
[81878ms] Prefetch of sector 0x00000001 failed: status 2
[85105ms] USB MSC: 477 commands, 3 kB/s
[90106ms] USB MSC: 617 commands, 0 kB/s
[95107ms] USB MSC: 555 commands, 0 kB/s
```

I couldn't get Git to find the exact a126663 SHA you used, but I see the PRs were merged to main so I just built main with my same hack to always use READ(10) and WRITE(10) in ZuluSCSI_msc_initiator.cpp and everything works 100% fine that way. So I think there's still something funky about how the READ(6) command interacts with this drive.

Thanks again for all your help on this!!! I'm just excited at this point because I was able to write the test data I needed to put on the drive.

PetteriAimonen, 2025-02-21:
Funny, the logs don't show anything going wrong on ZuluSCSI end, but somehow Windows is convinced that something is wrong.

But I decided now to implement the autodetection of READ10 support, with optional .ini file override.
Here is a test build (which also includes another requested config option, readonly mode):
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13452597483

dougg3, 2025-02-22:
Yeah, I'm not sure what was going on with those logs. It seems like they got cut off early maybe. It's also super weird that debugging enabled made it work.

I like your solution! Hopefully the READ10 autodetection doesn't adversely affect some other drives. Test results with my IBM drive:

- Normal initiator mode works great
- USB MSC initiator mode works great, both reading and writing. I did notice HxD showed the old data after the write finished, until I scrolled away and back. Might just be an HxD bug though.
- Confirmed that `InitiatorMSCReadOnly = 1` prevents writes
- Confirmed that I run into problems if I add `InitiatorUseRead10 = 0`, as expected since this drive doesn't like READ(6). It worked for a while and then hung up at `[77365ms] -- SCSI Initiator read on target 0 failed with status 2 and sense_key 0x00 (UNKNOWN)`
- Re-confirmed that all is good after I remove `InitiatorUseRead10 = 0`

All in all, I'd say I am 100% happy with this solution. Thank you so much for all your fast work on fixing this!

aperezbios, 2025-02-22:
@dougg3 great, thanks for all the feedback! These changes are incorporated in to the v2025.02.21 release that I cut earlier today, which can be downloaded from https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/v2025.02.21

closed 2025-02-22 by aperezbios