ZuluSCSI/ZuluSCSI-firmware #526: Toolbox filenames truncated before verifying extension

jhanssen, 2025-03-02:
It appears that several of the `Toolbox.cpp` calls truncates file name before validating that they're valid CD file names, for example if a file name is longer than 32 bytes (the truncation point) and the file is not a valid CD image (let's say a `.cue` file instead) and the user then performs a `TOOLBOX_SET_NEXT_CD` call with an index that lands on this file then `toolboxFilenameValid()` will return true due to the extension being truncated and the system will potentially hang when trying to list the directory of the emulated CD drive

aperezbios, 2025-03-02:
@jhanssen Thanks for the report. Can you be more explicit specify what "the system" you're referring to is? What's the OS/software you're using? With the information provided thus far, there's insufficient information to reproduce the problem.

jhanssen, 2025-03-02:
Sorry for the confusion. By "the system" I'm referring to the host computer the ZuluSCSI is installed in. I'm using a computer running MS-DOS 6.22 and using the escsitoolbox program to change the CD image. The issue can be reproduced by having one .bin and one .cue file on an SD card where the combined length of the base name and the extension is > 32 bytes, the problem lies with the following code in `get_file_from_index()`

```
        // truncate filename the same way listing does, before validating name
        size_t len = file_test.getName(name, MAX_FILE_PATH);
        if (len > MAX_MAC_PATH)
            name[MAX_MAC_PATH] = 0x0;
        // validate filename
        if(!toolboxFilenameValid(name, isCD))
```

`MAX_MAC_PATH` is defined to `32` and thus the file name is truncated to 32 bytes before running through the validation code and that means that the extension is no longer `.cue` when running the `scsiDiskFilenameValid()` function that `toolboxFilenameValid()` calls.

morio, 2025-03-12:
@jhanssen could you try this build. I did a blind fix, but hopefully it fixes your issue
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13821065412
Thanks for tracking down the issue in the code.

closed 2026-04-02 by aperezbios