ZuluSCSI/ZuluSCSI-firmware #527: Non-boot volumes fail to mount when toolbox is enabled

snhirsch, 2025-03-06:
I find the scuzEMU-0.5 tool to be very handy for shuffling CD images on my Pico Slim, but when I add the 'EnableToolbox = 1' line to zuluscsi.ini none of the secondary MacOS volumes get mounted (specifically, I boot from HD2 and rely on HD0 and HD1 being available).  Some of the SCSI probe software can see the devices, but nothing is able to mount them.  The network device (disabled in the trace below) is operational and the CD images can be mounted.  It's just HD0 and HD1 that are MIA.  If I do not enable the toolbox everything functions as it should.

Here's the startup log:

```
[12ms] Platform: ZuluSCSI Pico DaynaPORT
[12ms] FW Version: 24.12.12-release Dec 12 2024 20:03:44
[12ms] DIP switch settings: debug log 0, termination 1
[12ms] SCSI termination is enabled
[12ms] Flash chip size: 2048 kB
[12ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[30ms] SD card detected, FAT32 volume size: 30424 MB
[30ms] SD MID: 0x03, OID: 0x53 0x44
[30ms] SD Name: SK32G
[30ms] SD Date: 11/2024
[30ms] SD Serial: 0x83D33A56
[33ms] Reading configuration from zuluscsi.ini
[35ms] Active configuration (using system preset "Mac"):
[35ms] -- SelectionDelay = 255
[35ms] -- EnableUnitAttention = No
[35ms] -- EnableSCSI2 = Yes
[35ms] -- EnableSelLatch = No
[35ms] -- MapLunsToIDs = No
[35ms] -- EnableParity = Yes
[52ms] SCSI3 searching default optical image directory 'CD3'
[57ms] -- Opening 'CD3/MAC_OS_7-6-1_RETAIL.ISO' for id: 3
[84ms] ---- Configuring as CD-ROM drive
[84ms] ---- Read prefetch enabled: 4608 bytes
[105ms] Finding images in directory /:
[106ms] -- Opening /HD1.img for id:1 lun:0
[563ms] ---- Configuring as disk drive drive
[564ms] ---- Read prefetch enabled: 4608 bytes
[565ms] -- Opening /NE4.img for id:4 lun:0
[565ms] ---- Configuring as network based on image name
[566ms] -- Opening /HD0.img for id:0 lun:0
[618ms] ---- Configuring as disk drive drive
[619ms] ---- Read prefetch enabled: 4608 bytes
[620ms] -- Opening /HD2.img for id:2 lun:0
[1060ms] ---- Configuring as disk drive drive
[1061ms] ---- WARNING: This drive contains the LIDO driver and may cause issues.
[1061ms] ---- Read prefetch enabled: 4608 bytes
[1063ms] -- Platform supports ROM drive up to 1468 kB
[1063ms] ---- ROM drive image not detected
[1063ms] SCSI ID: 0, BlockSize: 512, Type: 0, Quirks: 1, Size: 524336kB
[1063ms] SCSI ID: 1, BlockSize: 512, Type: 0, Quirks: 1, Size: 2097152kB
[1063ms] SCSI ID: 2, BlockSize: 512, Type: 0, Quirks: 1, Size: 2097152kB
[1063ms] SCSI ID: 3, BlockSize: 2048, Type: 2, Quirks: 1, Size: 271964kB, Removable
[1063ms] SCSI ID: 4, Type: 6, Quirks: 1
[1063ms] Found 1 removable device, to set an eject button see EjectButton in the 'zuluscsi.ini', or the http://zuluscsi.com/manual
[1929ms]  
[1929ms] === Network Initialization ===
[2795ms] Wi-Fi MAC: 00:80:19:08:E3:15
[2795ms] Connecting to Wi-Fi SSID "" with no authentication
[2981ms] Clock set to: 133000000Hz
[2981ms] Initialization complete!
[4184ms] Toolbox enabled = 1
```
So ZuluSCSI is certainly aware of the other volumes.

aperezbios, 2025-03-06:
@snhirsch I believe this was a known bug that's already fixed. Please update to the latest version of the firmware and let us know if it resolves your issue. 

For your convenience: https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v2025.02.21/ZuluSCSI_Pico_DaynaPORT_2025-02-21.bin

Just copy that .bin file to the SD card, and the ZuluSCSI will detect it, self-update, and reboot. It takes under two seconds.

snhirsch, 2025-03-06:
Unfortunately that did not correct the issue.  I see no change in the mount behavior and no differences in the log (beyond calling out the later firmware versions).

aperezbios, 2025-03-06:
@snhirsch thanks for updating, and for the update. We will attempt to reproduce in house.

aperezbios, 2025-03-06:
@snhirsch which SCSI ID corresponds with your boot volume?

aperezbios, 2025-03-06:
> I have some further information: I believe it's actually the presence of a 'CD3' subdirectory (populated with ISO images) that's creating the issue. With that deleted the other volumes are visible again - even with the toolbox enabled. But, of course, this prevents me from swapping images dynamically.

Thanks, this is helpful information that will hopefully let us reproduce the issue.

snhirsch, 2025-03-06:
@aperezbios The boot volume is HD2.

snhirsch, 2025-03-06:
More info:  There's a device ordering issue involved.  If I reshuffle the image names such that the boot image is HD0, then all the volumes appear and the toolbox is functional.  Otherwise it seems to ignore disk images that are a lower SCSI id than the boot drive.