ZuluSCSI/ZuluSCSI-firmware #535: Latest firmware (25.03.10 release) locks up Amiga 3000 SCSI

jdryyz, 2025-03-15:
Hi all.

I noticed my RP 2040's firmware hadn't been updated for a over a year. I successfully applied the latest firmware but my Amiga 3000's SCSI bus now locks up when trying to boot (solid activity light).

Fortunately, I was able to just rollback to the last version I installed and all is well again. It would take a while to determine the specific firmware changes that made this happen since there were several since the last one I had installed.

I have attached my logs if that is useful to anyone.

[zululog_24.02.06-release.txt](https://github.com/user-attachments/files/19264779/zululog_24.02.06-release.txt)
[zululog_25.03.10-release.txt](https://github.com/user-attachments/files/19264778/zululog_25.03.10-release.txt)

aperezbios, 2025-03-15:
@jdryyz thanks for this report. Can you do the same thing again, but with the debug logging enabled? You can temporarily enable it with debug=1 in your ZuluSCSI.ini file, or via the DIP switch if you have a full-size board. Either will work, just make sure you disable it after you generate the logs.

jdryyz, 2025-03-15:
Here you go.

[zululog_25.03.10-rel-debug.txt](https://github.com/user-attachments/files/19265005/zululog_25.03.10-rel-debug.txt)

aperezbios, 2025-03-17:
@jdryyz thanks again. This log shows that everything appears to be working fine/as expected for the first ~30 seconds. Then, things go sideways with the following output. We will investigate further as time allows.

```[31912ms] DBG ---- SELECTION: 4
[31912ms] DBG ---- MESSAGE_OUT
[31913ms] DBG ------ OUT: 0xC0 
[31913ms] DBG ---- COMMAND: ReadCapacity
[31913ms] DBG ------ OUT: 0x25 0x00 0x00 0x00 0x00 0x00 
[31914ms] DBG ------ OUT: 0x00 0x00 0x00 0x00 
[31915ms] DBG ---- DATA_IN, syncOffset 12 syncPeriod 53
[31915ms] DBG ---- Total IN: 8 OUT: 0 CHECKSUM: 26140
[31916ms] DBG ---- STATUS: 0 GOOD
[31920ms] DBG ---- MESSAGE_IN
[31920ms] DBG ------ IN: 0x00 
[31921ms] DBG -- BUS_FREE
[31922ms] DBG -- BUS_BUSY
[31922ms] DBG ---- SELECTION: 4
[31922ms] DBG ---- MESSAGE_OUT
[31922ms] DBG ------ OUT: 0xC0 
[31923ms] DBG ---- COMMAND: ModeSense
[31923ms] DBG ------ OUT: 0x1A 0x00 0x04 0x00 0xFE 0x00 
[31924ms] DBG ---- DATA_IN, syncOffset 12 syncPeriod 53
[31924ms] WARNING: Host used command 0x1A which is affected by drive geometry. Current settings are 63 sectors x 255 heads = 16065 but image size of 31457280 sectors is not divisible. This can cause error messages in diagnostics tools.
[31927ms] DBG ---- Total IN: 36 OUT: 0 CHECKSUM: 23957
[31927ms] DBG ---- STATUS: 0 GOOD
[31931ms] DBG ---- MESSAGE_IN
[31931ms] DBG ------ IN: 0x00 
[31931ms] DBG -- BUS_FREE
[31933ms] DBG -- BUS_BUSY
[31933ms] DBG ---- SELECTION: 4
[31933ms] DBG ---- MESSAGE_OUT
[31933ms] DBG ------ OUT: 0xC0 
[31934ms] DBG ---- COMMAND: Read6
[31934ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00 
[31934ms] DBG ------ Read 1x512 starting at 0
[31935ms] DBG ---- DATA_IN, syncOffset 12 syncPeriod 53
[31936ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 65053
[31936ms] DBG ---- STATUS: 0 GOOD
[31940ms] DBG ---- MESSAGE_IN
[31940ms] DBG ------ IN: 0x00 
[31941ms] DBG -- BUS_FREE
[31942ms] DBG -- BUS_BUSY
[31942ms] DBG ---- SELECTION: 4
[31942ms] DBG ---- MESSAGE_OUT
[31943ms] DBG ------ OUT: 0xC0 
[31943ms] DBG ---- COMMAND: Read6
[31943ms] DBG ------ OUT: 0x08 0x00 0x00 0x01 0x01 0x00 
[31944ms] DBG ------ Read 1x512 starting at 1
[31944ms] DBG ---- DATA_IN, syncOffset 12 syncPeriod 53
[31945ms] DBG ------ Found 1 sectors in prefetch cache
[36946ms] scsi_accel_rp2040_stopWrite() timeout
[36946ms] SCSI DMA state: WRITE_DONE
[36946ms] Current buffer: 0x00000000/0x00000000, next 0x00000000 bytes
[36947ms] SyncOffset: 12 SyncPeriod 53
[36947ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 28, instr 0x000080A0
[36948ms] PIO Data SM: tx_fifo 0, rx_fifo 4, pc 3, instr 0x00007909
[36949ms] PIO Sync SM: tx_fifo 4, rx_fifo 0, pc 21, instr 0x0000200A
[36949ms] DMA CH A: ctrl: 0x00000011 count: 0x00000000
[36950ms] DMA CH B: ctrl: 0x01020809 count: 0x00000001
[36950ms] DMA CH C: ctrl: 0x00008805 count: 0x00000000
[36951ms] DMA CH D: ctrl: 0x01011809 count: 0xFFFFFF81
[36952ms] GPIO states: 0x3BF9EEFF
[36956ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 13771
[36957ms] DBG -- BUS_FREE
[36958ms] DBG ---- SELECTION: 4
[36959ms] DBG ---- MESSAGE_OUT
[36959ms] DBG ------ OUT: 0xC0 
[36959ms] DBG ---- COMMAND: Read6
[36959ms] DBG ------ OUT: 0x08 0x00 0x00 0x02 0x01 0x00 
[36960ms] DBG ------ Read 1x512 starting at 2
[36960ms] DBG ---- DATA_IN
[44137ms] DBG BUS RESET
```

jdryyz, 2025-03-17:
Thank you for the update. I might make an attempt to figure out which of the past updates may have triggered the problem since it could have started with something earlier.

aperezbios, 2025-03-17:
@jdryyz I wouldn't bother were I in your shoes, as I'm pretty sure we have a prime suspect. The real question for us to figure out is the best way to go about resolving the issue.

jdryyz, 2025-03-17:
Ohh good. I won't bother then.  :)

PetteriAimonen, 2025-03-17:
@jdryyz Looks like the initial guess at a recent commit causing the issue is not that likely.

If you have time to try, test this release (I expect it might work and the one after that might fail):
https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/v2024.10.16

jdryyz, 2025-03-17:
That firmware version is working.

PetteriAimonen, 2025-03-18:
Based on the debug log, the communication breaks down soon after the host enables synchronous mode.
The symptoms look like the host is missing REQ pulses during the transfer.
This is usually caused by timing violations in the bus communication.

I tested the synchronous mode timings with offset 12, period 53 on the two releases:

| Parameter | SCSI spec | FW 2024-10-16 | FW 2025-03-10 |
| -- | -- | -- | --| 
| REQ period | >= 212 ns |  >= 206 | >=  206 |
| REQ low (assert) | >= 90 ns | 94 .. 100 | 102 .. 126 |
| REQ high (negate) | >= 90 ns | >= 102 | >= 90 |
  
My theory is that the 90 ns negation period is too close to the spec for the Amiga.
This only affects the slowest SCSI-5 sync mode, as fast SCSI supports negation down to 30 ns.

Here is a test build where I have increased the negation period to >= 100 ns.
This also brings the period to >= 214 ns, which agrees better with the negotiated speed.

@jdryyz Can you try this?
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/13922328477

jdryyz, 2025-03-19:
Sorry for the late reply. That firmware is also working.  :)

Incidentally, I upgraded my WD SCSI chip from -04 to -08 today. My understanding is that the -08 rev is mostly about fixing removeable media problems (none attached at the moment). It made another change to my system that I wasn't expecting, however. It seems that it didn't like that my ext. DB25 SCSI port was not terminated. I was seeing some solid SCSI LED activity as well as read errors. Luckily for me I had a DB25 terminator available and also lucky for me I had a bit of a feeling of deja vu from the 1990s.

This is not the first time I tested the -08 in my A3K. The last time I did, I only had access to the beta KS 1.4 dual boot setup. I assumed  the trouble was caused by an incompatibility with the newer -08 chip since those old ROMs were limited. Thanks to a modern kickstart switcher, I can stick with my ks 1.4 setup or switch to 3.2.2, DiagROM and others. I had 3.2.2 selected prior to the -08 going back in, thus my surprise that it still triggered a problem.

The WD -08 chip not liking the lack of ext. termination probably exposes the -04 not following all the rules.  :) I should probably test that external port with my ZIP drive again now that the -08 is installed. It didn't quite work the way I expected when -04 was still in there.

Another thing I should mention-- my RP2040 is connected internally and does have termination enabled. It is also the only physical device attached to the ribbon cable, although I have three virtual drives on the SD media.

aperezbios, 2025-03-19:
@jdryyz thanks for confirming.

PR #539 (2025-03-19): RP2040: Adjust sync 5MB/s mode timings
In synchronous 5MB/s mode, the REQ negation period used to be >= 102 ns. After the RP2040 clocking updates in October 2024, the negation period is >= 90 ns. That is just barely within SCSI-2 specification.

This commit increases the negation period slightly to 100 ns.

This affects only the slowest synchronous mode, as fast SCSI supports negation period down to 30 ns.

Fixes #535.
closed 2025-03-19 by aperezbios