ZuluSCSI/ZuluSCSI-firmware #537: dfu-util possibly incompatible with a specific ZuluSCSI v1.1 bootloader

TheBoctor, 2025-03-18:
Hello,

While carefully following the flashing instructions for dfu-util on Linux and a v1.1 PCB, I was unable to exceed ~28% progress before bailing out with a libusb type -9 error, during a page erase command (dfuse_download) with somewhat random timing between attempts. Uploading the flash from the MCU to the host PC always worked fine, but even this identical backup image could not be flashed back onto the chip. Verbose mode revealed nothing new or unexpected.

Some past issues opened on the dfu-util bug tracker implied this error could be because the microcontroller itself was buggy. I tried previous builds of dfu-util on Arch Linux (packaged versions) to test for any regressions or changes, even recompiling it with more lenient timeouts, but these failed identically. I then tried prebuilt dfu-util 0.11 and 0.8 binaries on a Windows machine with the same results. At this point, I was concerned that I had soft bricked the device.

As a final resort, I made a Windows VM, passed the USB DFU device through to it, installed the proprietary GigaDevice GD32 DFU driver linked in the README*, and ran their official flashing utility for Windows. The 2025.03.10 firmware image then uploaded without issues, and the ZuluSCSI now worked perfectly.
<sub>*The link in README.md may be out of date. It searches for "dfu" on the vendor's website, but this only returns the driver download,  not the required GD32 flashing program.</sub>

Is this a known caveat with a specific batch/revision of ZuluSCSI v1.1 that could warrant a note regarding dfu-util, somewhere in the documentation? Has anyone else encountered this, when upgrading from a preinstalled 2022 firmware?

aperezbios, 2025-03-18:
@TheBoctor you should not need to use DFU to upgrade a V1.1/V1.2 board to newer firmware, and we _**do not**_ recommend you do so unless the firmware on your board is already hosed or otherwise non-functional. 

All you should need to do is copy the V1.1plus .bin file to an SD card, and then power on the board. The custom bootloader will automatically detect the .bin file and proceed to flash the new firmware, which takes a couple of seconds. Please download https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v2025.03.10/ZuluSCSIv1_1_plus_2025-03-10.bin and place it on an SD card.

TheBoctor, 2025-03-18:
Thank you, I did see the SD card method described, but was unsure of whether that feature was available on my model. I may have misread the current version of the README: The recommended method's heading begins with, "For RP2040/RP2350-based boards," whereas the heading over the alternate dfu-util method reads, "For ZuluSCSI V1.1 and V1.2," so I mistakenly thought the feature might be exclusive to newer models, with dfu-util being the only way to flash a v1.x.

Upon reading a 2022 version of the README, I now see that the easier SD flash method has always been supported, so this can be safely dismissed as user error; Sorry for the noise.

closed 2025-03-18 by TheBoctor