ZuluSCSI/ZuluSCSI-firmware #542: ZuluSCSI 1.1 Throws Check Conditions after getting unknown commands ( MESSAGE IN: 0x01 0x02 0x03 0x00)

retro-andie, 2025-03-29:
Platform: ZuluSCSI v1.1+
FW Version: 25.03.19-release Mar 19 2025 16:53:12
Board Version: ZuluSCSI v1.1 Standard Edition

HW: IBM RS/6000 Model 43P-140 
Attached are the debug logs from attempts to boot a CD-ROM of AIX 4.1.5, AIX 4.2.1, and AIX 4.3.3. AIX 4.3.3 succeeds without issue.
The ISO files were named CD6_512.iso. This was done attempted on both an older revision and latest version of the firmware.

Previous experience with a ZuluSCSI 1.1 on a Model 250 had no issues with booting AIX 4.1.5 from a CD-ROM. This is likely a hardware specific issue. I was able to verify with a physical CD-ROM that the ISO files that I used were good.

[zululog415.txt](https://github.com/user-attachments/files/19523035/zululog415.txt)
[zululog421.txt](https://github.com/user-attachments/files/19523034/zululog421.txt)
[zululog433.txt](https://github.com/user-attachments/files/19523033/zululog433.txt)

morio, 2025-04-07:
@retro-andie this is an attempt to fix the issue https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/14321597366
It looks like the host is sending commands to LUNs outside LUN 0, which is the only one supported. It also looks like it is trying to enable a wider bus when it fails.

The code attempts to send the proper message when it is querying for wide bus access. It also fixes a LUN check.

Not sure if these changes will fix it as I don't have a test machine.

retro-andie, 2025-04-09:
@morio I tried the firmware from your private build. It didn't fix the issue. I then tried the latest and several back version. It looks like 25.02.21 was the last version where booting the AIX 4.3 CD worked. It did not work in subsequent releases

[zululog415_test_nolun_bad.txt](https://github.com/user-attachments/files/19659078/zululog415_test_nolun_bad.txt)
[zululog421_lun_test_bad.txt](https://github.com/user-attachments/files/19659080/zululog421_lun_test_bad.txt)
[zululog421_nolun_test_bad.txt](https://github.com/user-attachments/files/19659081/zululog421_nolun_test_bad.txt)
[zululog433_250319_bad.txt](https://github.com/user-attachments/files/19659077/zululog433_250319_bad.txt)
[zululog433_latest_mainline_bad.txt](https://github.com/user-attachments/files/19659079/zululog433_latest_mainline_bad.txt)

During testing, the device hard crashed and I am now trying to recover it through the DFU utility. 
The link provided http://www.gd32mcu.com/en/download?kw=dfu&lan=en in the README.md file only shows the USB drivers. What is the recommended tool to flash the firmware directly over USB?

aperezbios, 2025-04-09:
@retro-andie have you tried recovering the firmware with the .bin file? Unless the custom bootloader got clobbered in flash, that should work. Just try placing https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v2025.02.21/ZuluSCSIv1_1_plus_2025-02-21.bin on an SD card and powering on the ZuluSCSI. If the custom bootloader finds it, the LED will rapidly flash for 1-2 seconds, and then it will re-boot and be running the latest firmware. Please confirm you're not able to recover using this method before we proceed.

retro-andie, 2025-04-09:
@aperezbios I was not able to recover using that method. This is why I'm trying the DFU recovery method.

retro-andie, 2025-04-09:
I was able to recover and get the device back to 25.02.21. I'll try to run debug logs again tomorrow to against https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/14321597366 and other releases. Are there any specific settings that you want to see logs for?

aperezbios, 2025-04-09:
@retro-andie we have some local copies. You can grab https://www.ZuluSCSI.com/GD32%20DFU%20Tool%20and%20Driver.zip
and https://github.com/ZuluSCSI/ZuluSCSI.com/blob/gh-pages/GD32-DFU-Windows.zip

Under Windows, you need to install the DFU driver first. Then, start-up the DFU utility itself. It will tell you there's an update available, and prompt you to download it, but _ignore this_ as it's bogus.

Let me know if you have any further questions or problems.

retro-andie, 2025-04-09:
@aperezbios thank you for the links. I was able to recover the device by using the DFU. 
I'll try some more testing tomorrow and see what I can capture for logs.

morio, 2025-04-09:
@retro-andie another user posted an issue installing Solaris with a CD image with 512 byte sector size.
I think this fixes his issue it might fix yours too.
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/14368971209

retro-andie, 2025-04-10:
I was able to test out that test build as well as the latest 25.04.10 release.
AIX 4.3.3 boots in both cases and failed with AIX 4.1.5 and AIX 4.2.1.
I was able to verify that this is a hardware-specific issue. I have a RS/6000 Model 250 that was able to boot AIX 4.1.5 on both Revision 22 and 25 of the ZuluSCSI 1.1 firmware. 

[zululog_415_250410_failed.txt](https://github.com/user-attachments/files/19695410/zululog_415_250410_failed.txt)
[zululog_415_failed.txt](https://github.com/user-attachments/files/19695415/zululog_415_failed.txt)
[zululog_421_250410_failed.txt](https://github.com/user-attachments/files/19695412/zululog_421_250410_failed.txt)
[zululog_421_failed.txt](https://github.com/user-attachments/files/19695416/zululog_421_failed.txt)
[zululog_433_250410_good.txt](https://github.com/user-attachments/files/19695413/zululog_433_250410_good.txt)
[zululog_433_good.txt](https://github.com/user-attachments/files/19695409/zululog_433_good.txt)
[zululog415_rev22_failed.txt](https://github.com/user-attachments/files/19695411/zululog415_rev22_failed.txt)
[zululog433_rev22_good.txt](https://github.com/user-attachments/files/19695414/zululog433_rev22_good.txt)

closed 2025-04-11 by aperezbios