ZuluSCSI/ZuluSCSI-firmware #556: Drive geometry ignored (IRIX fx)

CB-HK52, 2025-04-13:
RP2040 ZuluSCSI devices appear to function without issue when partitioning images using IRIX's fx utility. Drive geometry settings in the .ini file do not appear to affect fx's ability to handle partitioning despite the zulu log stating the sector and head settings do not allow for divisibility and may cause issues. IRIX can be installed and functions without issue.

RP2350B Zulu Blaster devices do not seem to behave the same way. The .ini file does not appear to be readable under fx. The image size is divisible per the default 255 / 63 settings in the .ini file, yet fx appears to search for the geometry settings and is unable to read what the .ini file has set. This causes fx to issue errors and prevents proper partitioning of the image file.

The behavior of the .ini file as well as some other unknown parameter appears to be different between the 2040 and Blaster versions. fx seems to have no issue with the 2040 version even when the .ini file uses default drive settings resulting in divisibility issues. The Blaster version, however, seems to have issues regardless of whether the image file is divisible by the default settings or not.

Logs and .ini files are attached for reference. Please let me know if there's any other information I can provide for troubleshooting. Thank you!

[zuluscsi (2040).txt](https://github.com/user-attachments/files/19726729/zuluscsi.2040.txt)

[zululog (2040).txt](https://github.com/user-attachments/files/19726731/zululog.2040.txt)

[IRIX 6.2 FX (Zulu 2040).txt](https://github.com/user-attachments/files/19726732/IRIX.6.2.FX.Zulu.2040.txt)

[zuluscsi (blaster).txt](https://github.com/user-attachments/files/19726734/zuluscsi.blaster.txt)

[zululog (blaster).txt](https://github.com/user-attachments/files/19726735/zululog.blaster.txt)

[IRIX 6.2 FX (Zulu Blaster).txt](https://github.com/user-attachments/files/19726737/IRIX.6.2.FX.Zulu.Blaster.txt)

PetteriAimonen, 2025-04-15:
From the logs it appears that both RP2040 and Blaster have problems responding to the host:

<pre>
[37621ms] DBG -- BUS_BUSY
[37621ms] DBG ---- SELECTION: 1
[37621ms] DBG ---- MESSAGE_OUT
[37621ms] DBG ------ OUT: 0xC0 
[37622ms] DBG ---- COMMAND: ModeSense
[37622ms] DBG ------ OUT: 0x1A 0x00 0x3F 0x00 0xFF 0x00 
[37623ms] DBG ---- DATA_IN, syncOffset 10 syncPeriod 25
[42624ms] scsi_accel_rp2040_stopWrite() timeout
[42624ms] SCSI DMA state: WRITE_DONE
[42624ms] Current buffer: 0x00000000/0x00000000, next 0x00000000 bytes
[42625ms] SyncOffset: 10 SyncPeriod 25
[42625ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 28, instr 0x000080A0
[42626ms] PIO Data SM: tx_fifo 0, rx_fifo 4, pc 3, instr 0x00007409
[42627ms] PIO Sync SM: tx_fifo 4, rx_fifo 0, pc 21, instr 0x0000201B
[42627ms] DMA CH A: ctrl: 0x00000011 count: 0x00000000
[42628ms] DMA CH B: ctrl: 0x04082009 count: 0x00000001
[42628ms] DMA CH C: ctrl: 0x00022005 count: 0x00000000
[42629ms] DMA CH D: ctrl: 0x04046009 count: 0xFFFFFFFF
[42630ms] GPIO states: 0xCCAFF040
[42630ms] DBG ---- Total IN: 112 OUT: 0 CHECKSUM: 27247
[42630ms] DBG ---- STATUS: 0 GOOD
[42637ms] DBG -- BUS_FREE
</pre>

From the logs it appears you have termination disabled on the ZuluSCSI board. Do you have a terminator at each end of the SCSI cable?

Can you try replacing the existing setting with `MaxSyncSpeed = 0` in `zuluide.ini` to disable synchronous mode, in case that is what is causing the communication issues?

CB-HK52, 2025-04-15:
I do indeed have a terminator at each end of the SCSI cable.

As for the MaxSyncSpeed, I went ahead and set it to zero on both the 2040 and Blaster. Interestingly, the 2040 continued to operate normally, as it had before, without any issues or changes to its behavior--it stopped displaying the sync error, though. The Blaster, however, is now working correctly! Which is very interesting. I had a bad assumption that the speed setting was required in order to attain the appropriate SCSI transfer speed. It appears that is not the case though.

I was able to successfully see the Blaster in fx and repartition the drive.

Thank you for the assistance! I'm not sure why the behavior isn't the same with the 2040 (as far as not being able to be seen by fx, since I've been incorrectly configuring my ZuluSCSI 2040 boards and somehow they've been working just fine with the bad max sync setting). If you'd like me to pull any other logs for the 2040, please let me know.

[IRIX 6.2 FX (Zulu Blaster) no sync.txt](https://github.com/user-attachments/files/19759999/IRIX.6.2.FX.Zulu.Blaster.no.sync.txt)

[zululog (blaster) no sync.txt](https://github.com/user-attachments/files/19760000/zululog.blaster.no.sync.txt)

PetteriAimonen, 2025-04-15:
This sounds like some signal integrity issues which make the synchronous mode transfer fail. I think for RP2040, it works well enough that the data gets transferred, even though ZuluSCSI thinks there is a problem. For Blaster, it fails more. There are necessarily some timing differences between the RP2040 model and Blaster, even though both are within SCSI spec.

If the transfer speed seems adequate, operating with `MaxSyncSpeed = 0` is reasonable. That's limited to around 5 MB/s. The synchronous mode would give higher transfer speed, but demands more of the cabling. If you want to experiment with it, I would suggest trying another cable and testing whether using ZuluSCSI termination instead of the cable terminator makes a difference.

CB-HK52, 2025-04-16:
I did indeed run into an issue as soon as I attempted to set a sync speed above 5. 10 caused the issue to appear again, with the logs showing an issue between synchronous and asynchronous. I saw the same failure with fx not being able to understand the drive geometry once it sent the query.

As far as hardware goes, the cable is short at only six inches, and is terminated almost right after the 50 pin interface to the Blaster.

[zululog (blaster) sync 10.txt](https://github.com/user-attachments/files/19770399/zululog.blaster.sync.10.txt)

CB-HK52, 2025-05-03:
I went ahead and tried the ZuluSCSI Blaster in an Indigo2 and experienced a similar issue. Updated to newest firmware from 1 May.

It looks like there's a read/write timeout occurring.

[zululog.txt](https://github.com/user-attachments/files/20024698/zululog.txt)

aperezbios, 2025-05-03:
@CB-HK52 thanks for the log. I see that you have SCSI termination disabled. Is that intentional? It seems likely to be incorrect, but without knowing what your SCSI device configuration looks like, I can't comment with certainty.

This debug log shows everything operating normally for about 84 seconds, before the first write timeout, so it could easily be termination-related.

CB-HK52, 2025-05-04:
@aperezbios for the Indy and Indigo2 there's termination on both ends of the SCSI chain (internal and external) and I've found that enabling termination hasn't affected this behavior. I've left it off in order to ensure there are no issues with the CD-ROM drive.

Here's the Indigo2 chain:  EXT. TERM ---  WD93 Controller --- Device 1 (ZuluSCSI) --- Device 2 --- Device 3 (CD-ROM) --- INT. TERM

I ran the same .ini file for both the RP2040 and RP2350B. I made sure both were on the same firmware as well. Interestingly, the RP2040 works with zero issues. I was able to use fx to create a partition and then was able to install IRIX with no issue. Again, only using the RP2040 with the same bus configuration, same ZuluSCSI firmware, same .ini file, and same HD file (I used the same SD card for testing).

I have an impression that something is going on that is unique to the 2350 that is not happening on the 2040. Not sure if it's speed related or something else. I've attached the log for the 2040 and the .ini file I used.

[zululog (2350).txt](https://github.com/user-attachments/files/20026121/zululog.2350.txt)
[zululog (2040).txt](https://github.com/user-attachments/files/20026122/zululog.2040.txt)

[zuluscsi_ini.txt](https://github.com/user-attachments/files/20026125/zuluscsi_ini.txt)

PetteriAimonen, 2025-05-14:
I think I found the Blaster issue with SYNC-5 mode, and it should be fixed in this test build:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/15016308222

CB-HK52, 2025-05-17:
Unfortunately the issue I'm running into with the 2350 hasn't been resolved with that fix. I'm still seeing the same issues with reading geometry and then failing to be recognized correctly.

[zululog.txt](https://github.com/user-attachments/files/20265648/zululog.txt)

aperezbios, 2025-05-17:
@CB-HK52 based on this most recent log, I don't think this has _anything_ to do with drive geometry being ignored. I suspect this may be an incorrect conclusion, based on what I see in the l og. Without further details as to what your SCSI chain and termination configuration looks like, I don't think we'll be able to make headway here. 

I see a lot of "DBG ---- COMMAND: Unknown" in the later parts of the log, which makes me suspect termination isn't quite right here. 

Please describe, in detail, your SCSI chain set up, buth configurationi and termination. It would also be helpful if you could describe the exact make, model, OS and OS version of the _current_ machine you're testing with.

In a previous comment, you wrote "I did indeed run into an issue as soon as I attempted to set a sync speed above 5. " but it's not clear which version of the firmware you were using when you made that comment. Can you confirm that if you set MaxSyncSpeed to 5 with the current/latest release firmware, these issues go away?

PetteriAimonen, 2025-05-19:
@CB-HK52 Thank you for testing!

Looks like I misremembered the gist of this issue, for some reason I remembered that SYNC-5 didn't work either.
I now double-checked the timings for SYNC-10 mode. They were within SCSI specification, but there was room for improving the timing margin. Here is another build if you can try:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/15107624283

The ZuluSCSI Blaster timing configuration has ended up quite complex in the push to get the best performance out of the hardware. Most of the testing has gone towards the fastest modes, but of course the intent is for the slower modes to have best possible compatibility also.

CB-HK52, 2025-06-06:
@PetteriAimonen this last update works! Thank you! I can report back that SYNC-10 mode is working with no issues. fx is correctly reading the disk geometry and modifying the partition of the disk. I can also install IRIX without any issue.

I will note that I have to leave termination disabled on the board. If I have it enabled, I end up with a stream of parity errors. Leaving termination off on the board isn't an issue based on being able to use the system without problem. The SCSI chain remains terminated on both ends even with disabling termination on the Blaster. I have no errors running in this configuration.